esp8266/Arduino #2331: Exception after compiling

jonny190, 2016-07-26:
I have a issue when i compile the below script when it runs i get an Exception that decodes to 

`Decoding 4 results
0x40203fe5: PubSubClient::connected() at C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x40201f59: loop at C:\Users\Jonny\Documents\Arduino\libraries\FastLED/controller.h line 170
0x40203c68: loop_wrapper at C:\Users\Jonny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Jonny\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
`
Any assistance would be greatly appreciated 

`
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#include "FastLED.h"

// UPDATE
const char* ssid = "*";
const char* password = "*";
const char* mqtt_server = "192.168.0.1";
#define LED_COUNT 24
#define PIN_LEDS D6

// END

WiFiClient espClient;
PubSubClient client(espClient);
CRGB leds[LED_COUNT];
long lastMsg = 0;
char msg[50];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is acive low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("testfastledcrash")) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  FastLED.addLeds<NEOPIXEL, PIN_LEDS>(leds, LED_COUNT);
  FastLED.setBrightness(120);
  for (uint8_t i = 0; i < 1; i++) {
    for (uint8_t dot = 0; dot < LED_COUNT; dot++) {
      leds[dot] = CRGB::Blue;
      FastLED.show();
      delay(20);
      leds[dot + 1] = CRGB::Red;
      FastLED.show();
      delay(20);
    }
  }
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 75, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}`

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

me-no-dev, 2016-07-29:
your sketch code is really hard to read. Look at how the example sketch is pasted and edit the above so we can see indentations and color coding. Core version is also important.

devyte, 2017-10-14:
@jonny190 I see:
`char msg[50];
`
but then:
`    snprintf (msg, 75, "hello world #%ld", value);
`
I don't think you're hitting size 50, though.

I see:
```
    for (uint8_t dot = 0; dot < LED_COUNT; dot++) {
      leds[dot] = CRGB::Blue;
      FastLED.show();
      delay(20);
      leds[dot + 1] = CRGB::Red;
      FastLED.show();
      delay(20);
    }
```

you are indexing and assigning `leds[dot + 1]`, which for the last iteration is past the end of the array.

If you need further assistance, please refer to a community forum for discussion. 
Closing due to code errors, see #3655 .

closed 2017-10-14 by devyte