esp8266/Arduino #2332: Memory loss of 184 bytes at each request

marcoschwartz, 2016-07-27:
Hello,

I am the creator of the aREST library, which is also compatible with the ESP8266 and uses this library to work. We found out that at each request to the board (the sketch is running a web server on the board), we loose 184 bytes of memory. I think it's related to this issue:

https://github.com/esp8266/Arduino/issues/1923

I confirmed that it's indeed due to this, as the memory comes back to normal after a while. However, for our library we often need to process a lot of requests/second, so after a while the board crashes. I read that this 'issue' with the TCP protocol is solved by the ESP8266WebServer library, so how to actually do it for a generic code using the WiFiServer library? Thanks!

igrr, 2016-07-27:
FYI, @me-no-dev did a modification to LwIP stack, now the number of connections kept in TIME_WAIT state is limited. So the memory "loss" should by limited by `MEMP_NUM_TCP_PCB_TIME_WAIT * 184`, and `MEMP_NUM_TCP_PCB_TIME_WAIT` is set to 5 in lwipopts.h.

Regarding ESP8266WebServer, the trick is to ask the client to close the connection instead of closing it on the ESP. In TCP, the party which does "active close" has to keep the connection for a long time, while the other party doesn't have to. ESP8266WebServer sends "Connection: close" header to the client, and then waits until client closes the connection.

closed 2016-08-01 by igrr

Xiaoli, 2016-11-15:
Thanks for the changes of LwIP stack #2260 made by @me-no-dev

CircuitSerialKiller, 2017-02-07:
Yes! Thanks so much for adding this @me-no-dev!!
Under no circumstances should it be acceptable to have any code that just lets the ESP8266 crash. 
Hopefully we can get a stable version on github soon... I noticed something was changed after 2.3.0 that easily makes the heap run out and TCP send fails with higher latency connections:
(Issue 2925).