esp8266/Arduino #2334: Nodemcu 2 HTTP arguments problem

mbostic, 2016-07-27:
I am using one nodemcu as web server and another nodemcu as client to connect to it via HTTP. I am using get method. It works, but only with one argument. If i use two arguments like this (client nodemcu): `http.begin("http://192.168.4.1/command?c=2&ip="+WiFi.localIP()); `
it doesnt work. but if I connect to server nodemcu with the same url with web browser, code works and server handles request as expected (reads arguments correctly). This means that problem is on client nodemcu.

client code:
```
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

#define USE_SERIAL Serial

void setup() {
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network
    WiFi.mode(WIFI_STA);
    WiFi.begin("ESPap", "thereisnospoon");

    Serial.println();
    Serial.println();
    Serial.print("Wait for WiFi... ");

    while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

}

void loop() {
  
        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        http.begin("http://192.168.4.1/command?c=2&ip="+WiFi.localIP());    

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
   

    delay(10000); 
}
```

closed 2016-07-27 by mbostic

ymgupta, 2017-04-12:
http.begin() takes 3 arguments. 
Try using it this way:
http.begin(ip address of server, 80, /resource path here);

iit2015143, 2017-10-06:
Did you get it resolved? I will appreciate fast response. I am stuck at the same problem .

josenjose, 2019-07-25:
I have this issue . Can u help

NayLynnYinMaung, 2019-09-07:
I know it's closed but I also have httpcode=-1 connection refused upon httpGet like that...