esp8266/Arduino #2335: ESP8266 crashes after writing to StaticJsonBuffer

forthlightning, 2016-07-27:
Hardware
Hardware:			ESP8266 Huzzah 
Core Version:      	        2.3.0-rc2

Description

I am reading sensor values and storing in a StaticJsonBuffer. These values will eventually be sent in an HTTP request, but that's not important for the moment.

For reference I'm using the ArduinoJson.h library.

The function sensorsToJSON() is called each time the sketch enters the loop. The sketch seems to make it past sending the buffer string to the Serial monitor, but then before it can loop a second time, the board crashes and displays this stack of error traces. Any help would be appreciated!

### Settings in IDE

Module:  			Adafruit Huzzah ESP8266
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:     80Mhz
Flash Frequency:	40Mhz
Upload Using:		SERIAL

### Sketch

Here is the part of the sketch that is throwing errors: 

String sensorsToJSON() {

	tempSensors.requestTemperatures();
	float T1 = tempSensors.getTempCByIndex(0);
	float T2 = tempSensors.getTempCByIndex(1);
	String T1String = String(T1);
	String T2String = String(T2);

	const int BUFFER_SIZE = JSON_OBJECT_SIZE(10);
	StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;
	Serial.print("Size of Sensor Buffer: ");
	Serial.println(BUFFER_SIZE);
	JsonObject& sensorArray = jsonBuffer.createObject();

	sensorArray["temp1"] = T1String;
	sensorArray["temp2"] = T2String;

	int size = 200;
	char buffer[size];
	sensorArray.printTo(Serial);
	Serial.print("Size of Sensor Array: ");
	Serial.println(sizeof(sensorArray));

	String temp = sensorArray.get<String>("temp1");
	Serial.println(temp);

	Serial.println("json buffered...");
}

### Debug Messages

Here is output from the exception decoder tool:

0x4010020c: _umm_free at /Users/maxfieldmorrison/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1287
0x4010068c: free at /Users/maxfieldmorrison/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c line 1733
0x402010fe: delay at /Users/maxfieldmorrison/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_wiring.c line 53
0x40209c58: String::~String() at /Users/maxfieldmorrison/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/WString.cpp line 519
0x4020298e: loop at /Users/maxfieldmorrison/Desktop/NSR/Sensor Board/solSense/sensor-board/solSenseMQTT_JSON_AutoConnect_v2/solSenseMQTT_JSON_AutoConnect_v2.ino line 378
0x4020a75c: loop_wrapper at /Users/maxfieldmorrison/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at /Users/maxfieldmorrison/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S line 109

And here is some serial output:

*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 172.20.10.3
Time Status: 0
Current Time is: 16
Wifi Initialized... IP number assigned by DHCP is 172.20.10.3
Starting UDP
Local port: 8888
waiting for sync
Transmit NTP Request
us.pool.ntp.org: 104.156.99.226
Receive NTP Response
$�  �  h�{H�C������        �C�{֥�=�C�{֩��Setup Complete...
Begin Loop...
Size of Sensor Buffer: 168
{"temp1":"23.06","temp2":"23.25"}Size of Sensor Array: 8
23.06
json buffered...

Exception (3):
epc1=0x4010010e epc2=0x00000000 epc3=0x00000000 excvaddr=0x400214a9 depc=0x00000000

ctx: cont 
sp: 3fff0530 end: 3fff0750 offset: 01a0

me-no-dev, 2016-07-28:
maybe have us see the rest of the sketch since the exception happens after exiting the method above. Also having it return String by reference might be better and actually returning the string will be great.
```cpp
String & sensorsToJSON() {
  ......
  return temp;
}
```

igrr, 2016-07-28:
doing something like this is not recommended:
```c++
String getStuff() {
 // no return statement
}
```

doing this is not recommended either:
String& getStuff() {

closed 2016-08-01 by igrr