esp8266/Arduino #2338: WiFi.scanNetwork() and websockets

Def152, 2016-07-28:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	        2.3.0

### Description

The problem is in my scanNet() function and to be more specific in WiFi.scanNetworks(), I'm in STA mode trying to scan networks around me and send them using websockets to my config page. Unfortunately WiFi.scanNetworks() returns 0 all the time. The websockets connection stays connected. At this moment I'm using STA mode because it is easier to test code. In future I would like scan in AP mode.

Websockets library I'm using: https://github.com/Links2004/arduinoWebSockets

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck 

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <WebSocketsServer.h>
#include <Hash.h>
#include <Ticker.h>
#include <FS.h>
#include <ArduinoJson.h>
#include <SPI.h>

const char* ssid = ".....";
const char* password = ".....";

//text/plain

ESP8266WebServer server(80);
WebSocketsServer webSocket = WebSocketsServer(81);
Ticker timer;
int value = 200;

void scanNet(){
  StaticJsonBuffer<200> jsonBuffer;
  int n = WiFi.scanNetworks();
  if (n == 0){
    Serial.println("no network");
  }
   else
   {
     JsonObject& json = jsonBuffer.createObject();
     json["action"] = "getNet"; // action name
     for (int i = 0; i < n; ++i)
     {
       char buffer[81] = "";
       // Print SSID and RSSI for each network found
       json["ssid"] = WiFi.SSID(i);
       json["rssi"] = WiFi.RSSI(i);
       json["enc"] = WiFi.encryptionType(i);
       json.printTo(buffer, sizeof(buffer));
       webSocket.broadcastTXT(buffer, sizeof(buffer));
       //String strn = String(buffer);
       //webSocket.broadcastTXT(strn);
       //Serial.print(buffer);
    }
  }
}

void broadcastINT(int *value){
  //
  String string = String(*value);
  //char string[17];
  //itoa(value, string, 10);
  webSocket.broadcastTXT(string);
}

void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t lenght) {

    switch(type) {
        case WStype_DISCONNECTED:
            Serial.printf("[%u] Disconnected!\n", num);
            break;
            timer.detach();
        case WStype_CONNECTED:
            {
                IPAddress ip = webSocket.remoteIP(num);
                Serial.printf("[%u] Connected from %d.%d.%d.%d url: %s\n", num, ip[0], ip[1], ip[2], ip[3], payload);
                timer.attach_ms(5000, scanNet);
            }
            break;
        case WStype_TEXT:
            Serial.printf("[%u] get Text: %s\n", num, payload);
//            // echo data back to browser
          //broadcastINT(value);
      //webSocket.sendTXT(num, payload, lenght);
//
//            // send data to all connected clients
      //webSocket.broadcastTXT(payload, lenght);
            break;
        case WStype_BIN:
            Serial.printf("[%u] get binary lenght: %u\n", num, lenght);
//            hexdump(payload, lenght);
//
//            // echo data back to browser
      //webSocket.sendBIN(num, payload, lenght);
            break;
    }

}

void sendFile(String name, String type){
     bool result = SPIFFS.begin();
  Serial.println("SPIFFS opened: " + result);

  // this opens the file in read-mode
  File f = SPIFFS.open("/" + name, "r");

  if (!f) {
    Serial.println("File open failed" + name );
    return;
  }
  server.streamFile(f, type);
  f.close();
}

void handleRoot() {
  sendFile("index.html", "text/html");
}

void handleAdmin() {
  sendFile("admin.html", "text/html");
}

void handleAdminJS (){
    sendFile("admin.js", "application/javascript");
}

void handleJquery (){
    sendFile("jquery.js.gz", "application/javascript");
}

void handleChart (){
    sendFile("highcharts.js.gz", "application/javascript");
}

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  //WiFi.mode(WIFI_AP);
  //WiFi.softAP(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  server.on("/", handleRoot);
  server.on("/admin", handleAdmin);
  server.on("/admin.js",handleAdminJS);
  server.on("/jquery.js",handleJquery);
  server.on("/highcharts.js", handleChart);
  //server.on("/", handleJS);

  server.begin();
  Serial.println("HTTP server started");
  webSocket.begin();
  webSocket.onEvent(webSocketEvent);
  Serial.println("WebSocket init");
}

void loop() {
  server.handleClient();
  webSocket.loop();

}
```

Def152, 2016-07-29:
This is happening because of Ticker. If I try to set flag in timer callback and put scanNet in void loop() it is working perfectly. 

Also from library documentation 

> It is currently not recommended to do blocking IO operations (network, serial, file) from Ticker callback functions. Instead, set a flag inside the ticker callback and check for that flag inside the loop function.

closed 2016-07-29 by Def152

PyramidSlayer, 2017-03-01:
I had the same problem and I made the conclusion the blocking functions as WiFi.scanNetworks() (is a network operation) aren't accepted by the ticker function, If I put my callback function containing WiFi.scanNetworks() in the loop without the ticker, it works perfectly.