esp8266/Arduino #2339: Unable to upload .bin file using "esp8266 flash download tool"

romuye123, 2016-07-28:
### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	        2.3.0

### Description

Generated a .bin file using "Export Compiled Binary"  with following settings in IDE.
Used EEPROM in the program. When we upload program using "esp8266 flash download tool" it uploads correctly for the first time. When we do some EEPROM related action it gives exception and restarts and repeated exception on boot "Exception (28):"

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:	        dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

me-no-dev, 2016-07-29:
It seems that you do not have problem uploading bin files, but instead probably have selected wrong flash size and the EEPROM lib overwrites system block. The information you give is way not enough for better guess. Please attach code and decoded exception stack trace

igrr, 2016-08-01:
Could you please paste a screenshot of your settings in "esp8266 flash download tool"?

romuye123, 2016-08-02:
@igrr 
Here are settings in "esp8266 flash download tool":

[image]

@me-no-dev 
And the program is very basic program which uses storing few variables and reading them using EEPROM.

`
void setup() {
  Serial.begin(115200);  
  EEPROM.begin(1124);
}
`

romuye123, 2016-08-02:
Exception occurring after you do some EEPROM related action. And the program restarts and exception appears and the cycle continues.......

[image]

romuye123, 2016-08-04:
@igrr @me-no-dev 
Any update on this ? Still not able to resolve the problem ?

me-no-dev, 2016-08-04:
There is [this](https://github.com/me-no-dev/EspExceptionDecoder) to decode that stack trace and I have no knowledge of that "esp8266 flash download tool". We do not use that.

romuye123, 2016-08-05:
@me-no-dev 
what should be used to upload bin files on esp12e? is there any other GUI tool ?
also request you to mention the address parameters on where the generated .bin file to uploaded ?

igrr, 2016-08-05:
With the settings as in above photo it should work correctly. I'll try to reproduce.
Just to be sure i got this right, did you select "Generic" module in the IDE and "4M(3M SPIFFS)" as flash size (in Tools menu)?

me-no-dev, 2016-08-05:
I use Arduino IDE to upload my sketches, which uses esptool-ck, which can be called from terminal (command prompt). Why are you using something else and not the Arduino IDE itself?

romuye123, 2016-08-05:
@igrr: Yes following are my correct settings in tools menu

Module: Generic ESP8266 Module
Flash Size: 4M(3M SPIFFS)
CPU Frequency: 80Mhz
Flash Mode: dio
Flash Frequency: 40Mhz
Upload Using: SERIAL
Reset Method: ck

I generate **myapp.cpp.bin** file using Arduino IDE with above settings by pressing Verify button. 
Then using 'ESP download tool' i upload the myapp.cpp.bin file on 0x00000 address and it uploads at very first time but gives eeprom related issues. Then after that it keeps giving Exception (28):

@me-no-dev Yes i also use Arduino IDE itself to upload sketch. But i want to share the binary file to someone else and do not want to share the source code with him. Thats why asking is there any other simple option to upload? Something like select file and upload as mentioned in my earlier screenshot.

krzychb, 2016-08-05:
> But i want to share the binary file to someone else and do not want to share the source code with him. Thats why asking is there any other simple option to upload?

Hi @romuye123,

Did you consider uploading from a [web browser](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#web-browser)?

Krzysztof

roboticboyer, 2016-08-09:
Why don't use directly the esptool to write the bin file?
It's the upload tool of Arduino Ide for ESP.
It can be found inside the Arduino Ide folders:
packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd ck -cb 115200 -cp /dev/ttyUSB1 -ca 0x00000 -cf /directories/file.bin

devyte, 2017-10-14:
Not the right place to ask for this kind of help. Please refer to a forum for discussion.
Closing per #3655 .

closed 2017-10-14 by devyte