esp8266/Arduino #2341: ESPAsyncWebServer

kiralikbeyin, 2016-07-29:
I uploaded same sketch maybe 10 times now it is giving exception again...

Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(tm*)':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope
strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
^
Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t)':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:72:60: error: 'gmtime' was not declared in this scope
return setLastModified((struct tm *)gmtime(&last_modified));
^
Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified()':
/Users/EvAkilli/Documents/Arduino/libraries/ESPAsyncWebServer-master/src/WebHandlers.cpp:77:25: error: 'time' was not declared in this scope
if(time(&last_modified) == 0) //time is not yet set

@igrr Can it be a core problem with time lib?

more : https://github.com/me-no-dev/ESPAsyncWebServer/issues/60

igrr, 2016-07-29:
Did you include `<time.h>` in WebHandlers.cpp?
Also please describe your environment: which versions of the ESP8266 Arduino core and ESPAsyncWebServer are you using, how you are building the code (Arduino IDE or other build system), and the minimal sketch to reproduce the issue. Thank you.

kiralikbeyin, 2016-07-29:
Yes

WebHandlers.cpp
```
#include "ESPAsyncWebServer.h"
#include "WebHandlerImpl.h"
#include <time.h>

AsyncStaticWebHandler::AsyncStaticWebHandler(const char* uri, FS& fs, const char* path, const char* cache_control)
  : _fs(fs), _uri(uri), _path(path), _default_file("index.htm"), _cache_control(cache_control), _last_modified("")
{
  // Ensure leading '/'
  if (_uri.length() == 0 || _uri[0] != '/') _uri = "/" + _uri;
  if (_path.length() == 0 || _path[0] != '/') _path = "/" + _path;

  // If path ends with '/' we assume a hint that this is a directory to improve performance.
  // However - if it does not end with '/' we, can't assume a file, path can still be a directory.
  _isDir = _path[_path.length()-1] == '/';

  // Remove the trailing '/' so we can handle default file
  // Notice that root will be "" not "/"
  if (_uri[_uri.length()-1] == '/') _uri = _uri.substring(0, _uri.length()-1);
  if (_path[_path.length()-1] == '/') _path = _path.substring(0, _path.length()-1);

  // Reset stats
  _gzipFirst = false;
  _gzipStats = 0xF8;
}

AsyncStaticWebHandler& AsyncStaticWebHandler::setIsDir(bool isDir){
  _isDir = isDir;
  return *this;
}

AsyncStaticWebHandler& AsyncStaticWebHandler::setDefaultFile(const char* filename){
  _default_file = String(filename);
  return *this;
}

AsyncStaticWebHandler& AsyncStaticWebHandler::setCacheControl(const char* cache_control){
  _cache_control = String(cache_control);
  return *this;
}

AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(const char* last_modified){
  _last_modified = String(last_modified);
  return *this;
}

AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(struct tm* last_modified){
  char result[30];
  strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
  return setLastModified((const char *)result);
}
//#ifdef ESP8266
AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t last_modified){
  return setLastModified((struct tm *)gmtime(&last_modified));
}

AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(){
  time_t last_modified;
  if(time(&last_modified) == 0) //time is not yet set
    return *this;
  return setLastModified(last_modified);
}
//#endif

..................
```

WebHandlerImpl.h
```
#ifndef ASYNCWEBSERVERHANDLERIMPL_H_
#define ASYNCWEBSERVERHANDLERIMPL_H_

#include "stddef.h"
#include <time.h>

class AsyncStaticWebHandler: public AsyncWebHandler {
  private:
    bool _getFile(AsyncWebServerRequest *request);
    bool _fileExists(AsyncWebServerRequest *request, const String path);
    uint8_t _countBits(const uint8_t value);
  protected:
    FS _fs;
    String _uri;
    String _path;
    String _default_file;
    String _cache_control;
    String _last_modified;
    bool _isDir;
    bool _gzipFirst;
    uint8_t _gzipStats;
  public:
    AsyncStaticWebHandler(const char* uri, FS& fs, const char* path, const char* cache_control);
    bool canHandle(AsyncWebServerRequest *request);
    void handleRequest(AsyncWebServerRequest *request);
    AsyncStaticWebHandler& setIsDir(bool isDir);
    AsyncStaticWebHandler& setDefaultFile(const char* filename);
    AsyncStaticWebHandler& setCacheControl(const char* cache_control);
    AsyncStaticWebHandler& setLastModified(const char* last_modified);
    AsyncStaticWebHandler& setLastModified(struct tm* last_modified);
  //#ifdef ESP8266
    AsyncStaticWebHandler& setLastModified(time_t last_modified);
    AsyncStaticWebHandler& setLastModified(); //sets to current time. Make sure sntp is runing and time is updated
//  #endif
};

...............

```

igrr, 2016-07-29:
Just to repeat, please describe your environment: which versions of the ESP8266 Arduino core and ESPAsyncWebServer are you using, how you are building the code (Arduino IDE or other build system), and the minimal sketch to reproduce the issue. Thank you.

kiralikbeyin, 2016-07-29:
sketch https://github.com/kiralikbeyin/convert-to-async

Tried with 1.6.8 and 1.6.9 no other platform
esp 2.3 stable and staging
Nodemcu v3 LoLin

kiralikbeyin, 2016-07-29:
I reinstall esp8266 2.3 and no problem now. 

I am not sure if it makes trouble again.

Thank you igor.

igrr, 2016-07-29:
No problem! I'm Ivan by the way :)

kiralikbeyin, 2016-07-29:
Nightmare again

Arduino: 1.6.8 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"
 
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

kiralikbeyin, 2016-07-29:
/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include
Make a copy of time.h and name it to "_time.h"

replace in WebHandlerImpl.h #include <time.h> to #include <_time.h>

I will give info when i get problem again...

igrr, 2016-08-01:
Do you have some other file called "Time.h" somewhere in your include paths?

kiralikbeyin, 2016-08-01:
Orginal arduino lib https://github.com/PaulStoffregen/Time and esp8266 core time lib nothing more... 
I didn't get any error with time lib before.

But my duplicate time.h _time.h files solved it for now.

Thanks for your attention.

bhcuong2008, 2016-10-16:
I've just faced error undefined reference to strftime. Arduino 1.6.10, ESP stable 2.3.0, on ESP-12.

In path "AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include", I see file time.h, including function strftime.

Please help me.

Thank you.

kiralikbeyin, 2016-10-16:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/60#issuecomment-253021661

closed 2016-10-16 by kiralikbeyin

jamesmyatt, 2018-12-05:
I have this problem also, on windows only, not linux. I solved it by deleting the "Time.h" from the Time library (https://github.com/PaulStoffregen/Time), and using TimeLib.h only in my code.