esp8266/Arduino #2343: Is there any "WiFi Direct " example available ?

waqas-ogc, 2016-07-29:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

igrr, 2016-08-01:
This isn't available in non-OS SDK at the moment, and therefore not available in Arduino core.

closed 2016-08-01 by igrr

stern0m1, 2017-04-03:
Is this possible with with rtos?
Can you post an example?
Thanks

Androbin, 2019-08-25:
Someone seems to have done it:
https://github.com/youmebangbang/esp8266-wifi-direct-examples

cristiandc, 2020-04-01:
> Someone seems to have done it:
> https://github.com/youmebangbang/esp8266-wifi-direct-examples

That isnt look like a wifi direct protocol. Instead its just a server-client simple communication with some-guy-made protocol