esp8266/Arduino #2352: Is switching between softAP and client connection possible?

treii28, 2016-07-31:
Is it possible to switch back and forth between connecting to an existing network or running a softAP in the arduino code? I'm interested in having a softAP landing page after programming where you can specify a network ssid and password. Once that is stored, have the device try to connect to that network as a client instead. If for any reason that fails, after so many attempts or a given period of time go back to softAP mode and the configuration page again until the information is again changed.

Someone in the ESP8266.com forums said it was possible with the .mode() method, but didn't say anything else so I'm not sure what all needs to be done. (i.e. does the old instance handle need to be re-initialize in any way? Do I have to 'shut down' one before starting the other in some way, etc)

### Basic Infos

#### Hardware
Hardware:		ESP-12 / nodemcu

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU v0.9
CPU Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

krzychb, 2016-07-31:
Hi treii28 ,

I believe the functionality you are looking for is already provided by great [WiFiManager](https://github.com/tzapu/WiFiManager). 

I would check it out before implementing on my own. WiFiManager should provide the best insight into nuances of changing the modes, etc. For instance you may face problems to establish connection if using SDK 1.5.3 (core 2.3.0) as documented [here](https://github.com/esp8266/Arduino/issues/2186#issuecomment-228581052).

If you like to start testing on your own this is also great and below is quick and rough summary.

Mode changes are possible with:
```cpp
WiFi.mode(WIFI_AP)
WiFi.mode(WIFI_STA)
WiFi.mode(WIFI_AP_STA)
WiFi.mode(WIFI_OFF)
```
In between of  `WIFI_STA` and `WIFI_AP_STA` mode changes I am using:
```cpp
WiFi.reconnect() 
```
After switching to `WIFI_STA` or`WIFI_AP_STA`  from other modes, use:
```cpp
WiFi.begin() 
```
or
```cpp
WiFi.config()
```

Extra reference on API and functionality of particular functions is [here](https://links2004.github.io/Arduino/d0/d5f/class_e_s_p8266_wi_fi_s_t_a_class.html), but I am not sure what core version it reflects.

Krzysztof

johnpengwot, 2016-08-04:
In NodeMCU, a simple way is to use the FLASH button(D3)  as a reset( or mode switch ) button, 
you can count the times of pushed or count hold time with digitalRead(D3) ( exmaple http://playground.arduino.cc/Code/HoldButton) to determine the trigger of mode change.

devyte, 2017-10-14:
Answer is documented, and already provided, closing.

closed 2017-10-14 by devyte