esp8266/Arduino #2353: New serial Rx buffer causes wdt on data recieve using bauds over 115200.

sticilface, 2016-07-31:
### Basic Infos

#### Hardware
Hardware:		any esp8266
Core Version:      	since https://github.com/esp8266/Arduino/pull/2239

### Description

Since this change i've not managed to get serial receive to work without a wdt using bauds above 115200.  The application is an Adalight implementation, that takes Serial data and pushes it out over WS2812s.  Reproducing it has been extremely difficult as it has been part of a very large sketch, with lots going on, but i have distilled it down to an example that reproduces it every time using a baud of 1,000,000.  This example does not reproduce it at lower speeds but it is hard to simulate adalight....  on my adalight set up, i get resets at 230400 and above. 

I should stress that prior to this it worked fine.  It also worked fine with the software implementation that was used prior to the hardware only (https://github.com/esp8266/Arduino/commit/e255f25cfd626cb902d8f55fbe006505435ccdad). 

Things I know
1) Occurs without driving WS2812
2) Occurs at 80 and 160Mhz
3) seems to require a few things in loop.. if getFreeHeap() is removed it works. ?wierd
requires OTA in loop. 
4)  Whereas before the data would be passed in 12 chunks, now it is in 3... might be significant..might not be. 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		?80Mhz?
Flash Mode:			qio
Flash Frequency:	80Mhz
Upload Using:		OTA or SERIAL
Reset Method:		na

### Sketch

https://gist.github.com/sticilface/bab7cfeb6096efaed76269b265c4d296

### Debug Messages

me-no-dev, 2016-08-01:
chunk size comes from RX FIFO THRESHOLD value which was set to 128 and I have now lowered to 64. Should maybe even up to 110-ish as that will make the isr be called less often on large packets. When packet is shorter than the full threshold, timeout flag is set after a few bytes time without data.

closed 2016-08-01 by me-no-dev

sticilface, 2016-08-23:
I'm afraid to say that this problem has not entirely gone away.  I'm still getting crashes on what was an entirely stable code base.  Would it be possible to implement a version of Serial where the RX buffer is disabled entirely, giving functionality similar to before with just raw hardware.  

The odd thing is that all my code worked before Serial was changed to hardware only, so it is only the mix of hardware and software serial buffers that seems to be a problem.

danriches, 2017-09-12:
I'm sorry to say that I also get crashes at 115200 baud. My application streams serial data from a serial port, incoming only no reply to serial device, and via a W5100 on Ethernet it works flawlessly. Via WiFi it's still a big issue whereas in the office where the WiFi is better it's nowhere near as frequent but still happens. I had to change the HardwareSerial class to use a larger 4096 byte buffer to get this working at all at this baud rate.

Anyone have any ideas what might be causing this as it's a massive problem for me. I can post parts of the loop in the sketch if that helps?!