esp8266/Arduino #2356: ESP8266mDNS stops working after some time

angelovAlex, 2016-08-01:
I have a project with OTA and mDNS that adds _temp domain. It works for awhile but after some time it stops to be discoverable. I dont know when exactly, but after one day it never worked. I found an easy fix is to run Arduino IDE, as soon as I run it it becomes discoverable with `avahi-daemon`.
Before I run Arduino IDE after one day of running a sketch:
```
# avahi-browse -t --all
+   eth0 IPv6 BedRoom Back                     AirTunes Remote Audio local
+   eth0 IPv6 BedRoom Front                    AirTunes Remote Audio local
+   eth0 IPv4 BedRoom Back                     AirTunes Remote Audio local
+   eth0 IPv4 BedRoom Front                    AirTunes Remote Audio local
+   eth0 IPv6 raspberrypi                                   Remote Disk Management local
+   eth0 IPv4 raspberrypi                                   Remote Disk Management local
+   eth0 IPv6 raspberrypi              Workstation          local
+   eth0 IPv4 raspberrypi               Workstation          local
+   eth0 IPv6 AlexMacbookPro (5581)                         VNC Remote Access    local
+   eth0 IPv4 AlexMacbookPro (5581)                         VNC Remote Access    local
+   eth0 IPv6 mobdev (2) _apple-mobdev2._tcp  local
+   eth0 IPv4 mobdev (2) _apple-mobdev2._tcp  local
```
Just after I ran Arduino IDE my esps becomes discoverable:
```
# avahi-browse -t --all
+   eth0 IPv6 BedRoom Back                     AirTunes Remote Audio local
+   eth0 IPv6 BedRoom Front                    AirTunes Remote Audio local
+   eth0 IPv4 BedRoom Back                     AirTunes Remote Audio local
+   eth0 IPv4 BedRoom Front                    AirTunes Remote Audio local
+   eth0 IPv6 raspberrypi                                   Remote Disk Management local
+   eth0 IPv4 raspberrypi                                   Remote Disk Management local
+   eth0 IPv6 raspberrypi              Workstation          local
+   eth0 IPv4 raspberrypi               Workstation          local
+   eth0 IPv6 AlexMacbookPro (5581)                         VNC Remote Access    local
+   eth0 IPv4 AlexMacbookPro (5581)                         VNC Remote Access    local
+   eth0 IPv6 mobdev (2) _apple-mobdev2._tcp  local
+   eth0 IPv4 mobdev (2) _apple-mobdev2._tcp  local
+   eth0 IPv6 esp2                                          _arduino._tcp        local
+   eth0 IPv6 esp3                                          _arduino._tcp        local
+   eth0 IPv4 esp2                                          _arduino._tcp        local
+   eth0 IPv4 esp3                                          _arduino._tcp        local
+   eth0 IPv6 esp2                                          _temp._tcp           local
+   eth0 IPv6 esp3                                          _temp._tcp           local
+   eth0 IPv4 esp2                                          _temp._tcp           local
+   eth0 IPv4 esp3                                          _temp._tcp           local
```

I tried to run `MDNS.update()` every hour, it didnt help. Also tried to do soft reset of devices, didnt help as well.  I can't run Arduino IDE every time I need to discover devices, so need to find a solution to make it work all the time with `avahi-browse`.

me-no-dev, 2016-08-01:
seems that issue is in your avahi browser because what the IDE does is query for services the same way that avahi should. Run tcpdump on udp port 5353 on the avahi machine and see if requests are sent by avahi-browse. If so, please post the full request here so we can examine it.
You can also enable debug in [ESP8266mDNS.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/ESP8266mDNS.cpp#L62-L64) and see if any messages arrive

angelovAlex, 2016-08-01:
I ran wireshark on windows machine (without running Arduino IDE) and ran `avahi-browse -t -all` and I catched queries and responses between my raspberry pi (the avahi machine) and esp devices. 
The interesting thing that I found is that it sends domain names in query
[https://www.amazon.com/clouddrive/share/beiXwj8OL23ppCklInOSbX7pFQp638t2edCEtTQTvc9?ref_=cd_ph_share_link_copy](url)
So I think Arduino IDE simply does something like `browse -d local _arduino._tcp` and next time I can do `avahi-browse -d local _temp._tcp` and it should work.

angelovAlex, 2016-08-02:
I tested it today and can confirm. You do need to specify the name of domain like `avahi-browse -d local _temp._tcp` if your domain is not standard like `_http`,`_ssh`, etc.

closed 2016-08-02 by angelovAlex