esp8266/Arduino #2357: ERROR - WPS_CB_ST_UNK not declared in scope

bdgordon10, 2016-08-01:
I wrote a sketch that I previously had working on my ESP8266-12E nodeMCU board, which is now giving me a compiling error. Below is the error I am receiving:

/Users/[MYCOMPUTER]/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp: In function 'void wifi_wps_status_cb(wps_cb_status)':
/Users/[MYCOMPUTER]/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:593:14: error: 'WPS_CB_ST_UNK' was not declared in this scope
         case WPS_CB_ST_UNK:
              ^

Any help would be greatly appreciated!

igrr, 2016-08-02:
This new constant is defined in tools/sdk/include/user_interface.h. The only way to get this error would be to update ESP8266WiFi library without updating tools/sdk/include/user_interface.h. Please describe your setup in more detail: how are you obtaining esp8266 core (boards manager, git, or some other way), how are you updating it, which version of the core you have, etc.

bdgordon10, 2016-08-02:
Thanks for the response, Ivan.

I got the ESP8266 core by adding the boards in the board manager with URL: http://arduino.esp8266.com/stable/package_esp8266com_index.json.

I have version 2.3.0 installed.

igrr, 2016-08-02:
Boards manager install should not place ESP8266WiFi library into your Documents/Arduino/libraries directory. Have you placed it there manually?

bdgordon10, 2016-08-02:
Yes, I downloaded the .zip file from this github, extracted, and placed the ESP8266WiFi folder in my arduino/libraries folder.

How should I have done it?

igrr, 2016-08-02:
I think you should follow installation instructions given in README.md file of this repository. There is no mention of downloading zip files and placing them into libraries directory. Just use boards manager to install the complete package.

Like other libraries in this git repository, ESP8266WiFi library is bundled with the core. So there is no need to install library separately. It is installed as part of boards manager package installation.

bdgordon10, 2016-08-02:
Right as you said that, I stumbled upon that file and the directions.

What I did now was uninstall the Arduino IDE and remove the ESP8266 files from the libraries folder, as well as deleted the core files. I then reinstalled 1.6.5 and reinstalled the ESP cores. But now I am receiving an error saying that the ESP8266WiFi.h does not exist. Am I missing a step here?

bdgordon10, 2016-08-02:
...I should also make sure I have the correct board selected!

Now it works! Thanks for your help, I really, truly appreciate it!

closed 2016-08-02 by igrr