esp8266/Arduino #2360: Issue: ESP8266 outputting random data types, all of which are garbage?

trek90, 2016-08-02:
I'm using the tutorial described here: https://learn.adafruit.com/esp8266-temperature-slash-humidity-webserver/uploading-to-the-esp8266

My setup and wiring is identical to their setup, except I'm using an Arduino Uno as a power source, and a DHT11.

However, instead of communicating the right type of data (C0), it seems to be randomly sending wrong data types, such as 47, 56, and FF.

### Sketch

```    /* DHTServer - ESP8266 Webserver with a DHT sensor as an input
     
       Based on ESP8266Webserver, DHTexample, and BlinkWithoutDelay (thank you)
     
       Version 1.0  5/3/2014  Version 1.0   Mike Barela for Adafruit Industries
    */
    #include <ESP8266WiFi.h>
    #include <WiFiClient.h>
    #include <ESP8266WebServer.h>
    #include <DHT.h>
    #define DHTTYPE DHT11
    #define DHTPIN  1
     
    const char* ssid     = "YourRouterID";
    const char* password = "YourRouterPassword";
     
    ESP8266WebServer server(80);
     
    // Initialize DHT sensor 
    // NOTE: For working with a faster than ATmega328p 16 MHz Arduino chip, like an ESP8266,
    // you need to increase the threshold for cycle counts considered a 1 or 0.
    // You can do this by passing a 3rd parameter for this threshold.  It's a bit
    // of fiddling to find the right value, but in general the faster the CPU the
    // higher the value.  The default for a 16mhz AVR is a value of 6.  For an
    // Arduino Due that runs at 84mhz a value of 30 works.
    // This is for the ESP8266 processor on ESP-01 
    DHT dht(DHTPIN, DHTTYPE, 11); // 11 works fine for ESP8266
     
    float humidity, temp_f;  // Values read from sensor
    String webString="";     // String to display
    // Generally, you should use "unsigned long" for variables that hold time
    unsigned long previousMillis = 0;        // will store last temp was read
    const long interval = 2000;              // interval at which to read sensor
     
    void handle_root() {
      server.send(200, "text/plain", "Hello from the weather esp8266, read from /temp or /humidity");
      delay(100);
    }
     
    void setup(void)
    {
      // You can open the Arduino IDE Serial Monitor window to see what the code is doing
      Serial.begin(115200);  // Serial connection from ESP-01 via 3.3v console cable
      dht.begin();           // initialize temperature sensor
     
      // Connect to WiFi network
      WiFi.begin(ssid, password);
      Serial.print("\n\r \n\rWorking to connect");
     
      // Wait for connection
      while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
      }
      Serial.println("");
      Serial.println("DHT Weather Reading Server");
      Serial.print("Connected to ");
      Serial.println(ssid);
      Serial.print("IP address: ");
      Serial.println(WiFi.localIP());
       
      server.on("/", handle_root);
      
      server.on("/temp", [](){  // if you add this subdirectory to your webserver call, you get text below :)
        gettemperature();       // read sensor
        webString="Temperature: "+String((int)temp_f)+" F";   // Arduino has a hard time with float to string
        server.send(200, "text/plain", webString);            // send to someones browser when asked
      });
     
      server.on("/humidity", [](){  // if you add this subdirectory to your webserver call, you get text below :)
        gettemperature();           // read sensor
        webString="Humidity: "+String((int)humidity)+"%";
        server.send(200, "text/plain", webString);               // send to someones browser when asked
      });
      
      server.begin();
      Serial.println("HTTP server started");
    }
     
    void loop(void)
    {
      server.handleClient();
    } 
     
    void gettemperature() {
      // Wait at least 2 seconds seconds between measurements.
      // if the difference between the current time and last time you read
      // the sensor is bigger than the interval you set, read the sensor
      // Works better than delay for things happening elsewhere also
      unsigned long currentMillis = millis();
     
      if(currentMillis - previousMillis >= interval) {
        // save the last time you read the sensor 
        previousMillis = currentMillis;   
     
        // Reading temperature for humidity takes about 250 milliseconds!
        // Sensor readings may also be up to 2 seconds 'old' (it's a very slow sensor)
        humidity = dht.readHumidity();          // Read humidity (percent)
        temp_f = dht.readTemperature(true);     // Read temperature as Fahrenheit
        // Check if any reads failed and exit early (to try again).
        if (isnan(humidity) || isnan(temp_f)) {
          Serial.println("Failed to read from DHT sensor!");
          return;
        }
      }
    }
```

### Debug Messages

```
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 57600
	setting port from COM1 to COM1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 40 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 54 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 47 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 47 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: FF instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 56 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 47 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 47 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: FF instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

```

RudyFiero, 2016-08-03:
What are you using to supply the 3.3V to the ESP module? If the UNO then this might be the problem as it does not have enough output capability.

trek90, 2016-08-03:
The power source is in fact the UNO, I'll try a battery pack and report back.

trek90, 2016-08-03:
Interesting...when I switched to a battery pack I still got errors, but they were of a completely different variety (data type that is):
`esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 57600
	setting port from COM1 to COM1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: F0 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 60 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 5C instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 67 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 48 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 48 instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 1B instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: A6 instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 5C instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed`

trek90, 2016-08-03:
Update: I switched to a Huzzah, and now I'm receiving this error:
```
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 2E instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

RudyFiero, 2016-08-03:
#define DHTPIN  1

This is probably your problem. The default TX is on D1. Move the sensor to a different pin.

trek90, 2016-08-05:
So what should I change it to?

RudyFiero, 2016-08-05:
What do you have open. And open is the important word. Different functions sharing the same pin causes problems.

devyte, 2017-10-14:
This looks like user error. In any case, not the right place to discuss. Closing per #3655 .

closed 2017-10-14 by devyte