esp8266/Arduino #2362: Stack dump from dns_check_entry

lonerzzz, 2016-08-03:
### Basic Info

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description

I am routinely, but not always, getting a core dump with DNS lookup that is happening in the background processing when I attempt to resolve the DNS address:
0x40231e78: dns_check_entry at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 690
0x4023202c: dns_enqueue at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 917
:  (inlined by) dns_gethostbyname at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/dns.c line 977

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

Cannot include

### Debug Messages

Only the stack trace

devyte, 2017-10-14:
@lonerzzz Without a sketch to look through, this could be anything.
Is the issue still present in
- latest git
- PR #3215 
- PR #3362 
?

devyte, 2018-02-28:
Closing due to lack of feedback and age.

closed 2018-02-28 by devyte