esp8266/Arduino #2363: Question: Problem uploading

Gorkde, 2016-08-03:
ESP 12F and E

I had a FTDI adapter as well as some electronics go defective because of a new power supply that was defective.

I bought a new FTDI adapter but now sometimes it uploads and then my working sketches cause WDT Resets as if they haven't been uploaded correctly.

But I most of the time get this when uploading:

```
Uploading 248208 bytes from C:\............bin to flash at 0x00000000
................................................................................ [ 32% ]
................................................................................ [ 65% ]
................................................................................ [ 98% ]
..warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed

```

For some reason randomly (seldom) it uploads flawlessly and the sketch works.

I tried using a extra breadboard with only microcontroller and voltage regulator. Also I tried multiple ESP Modules. All the same response.

Any idea what might be the reason?

RudyFiero, 2016-08-07:
With known working sketches I would look at hardware issues when it is operating erratically.  The power supply is an area that is under-appreciated by a lot of people. Poor regulation can cause a lot of problems. 

You didn't specify the regulator so I'll say that you should have one that is rated for at least 500mA continuous current. You also must have capacitors from Vin on the regulator to ground/common. Also a capacitor(s) on the output of the regulator to ground. Minimum values should be 10uF. I tend to use between 50-100uF on the output of the regulator. I include a 0.1uF capacitor in parallel with the larger caps. Keep wiring short.

Make sure that you test with a good ESP module. If the faulty power supply damaged all you ESPs then you might be seeing the result of that damage. (possible but I would consider that unlikely)

I have been having less than perfect loads when using OTA code. I go back to wired serial and sometimes it still doesn't work. I do a reset and still no go. Re-powering the module seems to fix the problem. I suspect that to be a code issue but I haven't spent enough time to try and diagnose the problem. So hard to say. Too many other things to do.

devyte, 2017-10-14:
@Gorkde previous comment is correct. This is most likely a power supply issue, or wiring issue.
Not the right place to discuss this. Closing per #3655

closed 2017-10-14 by devyte