esp8266/Arduino #2364: Undefined Reference to String.h Functions

EmmaRose, 2016-08-03:
I'm trying to run a simple sketch that works on other boards (Feather w/ WINC1500, Arduino MKR, etc.) and it won't compile because of undefined references to strncasecmp and strspn in an outside library I'm using. I've seen people have similar issues with different functions, but haven't found a fix yet. 

Hardware:               Adafruit Feather HUZZAH w/ ESP8266 WiFi

Module:  			Adafruit HUZZAH ESP8266
Flash Size: 		4M
CPU Frequency:	80Mhz
Upload Speed:        115200
Programmer:           AVR ISP

Error Message: 
libraries/PubNub/PubNub.cpp.o: In function `PubNub::PubNub()':
/Users/emma/Documents/Arduino/libraries/PubNub/PubNub.cpp:201: undefined reference to `strncasecmp'
libraries/PubNub/PubNub.cpp.o: In function `PubNub::_request_bh(WiFiClient&, unsigned long, int, char)':
/Users/emma/Documents/Arduino/libraries/PubNub/PubNub.cpp:201: undefined reference to `strncasecmp'
libraries/PubNub/PubNub.cpp.o:(.text._ZN6PubNub7publishEPKcS1_i+0x18): undefined reference to `strspn'
libraries/PubNub/PubNub.cpp.o: In function `PubNub::publish(char const*, char const*, int)':
/Users/emma/Documents/Arduino/libraries/PubNub/PubNub.cpp:201: undefined reference to `strspn'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.

a-andreyev, 2016-08-16:
Looks similar to #1480: make sure about libraries including order.

I'm also not sure that PubNub library you are using is 100% compatible with the ESP8266WiFi API (probably it was created for some arduino wifi shield): [WiFi(ESP8266WiFi library)](https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#wifiesp8266wifi-library)

Searched for esp8266 arduino libraries at github, take a look at: https://github.com/suryasundarraj/pubnub-esp8266-rest-api

Anyway, I have no experience with the PubNub

devyte, 2017-10-14:
@EmmaRose I don't see a build problem using strncasecmp() in latest git.
Closing as can't reproduce.

closed 2017-10-14 by devyte