esp8266/Arduino #2365: SoftwareSerial problem

skeezoo, 2016-08-04:
Hi, 
I working with a wemos d1 board that s trying to communicate with a PZEM-004T
my IDE is 1.6.8 and esp8266 library 2.2.3
I faced problems with Serial communication when receiving bytes . it just looked like rubbish.
tried a wemos d1 with a mega 2560 always the esp8266 receives wrong characters.
Double and trippled check all my configurations , but no difference.
Until I tried to use older version of SoftwareSerial. I replaced the SoftwareSerial library found in version 2.2.3 with the one in 2.0.0
and now everything works 
when using any other different than 2.0.0 same issue 
I even tried to used the latest IDE 1.6.10 but same issue.
anyone having problem with software serial library? I'm I missing some configuration?

9600, 2016-08-04:
You called?

Sandiep, 2016-08-04:
Hi @skeezoo,

I have tested [espsoftwareserial](https://github.com/plerup/espsoftwareserial.git) v3.1.0 with esp8266 v2.3.0 using Arduino IDE 1.6.10.
Following code seems to work fine.
`#include <SoftwareSerial.h>

SoftwareSerial soft_serial(5,4);
void setup() {
  soft_serial.begin(9600);
}

void loop() {
  soft_serial.println("Hello World");
  delay(1000);
}`

Can you share your code to regenerate issue?

Thanks.

skeezoo, 2016-08-05:
@9600 sorry wrong number ;-)

@Sandiep thx for your help but found the solution in issue #1426
seems that @juancgalvez was working on a PZEM and solved it by adding  half bit time delay just before starting to read bits
followed his recommendation and solved the problem
looks like a PZEM timing issue.
Thanks a lot juancgalvez

crosofg, 2017-04-28:
@skeezoo can you share the pzem software serial with esp8266 code?

skeezoo, 2017-06-16:
Hi @crosofg
sorry for late answer
here is a sample code 

////////////////////very important for serial communication with PZM///////////////////////////////////////////////
#include "SoftwareSerial.h" // used  espsoftwareserial-master -> change in void ICACHE_RAM_ATTR SoftwareSerial::rxRead()-> m_bitTime/3  to  m_bitTime/2 
SoftwareSerial PZEMSerial(14, 12, false, 256);
uint8_t PZEM004T_Address[7]= {0xB4,0xC0,0xA8,0x01,0x01,0x00,0x1E};
uint8_t PZEM004T_VoltRequest[7]={0xB0,0xC0,0xA8,0x01,0x01,0x00,0x1A};
uint8_t PZEM004T_AmpRequest[7]={0xB1,0xC0,0xA8,0x01,0x01,0x00,0x1B};// C0 A8 01 01 00 1B
uint8_t PZEM004T_WattRequest[7]={0xB2,0xC0,0xA8,0x01,0x01,0x00,0x1C};//B2 C0 A8 01 01 00 1C
uint8_t PZEM004T_WattHourRequest[7]={0xB3,0xC0,0xA8,0x01,0x01,0x00,0x1D};//B3 C0 A8 01 01 00 1D

byte byPZEM004TResp,byPZEM004TReq;

float rVoltMain,rAmpsMain;
char strVolt[5] = "0";
char strAmp[5] = "0";
char strWatt[5] = "0";
char strWattHour[10] =  "0";
byte rx_bytes[7];
bool bPZEMTxBusy,bPZEMRxTimeout,bDataAvailable,bPing;
unsigned long prevmillis,prevmillis1,prevmillis2,prevmillis3,prevmillis4,prevmillis5,prevmillis6;
int i;
uint32_t dnWattHour;
uint16_t nWatt;
char buf[100];

///////////////////////////////////////////////////////////////////////////////////////////////////////////////
//SETUP
///////////////////////////////////////////////////////////////////////////////////////////////////////////////
void setup() {
  Serial.begin(115200);
}

void loop() {
 ///////////////////////////////////////////////////////////////////////////////////
  //PZEM Communication Fault 
  ///////////////////////////////////////////////////////////////////////////////////
  if ((millis() - prevmillis > 5000) && (bPZEMTxBusy)){
    //detect and trigger fault , start req again
    bPZEMTxBusy = false;
    bPZEMRxTimeout = true;
    //Reset Values read
    rVoltMain = 0.0;
    rAmpsMain = 0.0;
    nWatt = 0;
    byPZEM004TReq = 0;
    Serial.println("PZEM RX timeout");
  }
 
  ///////////////////////////////////////////////////////////////////////////////////
  //Led Fast Flicker On PZEM RX timeout
  ///////////////////////////////////////////////////////////////////////////////////
  if ((millis() - prevmillis1 > 50) && (bPZEMRxTimeout)){
    //Toggle BuildIn Led
    digitalWrite(BUILTIN_LED, !digitalRead(BUILTIN_LED));
    prevmillis1 = millis();
  }
  
  ///////////////////////////////////////////////////////////////////////////////////
  // Send To PZEM Request
  ///////////////////////////////////////////////////////////////////////////////////
  if ((millis() - prevmillis > 100) && (!bPZEMTxBusy)) {
    bPZEMTxBusy = true;
    switch (byPZEM004TReq) {
      case 0:
          PZEMSerial.write(PZEM004T_Address,sizeof(PZEM004T_Address));
          //Serial.print("Addressreq @ ");
          //Serial.println(millis());
[... 75 lines cut ...]
        byPZEM004TReq = 4;
        break;
      case 0xA3:
        Serial.print("Wh = ");
        dnWattHour = ((uint32_t)rx_bytes[1]<<16) + ((uint32_t)rx_bytes[2]<<8) + (uint32_t)rx_bytes[3];
        memset(strWattHour, 0, sizeof(strWattHour));
        ltoa(dnWattHour,strWattHour,10);
        Serial.println(strWattHour); 
        Serial.println(""); 
        byPZEM004TReq = 1;
        break;
      case 0xA4:
        if (rx_bytes[6]=0xA4){
          Serial.print("Address Ok ");
          byPZEM004TReq = 1;
        }
        else Serial.print("Address not Defined ");
    }
    
    
  }
  
  
  
}//End of Loop

devyte, 2017-10-14:
Per previous comments, closing as resolved.

closed 2017-10-14 by devyte