esp8266/Arduino #2366: #include <avr/pgmspace.h>

GeorgeBgk, 2016-08-04:
I am trying to use nokia 5110 screen with my ESP and when I compile my code i receive this:

C:\Users\g\Documents\Arduino\libraries\Adafruit-PCD8544-Nokia-5110-LCD-library-master\Adafruit_PCD8544.cpp:20:26: fatal error: avr/pgmspace.h: No such file or directory

 #include <avr/pgmspace.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0-rc2

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	?
Upload Using:		SERIAL
Reset Method:		?

### Sketch
/*********************************************************************
This is an example sketch for our Monochrome Nokia 5110 LCD Displays

  Pick one up today in the adafruit shop!
  ------> http://www.adafruit.com/products/338

These displays use SPI to communicate, 4 or 5 pins are required to
interface

Adafruit invests time and resources providing this open source code,
please support Adafruit and open-source hardware by purchasing
products from Adafruit!

Written by Limor Fried/Ladyada  for Adafruit Industries.
BSD license, check license.txt for more information
All text above, and the splash screen must be included in any redistribution
*********************************************************************/

#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_PCD8544.h>

// Software SPI (slower updates, more flexible pin options):
// pin 7 - Serial clock out (SCLK)
// pin 6 - Serial data out (DIN)
// pin 5 - Data/Command select (D/C)
// pin 4 - LCD chip select (CS)
// pin 3 - LCD reset (RST)
Adafruit_PCD8544 display = Adafruit_PCD8544(7, 6, 5, 4, 3);

// Hardware SPI (faster, but must use certain hardware pins):
// SCK is LCD serial clock (SCLK) - this is pin 13 on Arduino Uno
// MOSI is LCD DIN - this is pin 11 on an Arduino Uno
// pin 5 - Data/Command select (D/C)
[... 292 lines cut ...]
    display.drawLine(display.width()-1, display.height()-1, i, 0, BLACK);
    display.display();
  }
  for (int16_t i=display.height()-1; i>=0; i-=4) {
    display.drawLine(display.width()-1, display.height()-1, 0, i, BLACK);
    display.display();
  }
  delay(250);

  display.clearDisplay();
  for (int16_t i=0; i<display.height(); i+=4) {
    display.drawLine(display.width()-1, 0, 0, i, BLACK);
    display.display();
  }
  for (int16_t i=0; i<display.width(); i+=4) {
    display.drawLine(display.width()-1, 0, i, display.height()-1, BLACK); 
    display.display();
  }
  delay(250);
}
### Debug Messages

```
messages here
```

reaper7, 2016-08-04:
pgmspace.h for esp8266 is located inside cores root direcory,
so You need to little mod Adafruit_PCD8544.cpp file
and replace #include <avr/pgmspace.h> with:

```
#if defined ( ESP8266 )
  #include <pgmspace.h>
#else
  #include <avr/pgmspace.h>
#endif

btw...maybe we must create avr directory for future esp8266 arduino release and move avr specific files to this dir?

alpeza, 2017-05-19:
no funca una mierda

devyte, 2017-10-14:
@GeorgeBgk you are requesting help for a 3rd party lib, which seems to not support the ESP.
Per last comment, feel free to tweak the lib, or request support from the author.
Closing per #3655 .

closed 2017-10-14 by devyte

aaronchih, 2018-06-30:
I also have same issue. 
This is reper7's solution for it. I modified in "***.h" file.
#if defined ( ESP8266 )
  #include <pgmspace.h>
#else
  #include <avr/pgmspace.h>
#endif
working well.

d-a-v, 2018-06-30:
Fwiw, since at least core 2.4.1 and latest git master, <avr/pgmspace.h> should work.