esp8266/Arduino #2367: wpa2 Enterprise (eap_tls) problems, how to understand wifi debug code ??

teji24, 2016-08-04:
Hello,
i use arduino 1.6.9 with 2.3.0 esp add-on...
Work fine, thanks for theso good job ! 
i need to connect to eap-tls network. Then i begin to test with a home labo test.
with help of "eduram" exemple, i can connect to wpa enterprise eap-tls network, with certificate.
GOOD !
But at customer site, it don't want to connect.
i received a pfx certificate, has extracted cer&key to use in "eduram" example, without success... :-(
i used "Serial.setDebugOutput(true);" to obtain more infos.
indeed, i received code like: 
---------------cut cut cut--------------------
set cert key
scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
..........pm open,type:2 0
....................state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
.......
---------------cut cut cut--------------------
some text like "." or "set cert key" are sent by myself.
but HOW TO UNDERSTAND the debug informations ?
i would like to understant why it don't wanna connect on customer site, but accept to work in my "eap_tls testing labo" .....
but no doc about wifi debugging message anywhere.
Then before asking help about eap-tls itself, i prefer to try to understand wifi debug messages....
cause i was trying different kind of certificate, and got different debug codes.
Then understand can probably help me...
Thanks a lot for your answers ! Anyone can understand thoses codes ??? 
(anyway, if someone can talk about similar experiences....) 
TéJi

teji24, 2016-08-05:
WPA2 enterprise connection @ customer site is now OK, fault was "identity not set" in my code.
BUT the explaination about wifi debug code remain !!! 
Is someone understand thoses codes ? where is the doc ??
Thanks !

gorghino, 2016-10-22:
Did you manage a WPA2-PEAP connection? How did you do it? Could you please share your sketch and configuration?

teji24, 2016-10-25:
Hello. Not peap. I use wpa2 eap.
But it slow.....

shmrymbd, 2017-08-17:
hi to connect to  WPA2 Enterprise + MsChapv2.. anyone can advice?

devyte, 2017-10-14:
Per next to last of OP's comments, the underlying connection issue is solved. Also, in #3215 (sdk 2.1.0), the new sdk now has an api to set the outer identity.
Closing as resolved.

closed 2017-10-14 by devyte