esp8266/Arduino #2369: portmoderegister problems

tablatronix, 2016-08-06:
This is a dup but the other title is not relevant to actual issue
#1914 

I have a problem setting gpio mode using
`portModeRegister`

tablatronix, 2017-01-30:
```
port    = digitalPinToPort(PIN);    // get the pin's port
bitmask = digitalPinToBitMask(PIN); // pin bitmask
ddr     = portModeRegister(port);   // get the ddr 
*ddr |= bitmask;  // does not work
```

Not sure how this is supposed to toggle the correct bits when `&GPE` and `pinmask` do not align for pins above #3

Maybe someone with better understanding of the registers can look at this.

devyte, 2017-10-14:
@tablatronix I'm not sure what can be done here, if the register bits don't align. What is your expectation?

igrr, 2017-10-14:
Sorry, what do you mean by "GPE and pinmask do not align?" What is "pinmask"?

tablatronix, 2017-10-14:
I think I meant GPC(p) or bitmask ( I had some tests for this, but who knows where i put them )
I noticed pinmode uses alot of voodoo, like gpiomode and GPC for setting pinmode

EIther way I had problems with the avr compat mapping, and using the above code and simple
GPE - bitmask masking alignment

Was found using code similar to the linked issue above

```
#define GPC(p) ESP8266_REG(0x328 + ((p & 0xF) * 4))
```

```
      GPF(pin) = GPFFS(GPFFS_GPIO(pin));//Set mode to GPIO
      GPC(pin) = (GPC(pin) & (0xF << GPCI)); //SOURCE(GPIO) | DRIVER(NORMAL) | INT_TYPE(UNCHANGED) | WAKEUP_ENABLE(DISABLED)
      if(mode == OUTPUT_OPEN_DRAIN) GPC(pin) |= (1 << GPCD);
      GPES = (1 << pin); //Enable
    } else if(mode == INPUT || mode == INPUT_PULLUP){
      GPF(pin) = GPFFS(GPFFS_GPIO(pin));//Set mode to GPIO
      GPEC = (1 << pin); //Disable
      GPC(pin) = (GPC(pin) & (0xF << GPCI)) | (1 << GPCD); //SOURCE(GPIO) | DRIVER(OPEN_DRAIN) | INT_TYPE(UNCHANGED) | WAKEUP_ENABLE(DISABLED)
      if(mode == INPUT_PULLUP) {
          GPF(pin) |= (1 << GPFPU);  // Enable  Pullup
      }
```

```
#define digitalPinToBitMask(pin)    (1UL << (pin))
#define GPE    ESP8266_REG(0x30C) //GPIO_ENABLE R/W (Enable)
#define portModeRegister(port)      ((volatile uint32_t*) &GPE)
```

earlephilhower, 2019-01-16:
The problem is that the macro used is defined on AVR to map to a specific HW register with specific bit configs.  The 8266 is not compatible in any way, shape, or form with the AVR, so you will need to use the appropriate high-level `pinMode()` calls.

closed 2019-01-16 by earlephilhower

dagnall53, 2019-01-25:
@earlephilhower 
Sorry to ask in a closed issue: (only just found this)
I am trying to track down some very strange behavior with PWM that I suspect is due to unexpected pinMode changes after EEPROM commit, so was wanting to see EXACTLY what each pin is set to.
I can see that the bitmask produces an apparently sensible bit mask but is there ANY way to access the ESP low level port direction registers (or equivalent). 

Many thanks.. I will keep trying alternate bug finding techniques, but directly looking at the actual registers would simplify the process!.

dagnall53, 2019-01-26:
I think you mean 'core_esp8266_wiring_digital.c',but yes, WOW. that is truly convoluted!.