esp8266/Arduino #2370: After updating the flash size to match the flash script constant crashing

AdamMiltonBarker, 2016-08-06:
### Basic Infos

#### Hardware
Hardware:			ESP-12f
Core Version:      	2.3.0

### Description

After updating the flash size to 4m (1m spiffs) and also trying 3m spiffs, I am getting constant crashing, this was all to try and fix an error with HTTP update failing to connect to a secure server. This is my last attempt at trying to get these issues solved. I would appreciate if anyone can help get to the bottom of this before we finally decide to move to other hardware.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/3MB?
CPU Frequency:		80Mhz?
Flash Mode:			DIO
Flash Frequency:	40Mhz?
Upload Using:		SERIAL?
Reset Method:		ck 

### Sketch

```/*
 Copyright (c) 2016 TechBubble Technologies and other Contributors.
 
 Contributors:
    Adam Milton-Barker  - Initial Contribution
*/

#include <FS.h> 
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h> 
#include <ArduinoJson.h>
#include <ESP8266mDNS.h>
#include <DHT.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
 
const int powerPin = 4;
const int sensorPin = 5;
DHT dht(sensorPin, DHT11,15);
float humidity, temperature;
unsigned long previousMillis = 0;   
const long interval = 2000; 

ESP8266WebServer *server;

WiFiClientSecure espClient;
PubSubClient client(espClient);

extern "C" {
  #include "user_interface.h"
}

ADC_MODE(ADC_VCC);

const char* ssid = "TechBubbleAP";
const char* passphrase = "thereisnospoon";
String st;
String content;
String serverType ="";

String esid = "";
String epass = "";
  
String locationID = "";
String zoneID = "";
[... 418 lines cut ...]
Decoding 25 results
0x401003e9: _umm_malloc at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1382
0x40203490: spiffs_obj_lu_scan_v at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_nucleus.c line 941
0x401004f4: calloc at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1684
0x40107348: pvPortZalloc at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/heap.c line 33
0x40229a71: wifi_param_save_protect_with_check at ?? line ?
0x40105dbe: spi_flash_read at ?? line ?
0x40229b59: system_param_save_with_protect at ?? line ?
0x40229f48: wifi_station_ap_number_set at ?? line ?
0x4022a0cf: wifi_station_ap_number_set at ?? line ?
0x4022a15c: wifi_station_ap_number_set at ?? line ?
0x4022a198: wifi_station_set_config at ?? line ?
0x40208eac: ESP8266WiFiSTAClass::disconnect(bool) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 510
0x4010053d: _umm_realloc at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4020f6ac: String::changeBuffer(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4020f6fb: String::reserve(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4020f72d: String::copy(char const*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x402297c6: wifi_get_opmode at ?? line ?
0x40208c4f: ESP8266WiFiGenericClass::mode(WiFiMode) at c:\users\usuario\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1954
0x4020a6b0: HTTPUpload::HTTPUpload() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40207325: initiateAccessPoint() at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\ThermaSenseDSB/ThermaSenseDSB.ino line 265
0x40207a20: setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\ThermaSenseDSB/ThermaSenseDSB.ino line 338
0x402102d8: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

AdamMiltonBarker, 2016-08-07:
After changing the settings back to 512k and 128 k spiffs the above no longer happens but the flash check script tells me these are incorrect settings for my 12f.

igrr, 2016-08-08:
Loaded the sketch into NodeMCU (ESP-12F), set flash size to 4M(1M SPIFFS).
Haven't been able to reproduce the crash, here's the output:
```
0�~?�4�����_SAH��System Initiating...
Mounted file system!
 networks found
SoftAP initiated
Server started
{
  "esid": "#!/bin/test",
  "epass": "test_bin_9060316",
  "locationID": "1",
  "zoneID": "1",
  "deviceID": "21",
  "deviceName": "Foo",
  "mqttUsername": "Aaa",
  "mqttPassword": "Bbb"
} �n?�$�����_SAX��System Initiating...
Mounted file system!
Waiting for Wifi to connect...
..............Successfully connected to WiFi
HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused
CURRENT BATTERY: 3.05
Failed to read from DHT sensor!
```

AdamMiltonBarker, 2016-08-08:
Yes and others have but in my case, after changing the flash size it is crashing so how do I fix it ?

AdamMiltonBarker, 2016-08-08:
if I change the flash back no more crashes

igrr, 2016-08-08:
Could you please try running the following sketch on your ESP-12F

```c++
#include <ESP8266WiFi.h>
extern "C" {
  #include "spi_flash.h"
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF);
  delay(3000);
  Serial.println();
  Serial.println("------ cut here ------");
  for (size_t size=512*1024; size<=32*1024*1024; size*=2) {
    flashchip->chip_size = size + 64*1024;
    uint32_t val;
    ESP.flashRead(size, &val, 4);
    Serial.printf("%5d %08x\n", size / 1024, val);
  }
  Serial.println("------ cut here ------");
}

void loop() {}
```

I'm interested in the output, it should look roughly as follows:
```
------ cut here ------
  512 3350020c
 1024 ffffffff
 2048 ffffffff
 4096 0f0201e9
 8192 0f0201e9
16384 0f0201e9
32768 0f0201e9
------ cut here ------
```

AdamMiltonBarker, 2016-08-08:
Thanks with it set on 4m yes ? 3m spiffs or 1m spiffs ?

AdamMiltonBarker, 2016-08-08:
@igrr after lots of garbled print out which included "illegal operation"

`usl
mode : null

------ cut here ------
  512 ffffffff
 1024 ffffffff
 2048 ffffffff
 4096 400201e9
 8192 400201e9
16384 400201e9
32768 400201e9
------ cut here ------
`

igrr, 2016-08-09:
Okay, this means that flash chip size is indeed 4 megabytes, and CheckFlashConfig tool was not mistaken. This means that we have to investigate why doesn't full flash chip erase help in this situation. It could be that WiFi config gets corrupted again after flash erase, but the way this happens is not obvious.
Could you please do the following steps:
- erase the flash chip using esptool.py
- open serial monitor in Arduino IDE, set baud rate to 74880 baud and reset the chip. Verify that a message ending with `ets_main.c` is printed. This means that flash erase was successful and there is no valid program in flash memory.
- Change Serial.begin line in your sketch to 115200 baud, and add `Serial.setDebugOutput(true);` line after it.
- Connect to the AP, enter connection parameters, reset the chip as usual. Keep collecting serial output all the time.
- Assuming that the issue persists, on the next bootup the program should crash.
- Use `esptool.py --port <your_serial_port> read_flash 0 0x400000 flash_dump.bin` command to read the contents of entire flash memory into flash_dump.bin file. This will take a while, perhaps five minutes or so.
- Upload this bin file somewhere and share a link to it.
- Upload all the serial output collected in serial monitor, up to the point of crash, and share a link to that as well.
Thanks.

AdamMiltonBarker, 2016-08-09:
Thank you will try this now, by the way I added the cipher suites and still got A+ (thanks for letting me know those were required :+1:  ) doing the above now won't be long.

AdamMiltonBarker, 2016-08-09:
`'''
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

ets_main.c 

''`

Flash erase confirmed above.

It has just gone absolutely crazy without letting me connect to the AP, do you want me to read the flash now as it is not even getting to AP, one thing I wondered what pullups/pulldowns are required for a 12f

AdamMiltonBarker, 2016-08-09:
Its just loaded into AP now one sec

igrr, 2016-08-09:
Please expand on "gone absolutely crazy". Without the serial output log this isn't very useful to help you troubleshoot the issue.

AdamMiltonBarker, 2016-08-09:
Stack:

`'' 

Decoding 45 results
0x40105d74: spi_flash_write at ?? line ?
0x402104ec: __yield at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x4020e7a2: EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Esp.cpp line 512
0x40211227: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 116
0x40105859: ets_timer_disarm at ?? line ?
0x401018fe: ppEnqueueRxq at ?? line ?
0x401016cb: ppProcessTxQ at ?? line ?
0x40101d6a: pp_post at ?? line ?
0x4010502c: lmacTxFrame at ?? line ?
0x4010423b: lmacRecycleMPDU at ?? line ?
0x4010469e: lmacRecycleMPDU at ?? line ?
0x40104182: lmacProcessTxSuccess at ?? line ?
0x40103fb1: lmacProcessTXStartData at ?? line ?
0x401030e7: wDev_ProcessFiq at ?? line ?
0x40103fb4: lmacProcessTXStartData at ?? line ?
0x40103fb1: lmacProcessTXStartData at ?? line ?
0x40105859: ets_timer_disarm at ?? line ?
0x40102f8e: wDev_ProcessFiq at ?? line ?
0x4021116c: spiffs_hal_read(unsigned int, unsigned int, unsigned char*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 67
0x40102de4: wDev_ProcessFiq at ?? line ?
0x402137ff: pp_soft_wdt_feed at ?? line ?
0x40105d1f: spi_flash_erase_sector at ?? line ?
0x40203490: spiffs_obj_lu_scan_v at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_nucleus.c line 941
0x40203bc8: spiffs_obj_lu_scan at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_nucleus.c line 941
0x40205cac: spiffs_phys_wr at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_cache.c line 204
0x4020e768: EspClass::flashEraseSector(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Esp.cpp line 508
0x4021052c: optimistic_yield at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402039e0: spiffs_erase_block at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_nucleus.c line 941
0x40210429: loop_task at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402111b4: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 102
0x402112f8: spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 168
0x40202550: SPIFFS_format at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_hydrogen.c line 980
0x4021101a: SPIFFSImpl::format() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_api.h line 158
0x402119cd: fs::FS::format() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/FS.cpp line 188
0x402079f1: setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\ThermaSenseDS\ThermaSenseDS/ThermaSenseDS.ino line 333
0x402104b8: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

''`

It won't let me paste the garble

'''
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
áSystem Initiating...

Soft WDT reset

ctx: cont 
sp: 3fff2090 end: 3fff2380 offset: 01b0
'''

'''

Decoding 18 results
0x40105d2a: spi_flash_erase_sector at ?? line ?
0x40205cac: spiffs_phys_wr at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_cache.c line 204
0x4020e754: EspClass::flashEraseSector(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Esp.cpp line 505
[... 10 lines cut ...]
0x402079f1: setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\ThermaSenseDS\ThermaSenseDS/ThermaSenseDS.ino line 333
0x402104b8: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

'''

'''
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
áSystem Initiating...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset

'''
 

Over and over again. Before it was throwing pages and pages of exceptions.

AdamMiltonBarker, 2016-08-09:
Ahh  one sec spiffs format is still on.

AdamMiltonBarker, 2016-08-09:
I have deleted spiffs.format() but after reflashing it still shows it in the stack:

```
Decoding 22 results
0x40105d2a: spi_flash_erase_sector at ?? line ?
0x40205cac: spiffs_phys_wr at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_cache.c line 204
0x4020e750: EspClass::flashEraseSector(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Esp.cpp line 505
0x4020e759: EspClass::flashEraseSector(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Esp.cpp line 505
0x40210528: optimistic_yield at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40211323: spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 178
0x4020398b: spiffs_erase_block at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_nucleus.c line 941
0x40210429: loop_task at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x402111b0: spiffs_hal_write(unsigned int, unsigned int, unsigned char*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 102
0x402112f4: spiffs_hal_erase(unsigned int, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_hal.cpp line 168
0x40202550: SPIFFS_format at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\spiffs/spiffs_hydrogen.c line 980
0x40210fd0: SPIFFSImpl::begin() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/spiffs_api.h line 128
0x4010570e: wdt_feed at ?? line ?
0x402119ad: fs::FS::begin() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/FS.cpp line 188
0x401057ca: ets_timer_setfn at ?? line ?
0x4020776d: readSpiffs() at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\ThermaSenseDS\ThermaSenseDS/ThermaSenseDS.ino line 114
0x4020eca9: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 188
0x4020eda4: Print::println() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 188
0x40207a06: String::operator!=(char const*) const at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.h line 190
:  (inlined by) setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\ThermaSenseDS\ThermaSenseDS/ThermaSenseDS.ino line 338
0x402104b4: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
  ```

```

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

AdamMiltonBarker, 2016-08-09:
```

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
áSystem Initiating...
Fatal exception 29(StoreProhibitedCause):
epc1=0x40203a4a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00008056, depc=0x00000000

Exception (29):
epc1=0x40203a4a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00008056 depc=0x00000000

ctx: cont 
sp: 3fff20a0 end: 3fff2380 offset: 01a0

>>>stack>>>
3fff2240:  00000000 00000100 3fff3b80 40203a30  
3fff2250:  40210429 402111b0 402112f4 000fb000  
3fff2260:  00300000 00002000 00002000 00000100  
3fff2270:  00000200 00000fb0 feefeffe 3fff1350  
3fff2280:  00000000 00000000 3fff3b80 40202550  
3fff2290:  3fffdad0 3fff3b80 3fff3b7c 40210fd0  
3fff22a0:  4010570e 00000000 3fff23a4 402119ad  
3fff22b0:  401057ca 3fff07c8 000129ea 4020776d  
3fff22c0:  3fffdad0 00000014 3fff1294 4020eca9  
3fff22d0:  3ffe8fe0 3fff11cc 3fff1360 3fff11cc  
3fff22e0:  3fffdad0 00000013 3fff1294 4020eda4  
3fff22f0:  3fffdad0 00000000 3fff1294 3fff1350  
3fff2300:  3fffdad0 00000000 3fff1294 40207a06  
3fff2310:  feefeffe feefeffe feefeffe feefeffe  
3fff2320:  feefeffe feefeffe feefeffe feefeffe  
3fff2330:  feefeffe feefeffe feefeffe feefeffe  
3fff2340:  feefeffe feefeffe feefeffe feefeffe  
3fff2350:  feefeffe feefeffe feefeffe 3fff1350  
3fff2360:  3fffdad0 00000000 3fff1348 402104b4  
3fff2370:  feefeffe feefeffe 3fff1360 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
áSystem Initiating...

Soft WDT reset

ctx: cont 
sp: 3fff2040 end: 3fff2380 offset: 01b0
```

igrr, 2016-08-09:
Do you have a NodeMCU or Adafruit Huzzah to try the same sketch on?

AdamMiltonBarker, 2016-08-09:
No I don't mate

AdamMiltonBarker, 2016-08-09:
Could be related to pull ups/pull downs? Which are required for the 12f ?

AdamMiltonBarker, 2016-08-09:
I re did the process this is what happened but it did go to AP:

```wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xad
csum 0xad
v09f0c112
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
[... 647 lines cut ...]
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0¸0øþ```

AdamMiltonBarker, 2016-08-09:
This time no crash on adding the config in the PA form, continuing now,

AdamMiltonBarker, 2016-08-09:
IT APPEARS TO BE WORKING :D Including the http update.

AdamMiltonBarker, 2016-08-09:
@igrr dude it is working I have reset loads of times and it is working, just going to reflash without the debug etc and check.

AdamMiltonBarker, 2016-08-09:
@igrr confirmed mate, now if I just need to get the actual update script working correctly and I can shove this on a PCB at last :+1: Thank you for the help, the only thing I hadn't done previously was changed the baud rate though not sure what has actually fixed the issue but thanks for your time anyway.

closed 2016-08-09 by AdamMiltonBarker