esp8266/Arduino #2371: ESP-8266 WiFi.config(ip, gateway, subnet);

dvukovic, 2016-08-07:
I am using a NodeMcu V3 board.
I am using IDE 1.6.8
 I updated the ESP-8266 library to ver 2.2.0.

I loaded the HelloServer program and tested that.
It does work.

With the original file, I get an [IP Address: 10.0.0.14]  and everything appears to work.
--
hello from esp8266!
--
When I add :
IPAddress ip(192, 168, 0, 50);  // this 3 lines for a fix IP-address
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);
WiFi.config(ip, gateway, subnet);  //  before or after Wifi.Begin(ssid, password);

The message states [IP Address: 192.168.0.50]  but it does not connect:
--
This site can't be reached
192.168.0.50 took too long to respond.
--

Any ideas about this ?

Don

kiralikbeyin, 2016-08-07:
Syntax

WiFi.config(ip); 
WiFi.config(ip, dns); 
WiFi.config(ip, dns, gateway); 
WiFi.config(ip, dns, gateway, subnet); 
Parameters

ip: the IP address of the device (array of 4 bytes)

dns: the address for a DNS server.

gateway: the IP address of the network gateway (array of 4 bytes). optional: defaults to the device IP address with the last octet set to 1

subnet: the subnet mask of the network (array of 4 bytes). optional: defaults to 255.255.255.0

https://www.arduino.cc/en/Reference/WiFiConfig

Good luck

luc-github, 2016-08-07:
@kicurrently ESP does not follow same order as arduino, dns is at the end
have a check here : https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h#L42

dvukovic, 2016-08-07:
I changed the single line as suggested.
Still will not connect.

IP Address: 192.168.0.50
gateway Address: 192.168.0.1
subnetMask: 255.255.255.0
signal strength (RSSI):-59 dBm
HTTP server started

brendanmatkin, 2016-08-23:
@dvukovic it sounds like 192.168.0.1 is not the gateway of the router you are trying to connect to - because you get an assigned IP of 10.0.0.14 from your router, it is likely that the router gateway is 10.0.0.1. 

The way I understand it, if you want to connect to the 192.168.0.1/24 subnet you need to set your router's IP address to 192.168.0.1, which is the same as the gateway IP for your ESP. 

Think of it like the gateway IP is the gate/entrance to a given subnet (mask of 255.255.255.0 = subnet ip range of 192.168.0.1-192.168.0.255). 

Is this your problem?

dvukovic, 2016-08-23:
Let me get back to some basics.
I want to connect to the ESP-8266 from my phone or tablet.
If the ESP-8266 connects via DHCP too that router, I can not comm with that ESP-8266 without knowing what address DHCP that router assigned to it.
Having the ESP-8266 ask for a specific address, as in:  WiFi.config(ip, gateway, subnet, gateway); 
The WiFi.config(ip...) should be assigned to the ESP-8266 by the router.
( as is my understanding )
But, it does not work and the router will not assign that (ip) address to the ESP-8266.
 So, maybe my question should be, how to I get the ESP-8266 address after it gets an DHCP address ?

I hope I have cleared up my confusion.

brendanmatkin, 2016-08-23:
Here are four ways, in random order. There are probably more: 

1. Static Lease in router (IMO the best way), aka Address Reservation. Again, logging in to your router, you can assign a "static lease" - by doing this your router always  reserves the same IP address to devices you assign, based on MAC address. See [this tutorial:](http://www.howtogeek.com/69612/how-to-set-up-static-dhcp-on-your-dd-wrt-router/). This is maybe the best way IF you always use the same router for your application.

2. Print out dynamic IP (easiest). Print out the IP address of your device to a serial port using [WiFi.localIP()](http://esp8266.github.io/Arduino/versions/2.3.0/doc/libraries.html#wifi-esp8266wifi-library): ``Serial.println(WiFi.localIP())``. Obviously this is only useful if you have a wired serial connection, but it's also the easiest.

3. Find ESP's dynamic IP in router. Once your ESP has connected to the network, log into your router and look at it's DHCP client table. You'll see your device listed there with the IP which has been assigned to it. A router will try to assign the same IP to a device every time it connects but no guarantee, especially over time. Helps to use: ``WiFi.hostname(deviceName);`` - deviceName will show up as Client Name in DHCP client table. 

4. Static IP. Sounds like this is the method you have been trying, it is also the most finicky. I find that the code below seems to work best (changing the order of these calls can change the behaviour a little):
```
WiFi.mode(WIFI_STA);
WiFi.hostname(deviceName);      // DHCP Hostname (useful for finding device for static lease)
WiFi.config(staticIP, gateway, subnet);  // (DNS not required)
WiFi.begin(ssid, password);
```

**How to choose static IP:**
Again, when setting a static IP you should log in to your router. Check the DHCP address range. It might be something like 10.0.0.2-10.0.0.99, but it can vary depending on the router (my router right now is 10.0.0.100-10.0.0.199). Make sure to choose a static address outside of this range (e.g. 10.0.0.201).

Also make sure to note the first three octects (numbers) of your router IP (probably 10.0.0.x, 192.168.0.x, or 192.168.1.x). Make sure your chosen static IP matches those first 3. Your router IP is the IP Address of the LAN in the settings of your router (also the address you type in to the browser to get to get there). This router IP is your gateway for your ESP. 

SO, in your case I would guess:
- static IP: 10.0.0.201 (or any 4th octet that is outside the DHCP range of your router)
- gateway: 10.0.0.1
- subnet: 255.255.255.0

Notice how the first three numbers (octets) of the static IP and gateway are the same. The gateway is the IP address of the Router. The 4th octet of the static IP is outside of the DHCP range of the router. For now, just always use the above subnet.

dvukovic, 2016-08-23:
Thanks, I'll try these things.

kentaylor, 2016-10-11:
I use an android app called [Fing](https://play.google.com/store/apps/details?id=com.overlook.android.fing&hl=en) which will show you the IP address of all the esp8266 devices on your network.

mistergreen, 2017-02-22:
Hi, 
WiFi.config(ip);
Throws an error. It's expecting ip, gateway, subnet.

> bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);

I have an arduino ethernet sever where if I include the gateway and subnet, it wouldn't POST data to my php script on a web server. I'm seeing the same problem on the WiFi esp8266.

If any body knows how to solve 1 or 2 of the issue, let me know.

akashash, 2017-06-27:
Hi I am getting this, ets Jan 8 2013,rst cause:4, boot mode:(3,6), while using "Wifi.config()" to set static IP. If I am not using this...my code is working fine. But with this...I am getting this issue. Can anyone please help??

lrmoreno007, 2017-06-27:
Upload your code. I'm working with fixed IP and I don't have problems.

akashash, 2017-06-28:
Thanks for helping out. Here is my code. Please have a look.

/*********
  Rui Santos
  Complete project details at http://randomnerdtutorials.com
*********/

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

MDNSResponder mdns;

// Replace with your network credentials
const char* ssid = "Vertos";
const char* password ="aNonymous#1797";

ESP8266WebServer server(80);

String webPage = "";

int gpio0_pin = 0;
int gpio2_pin = 2;

IPAddress ip(192, 168, 1, 24); // where xx is the desired IP Address
IPAddress gateway(192, 168, 1, 1); // set gateway to match your network

IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network

void setup(void){

   webPage += "<h1>ESP8266 Web Server</h1><p>Socket #1 <a
href=\"socket1On\"><button>ON</button></a> <a
href=\"socket1Off\"><button>OFF</button></a></p>";
  webPage += "<p>Socket #2 <a
href=\"socket2On\"><button>ON</button></a> <a
href=\"socket2Off\"><button>OFF</button></a></p>";

  // preparing GPIOs
  pinMode(gpio0_pin, OUTPUT);
  digitalWrite(gpio0_pin, LOW);
  pinMode(gpio2_pin, OUTPUT);
  digitalWrite(gpio2_pin, LOW);

  delay(1000);

  Serial.begin(115200);
  Serial.println("Setting static ip to : ");
  Serial.println(ip);
  Serial.println("Connecting to Vertos...");
  WiFi.config(ip, gateway, subnet);

  //checking for SHIELD
  if (WiFi.status() != WL_NO_SHIELD) {
    Serial.println("WiFi shield is present");
    while(true);  // don't continue
  }

  WiFi.begin(ssid, password);
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }

  server.on("/", [](){
    server.send(200, "text/html", webPage);
  });
  server.on("/socket1On", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio0_pin, HIGH);
    delay(1000);
  });
  server.on("/socket1Off", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio0_pin, LOW);
    delay(1000);
  });
  server.on("/socket2On", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio2_pin, HIGH);
    delay(1000);
  });
  server.on("/socket2Off", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(gpio2_pin, LOW);
    delay(1000);
  });
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}

lrmoreno007, 2017-06-29:
You must set "WiFi.mode(WIFI_STA);" after WiFi.config and before WiFi.begin

Please close this.

closed 2017-06-29 by dvukovic

amirusamah, 2017-10-10:
hi all. currently i am doing this also. i have done for fixing the ip address. right now, my problem is to connect to vb. 
i want to display in the vb what were displayed on the ip address that i fixed it using your coding. in the ip address, i have enter a little bit information. so i need the information in the ip, to display in my vb

very appreciated if you can help

suryanarayanan15, 2017-12-05:
@Irmoreno007
#include<ESP8266WiFi.h>
const char* ssid="narayan"; 
const char* password="12345678";

int ledpin=2;
WiFiServer server(80);

void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
delay(10);
pinMode(ledpin,OUTPUT);
digitalWrite(ledpin,LOW);
Serial.println();
Serial.println();
Serial.println("connecting to");
Serial.println(ssid);
IPAddress ip(); // this 3 lines for a fix IP-address
IPAddress gateway();
IPAddress subnet();
WiFi.config(ip, gateway, subnet);
WiFi.mode(WIFI_STA);
WiFi.begin(ssid,password);
while(WiFi.status()!=WL_CONNECTED)
{ delay(500);
Serial.print(".");
}
Serial.print("");
Serial.print("WiFi connected");
server.begin();
Serial.println("server started");

Serial.print( " use the url to connect: ");
Serial.print("http://");
Serial.print(WiFi.localIP());
Serial.println("/");
}
void loop() {
  // put your main code here, to run repeatedly:
WiFiClient client=server.available() ;
if(!client)
{
  return;
}
Serial.println("new client");
while(!client.available())
{
  delay(1);
 
 }
 String request =client.readStringUntil('\r');
 Serial.println(request);
 client.flush();

 int value = LOW;
 if(request.indexOf("/LED=ON")!=-1)
 {
  digitalWrite(ledpin,1);
  value=HIGH;
 }
 if(request.indexOf("/LED=OFF")!=-1)
 {
  digitalWrite(ledpin,0);
  value=LOW;
 }
 client.println("HTTP/1.1 200 ok");
 client.println("Content-type:text/html");
 client.println("");
 client.println("<!DOCTYPE HTML>");
 client.println("<html>");

 client.print("ledpin is now: ");
 if(value==HIGH)
 {
  client.print("on");
 }
 else
 {
  client.print("off");
 }
 client.println("<br><br>");
 client.println("<a href=\"/LED=ON\"\"><button>turn on</button></a><br/>");
 client.println ("<a href=\"/LED=OFF\"\"><button>turn off</button></a><br/> ");
 client.println ("</html>");
 delay(1);
 Serial.println("client disconnected");
 Serial.println("");
 
 }

this is my code ....how can i connect my esp8266 to fixed client..no other client cant be able to access  my esp8266 wifi.

amirusamah, 2017-12-06:
Add this code to fixed your ip address :

IPAddress ip(192, 168, 11, 23);
IPAddress gateway(192, 168, 11, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(192, 168, 11, 1);
WiFi.config(ip, dns, gateway, subnet); 
WiFi.begin(ssid, password);

suryanarayanan15, 2017-12-06:
@amirusamah i have tried the above code with my laptop's ip ,gateway,subnet,dns. but it does not work.
site can't be reached is the webpage i get. what i can do

suryanarayanan15, 2017-12-07:
serial moniter returns me a url http://198.163.11.1.......can i control the clients connecting to this url

amirusamah, 2017-12-07:
your serial monitor returns you the url because in your code have this "IPAddress gateway(192, 168, 11, 1);" which i have give you before. you can change it to any url as long a "192.168........" . what i dont understanding is, what did you mean by control the client connecting to the url? what do you want to control?

suryanarayanan15, 2017-12-07:
can i control the site from unknown client? i need that site to be connected to some specific clients.....i think  to control the clients for the site , it is not the part of the code..is there any way to protect the site

amirusamah, 2017-12-07:
input and output is based from your esp8266. The information on the url is just to display the output from your esp8266. so anything should be control from esp8266

suryanarayanan15, 2017-12-07:
my code will make a url like http..
but my need is to make https...
how can i make

suryanarayanan15, 2017-12-07:
i want to make a password protected website...please help

devyte, 2017-12-07:
@suryanarayanan15 this is not the right place too ask for help, and this issue is closed. Please refer to a community forum for assistance. See our issue policy doc for more details.

suryanarayanan15, 2017-12-07:
OK.sorry for the disturbances

castillo92, 2019-04-03:
> Here are four ways do deal with ESP IP Addressing, in random order. There are probably more:
> 
> 1. Static Lease in router (IMO the best way), aka Address Reservation. Again, logging in to your router, you can assign a "static lease" - by doing this your router always  reserves the same IP address to devices you assign, based on MAC address. See [this tutorial:](http://www.howtogeek.com/69612/how-to-set-up-static-dhcp-on-your-dd-wrt-router/). This is maybe the best way IF you always use the same router for your application.
> 2. Print out dynamic IP (easiest). Print out the IP address of your device to a serial port using [WiFi.localIP()](http://esp8266.github.io/Arduino/versions/2.3.0/doc/libraries.html#wifi-esp8266wifi-library): `Serial.println(WiFi.localIP())`. Obviously this is only useful if you have a wired serial connection, but it's also the easiest.
> 3. Find ESP's dynamic IP in router. Once your ESP has connected to the network, log into your router and look at it's DHCP client table. You'll see your device listed there with the IP which has been assigned to it. A router will try to assign the same IP to a device every time it connects but no guarantee, especially over time. Helps to use: `WiFi.hostname(deviceName);` - deviceName will show up as Client Name in DHCP client table.
> 4. Static IP. Sounds like this is the method you have been trying, it is also the most finicky. I find that the code below seems to work best (changing the order of these calls can change the behaviour a little):
> 
> ```
> WiFi.mode(WIFI_STA);
> WiFi.hostname(deviceName);      // DHCP Hostname (useful for finding device for static lease)
> WiFi.config(staticIP, gateway, subnet);  // (DNS not required)
> WiFi.begin(ssid, password);
> ```
> **How to choose static IP:**
> Again, when setting a static IP you should log in to your router. Check the DHCP address range. It might be something like 10.0.0.2-10.0.0.99, but it can vary depending on the router (my router right now is 10.0.0.100-10.0.0.199). Make sure to choose a static address outside of this range (e.g. 10.0.0.201).
> 
> Also make sure to note the first three octects (numbers) of your router IP (probably 10.0.0.x, 192.168.0.x, or 192.168.1.x). Make sure your chosen static IP matches those first 3. Your router IP is the IP Address of the LAN in the settings of your router (also the address you type in to the browser to get to get there). This router IP is your gateway for your ESP.
> 
> SO, in your case I would guess:
> 
> * static IP: 10.0.0.201 (or any 4th octet that is outside the DHCP range of your router)
> * gateway: 10.0.0.1
> * subnet: 255.255.255.0
> 
> Notice how the first three numbers (octets) of the static IP and gateway are the same. The gateway is the IP address of the Router. The 4th octet of the static IP is outside of the DHCP range of the router. For now, just always use the above subnet.

Thanks you!! It works!

I forgot to write:

`  WiFi.mode(WIFI_STA);
`

jstuewe, 2019-09-30:
I had some old code that used to work fine with WiFi.config(staticIP, gateway, subnet) but WiFi.hostByName in my NTP routine stopped working with later (ie. 2.5.0) libraries.

Apparently the older libraries didn't need the DHCP server in WiFi.config but the newer libraries do.
I fixed it by adding the 'gateway' at the end as the DHCP server.  ie. WiFi.config(staticIP, gateway, subnet, **gateway**)

chupocro, 2021-03-20:
> I had some old code that used to work fine with WiFi.config(staticIP, gateway, subnet) but WiFi.hostByName in my NTP routine stopped working with later (ie. 2.5.0) libraries.
>
> Apparently the older [...]

You are right, a few years ago static IP worked without specifying DNS - probably the same IP as gateway was used as DNS IP if DNS wasn't specified in `wifi.config()`

I have some older code from 2016. and 2017. I had to rewrite for compiling it with present libraries because the devices couldn't connect to the server with the old code until I added DNS. DHCP has nothing to do with the static IP address, someone will probably confirm the IP used for DNS was the same as gateway in the old libraries.