esp8266/Arduino #2372: Parse HTTP Request Arduino

anujmattoo, 2016-08-07:
### Basic Infos

#### Hardware
Hardware:	       ESP-12E
Core Version:       2.3.0

### Description

How can I parse this string `GET /STA/ID=HelloWorld/Pass=Testin123 HTTP/1.1 `First I need to check for `STA`, if it exists, continue to scan the string. Put Value of `ID` in this case `HelloWorld` should be store in String data type `SSID` and Value of `Pass`, in this case `Testin123` should be store in String data type `Pass`

It should confirm the presence of `STA` in string first. If it is not present, do not enter into loop. If it exits, search for `ID` and `Pass`. Store it.

The code I used, seems to be not working perfectly.

`#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

char GetString[100];

String SSID;
String Pass;

String s = "HTTP/1.1 200 OK\r\nConnection: close\r\nContent-Length: 0\r\n\r\n";

WiFiServer server(80); //Initialize the server on Port 80

void setup()
{
  Serial.begin(115200); //Start communication between the ESP8266-12E and the monitor window
  WiFi.mode(WIFI_AP); //Our ESP8266-12E is an AccessPoint
  WiFi.softAP("Hello_IoT", "12345678"); // Provide the (SSID, password); .
  server.begin(); // Start the HTTP Server

  IPAddress HTTPS_ServerIP = WiFi.softAPIP(); // Obtain the IP of the Server
  Serial.print("\n");
  Serial.print("Server IP is: "); // Print the IP to the monitor window
  Serial.println(HTTPS_ServerIP);
}

void loop()
{
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  //Looking under the hood
  Serial.println("Somebody has connected :)");

  String incoming = client.readStringUntil('\r');
  Serial.println(incoming);
  incoming.toCharArray(GetString, 100);

  char *get = strtok(GetString, " ");
  char *request = strtok(NULL, " ");
  char *rtype = strtok(NULL, " ");

  Serial.print("\n");
  Serial.print("GetString :");
  Serial.println(GetString);

  Serial.print("\n");
  Serial.print("Request :");
  Serial.println(request);

  Serial.print("\n");
  Serial.print("Rtype :");
  Serial.println(rtype);

  if (request != NULL)
  {
    char *part = strtok(request, "/");

    while (part)
    {
      if (!strcmp(part, "STA"))
      {
        if (!strncmp(part, "ID=", 3))
        { // We have the ID
          SSID = String(part + 3);
          Serial.print("\n");
          Serial.print("SSID: ");
          Serial.print(SSID);

        }

        if (!strncmp(part, "Pass=", 5))
        { // We have the password
          Pass = String(part + 5);
          Serial.print("\n");
          Serial.print("PWD: ");
          Serial.print(Pass);
        }
      }
      else
      {
          Serial.print("\n");
          Serial.print("Not Found");
      }
      part = strtok(NULL, "/");

    }
  }
}`

### Settings in IDE

Module:  		        NodeMCU(ESP-12E)
Flash Size: 		        4MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	        80Mhz
Upload Using:		OTA
Reset Method:		nodemcu

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

majenkotech, 2016-08-07:
I have told you what you need to do on Arduino.SE. This is not a discussion forum. This is a bug tracking system.

anujmattoo, 2016-08-07:
I read that, but I am unable to find out other way to do this. I didn't get reply there. Had to post on other forums as well. Please help! I'm in real trouble.

closed 2016-08-07 by anujmattoo

majenkotech, 2016-08-07:
You haven't looked recently then.