esp8266/Arduino #2373: Search mDNS when resolving DNS names

madpilot, 2016-08-08:
Feature request: When automatically resolving domain names, would it be possible to search using the mDNS broadcast address if the domain name ends with .local?

If the domain name ends with .local, using a DNS server ip of 224.0.0.251 and a port of 5353 is all that needs to happen, so there is minimal code change.

If someone wants to point the right file, I'm happy to change it and put it in a PR.

n8henrie, 2016-09-12:
This would be great. I'm trying to get mdns to work right now just for this purpose but not having any luck.

@madpilot -- do you have a functioning `hostname.local` resolution workaround currently?

madpilot, 2016-09-12:
@n8henrie Not right now - I finally have master running in my IDE, so I'm hoping to have a poke around this week.

railnerd, 2017-03-18:
should be possible to pull latest stable lwip (2_0_2) and

#define LWIP_DNS_SUPPORT_MDNS_QUERIES 1

me-no-dev, 2017-03-18:
we should actually port the mdns from ESP32. It has host and service queries and all. I need to add a couple of features before merging though.

railnerd, 2017-03-19:
well, the root cause for not being able to resolve .local addresses lies in lwip, but certainly parity between ESP32 and ESP8266 would be nice.

madpilot, 2017-03-19:
I ended up writing a small library that does this: https://github.com/madpilot/mDNSResolver

devyte, 2017-10-14:
@me-no-dev will you be porting the esp32 mdns? no hurry. Also, it may make more sense after migrating to lwip2.

devyte, 2017-10-14:
@madpilot I assume from your last comment that your mDNSResolver addresses this issue?

me-no-dev, 2017-10-14:
i hope some not so full version of it can be ported. 3 interfaces and IPv6 on ESP32 make things a bit intensive and i do not know how 8266 will coop even with v4 on 2 interfaces. But there is a way I imagine to port most of it (once done)

devyte, 2019-01-11:
Automatic switchover to MDNS lookup in HostByName() for .local hostnames would forcibly pull in the MDNS code for all users, which is undesirable. I suggest a simple app function along the lines of:
```
myLookup(hostname)
{
  if(hostname ends in .local)
    look up with MDNS lib
  else
    look up with hostByName()
}
```
Closing.

closed 2019-01-11 by devyte