esp8266/Arduino #2375: Idea: Integrate ESP8266 USB Software Driver

probonopd, 2016-08-08:
https://github.com/cnlohr/espusb is a software USB stack running on the ESP8266/ESP8285. It requires only one external resistor. As is shown on http://www.esp8266.com/viewtopic.php?f=6&t=11188#p52430 it can already be used to flash the ESP8266.

I hope it can be nicely integrated nicely into esp8266/Arduino (e.g., to do the uploading and "serial" monitor using just the ESP and no serial adapter).

I also cross-posted this at https://github.com/cnlohr/espusb/issues/1

cnlohr, 2016-08-08:
I intend to keep most of my posts in the espusb copy of the ticket -- but -- that said, I am very interested in facilitating this, but I don't have any expertise with Arduino, and I'm not really tooled to get into it... so I want to help any way I can, but someone else is going to have to run point on the Arduino side.  I can then take requests on my side.

me-no-dev, 2016-08-09:
I would like to take on integrating it. I have experience with usb

cnlohr, 2016-08-09:
What would be the general architecture you would propose?

cnlohr, 2016-08-09:
I still think control messages would probably be the way to go.  When I say architecture - I mean would you put USB code somewhere that never gets overwritten?  So, when it boots it immediately enumerates as USB, after a brief time, without any requests to re-flash, it continues booting the main code?  Otherwise, it stays in the USB portion?

P.S. I still intend to greatly simplify my low-speed implementation and shrink it, as well as further detaching it form the Espressif SDK so it could run stand-alone.

me-no-dev, 2016-08-11:
You can enumerate different devices at about any time. Just pull D+ I think it was to GND and the host will reenumerate the device, so you can have one device in the bootloader which will wait some time for upload and then reenumerate as another device once it boots.
Since we are trying to conform to official Arduino code, I would probably implement it as it is in for leonardo or similar or like v-usb which will do the same thing.
I glanced at the code but have not had time to disect it and see what's needed where to make it work :) will ping you soon with questions though :)

One question: I saw a discussion about pin interrupts on the forum and wanted to ask how you are using the interrupt? Are you detecting just the first pulse that initiates communication or are you detecting every other pulse as well?

cnlohr, 2016-08-11:
I will likely be adding another layer to my code in the next few days.  Currently, the code to handle descriptors, custom endpoints and control messages is in the same file as the mid-level USB stack (What handles the DATAx toggling, IN/OUT, etc.)

Other than that, it shouldn't change too much.  I've already reorganized the way endpoints works to be much cleaner than I first did.

Only the first pulse. There's too much state to maintain between pulses and, interrupts on the ESP are SOOO SLOWW You don't even get into the handler till like the third bit!  Once in the interrupt handler from the start of a message, I use "ccount" to track progress and advance 53 (or 54) cycles along for every bit.

The discussion about interrupts stemmed from me wanting to make my devices enumerate as full-speed.  I hope to start on this in earnest this weekend.  Though after working through the numbers more, I am afraid full-speed will add too much IRAM/DRAM overhead to be practical.

romansavrulin, 2017-09-25:
@me-no-dev Is there any progress on this?

devyte, 2017-10-14:
@cnlohr I took a look at your espusb repo, and your high speed adc in colorchord, and I think they are awesome. I expect to take a look at the high speed adc at some point, with a view of getting it to work with this repo.
I may also take a look at this usb code further down the line. I know very little of usb, but I have a shelved project about it, and this may be a good way to learn.
A possibly stupid question: do you think the ESP could act as a simplistic USB host?

cnlohr, 2017-10-16:
Yes, very possible, but until someone churns through it.... OOORRRRR espressif grants access to GPIO via the system registers on future silicon revisions, it'll be stuck in low speed :-/