esp8266/Arduino #2376: How can I auto load default web page after ESP8266HTTPUpdateServer successful?

WheresWally, 2016-08-09:
I am loving the OTA updates using ESP8266HTTPUpdateServer.

however after loading /update and successfully selecting the new INO file and all that,
I am seeing the **http://192.169.0.222/update** web page.

Is there a way that I can make the browser show **http://192.169.0.222/** ????

Has anyone found a way to avoid then need to manually reload a different web page after updating?
Terry

penfold42, 2016-08-11:
You would need to modify the "success... Rebooting" with some JavaScript to get the browser to redirect after a few seconds

everslick, 2016-08-11:
```
static void send_redirect(const String &redirect) {
  html += F("HTTP/1.1 301 OK\r\n");
  html += F("Location: ");
  html += redirect;
  html += F("\r\n");
  html += F("Cache-Control: no-cache\r\n\r\n");

  webserver->sendContent(html);
}
```

devyte, 2017-10-14:
Closing per #3655 .

closed 2017-10-14 by devyte