esp8266/Arduino #2377: WatchDogTimer issue

rudydevolder, 2016-08-10:
#### Hardware:			ESP-12

### Problem Description:
The Example below is adapted to work but WatchDogTimer Resets the board and dumps the stack when the delay-commands are removed or when you increase the amount of steps in **myMotor->step(100)**

### Settings in IDE

Module:  			Feather Huzzah
Flash Size: 		4MB
CPU Frequency:	80Mhz

Upload Using:		SERIAL

### Sketch

/* 
This is a test sketch for the Adafruit assembled Motor Shield for Arduino v2
It won't work with v1.x motor shields! Only for the v2's with built in PWM
control

For use with the Adafruit Motor Shield v2 
---->	http://www.adafruit.com/products/1438
*/

#include <Wire.h>
#include <Adafruit_MotorShield.h>
#include "utility/Adafruit_MS_PWMServoDriver.h"

// Create the motor shield object with the default I2C address
Adafruit_MotorShield AFMS = Adafruit_MotorShield(); 
// Or, create it with a different I2C address (say for stacking)
// Adafruit_MotorShield AFMS = Adafruit_MotorShield(0x61); 

// Connect a stepper motor with 200 steps per revolution (1.8 degree)
// to motor port #2 (M3 and M4)
// My Mini stepper has 513 steps per revolution =>
Adafruit_StepperMotor *myMotor = AFMS.getStepper(513, 2);

void setup() {
  Serial.begin(9600);           // set up Serial library at 9600 bps
  Serial.println("Stepper test!");
  Serial.println("Stepper test!");
  
  AFMS.begin();  // create with the default frequency 1.6KHz
  //AFMS.begin(1000);  // OR with a different frequency, say 1KHz
  
  myMotor->setSpeed(10);  // 10 rpm   
}

void loop() {
  Serial.print("Single coil steps    : FORWARD ");
  myMotor->step(100, FORWARD, SINGLE); 
  Serial.println(", BACKWARD"); delay(1);
  myMotor->step(100, BACKWARD, SINGLE); 
  delay(1);

  Serial.print("Double coil steps    : FORWARD ");
  myMotor->step(100, FORWARD, DOUBLE); delay(1);
  Serial.println(", BACKWARD");
  myMotor->step(100, BACKWARD, DOUBLE);
  delay(1);
  
  Serial.print("Interleave coil steps: FORWARD ");
  myMotor->step(100, FORWARD, INTERLEAVE); delay(1);
  Serial.println(", BACKWARD");
  myMotor->step(100, BACKWARD, INTERLEAVE); 
  delay(1);
  
  //Serial.println("Microstep steps: FORWARD ");
  //myMotor->step(50, FORWARD, MICROSTEP); delay(1);
[... 254 lines cut ...]
  //myMotor->step(50, FORWARD, MICROSTEP); delay(1);
  //Serial.print(", BACKWARD");
  //myMotor->step(50, BACKWARD, MICROSTEP);
  Serial.println();
}

### Debug Messages

```
Soft WDT reset

ctx: cont 
sp: 3ffef320 end: 3ffef550 offset: 01b0

>>>stack>>>
3ffef4d0:  00000002 00000009 3ffee3b8 402024fc  
3ffef4e0:  40107010 00002daf 0000005d 40202724  
3ffef4f0:  00000002 0000005d 00002daf 4020278c  
3ffef500:  00000001 feefeffe 3ffee504 40202be8  
3ffef510:  00000640 3ffee3d0 3ffee504 3ffee528  
3ffef520:  3ffe8478 3ffee368 3ffee504 40202125  
3ffef530:  3fffdad0 00000000 3ffee520 40202d9c  
3ffef540:  feefeffe feefeffe 3ffee530 40100114  
<<<stack<<<
```

WereCatf, 2016-08-10:
Working as intended: watchdog has to get to do its job.

https://github.com/esp8266/Arduino/blob/master/doc/reference.md#timing-and-delays

rudydevolder, 2016-08-11:
Thank you very much @WereCatf 
I went looking in the sourcecode: "Adafruit Motorshield" for the step.command and added `delay(0)` => problem solved. ;)
Now I can run with any numbers of steps and without the delay command in my code.

devyte, 2017-10-14:
Closing as resolved.

closed 2017-10-14 by devyte