esp8266/Arduino #2380: Serial.swap() generates mangled characters.

PhilColbert, 2016-08-10:
When I do a Serial.swap() in a loop, many characters come out messsed up.

If I slow the loop down by putting a delay of perhaps 100, it then works, still some characters are bad, any ideas how to stop this?

I will be using the swap in a program that needs lots of speed ! 

Thanks

On a Nodemcu.

extern "C" {
#include <user_interface.h>
}

int i=0;

void setup()
{
  Serial.begin(9600);
}

void loop()
{
 
  Serial.print("Loop No: ");
  Serial.println(i); 
 // delay(100);
  Serial.swap();
  i++;
}

WereCatf, 2016-08-11:
I would assume you are doing Serial.swap() before it has actually emptied its outgoing buffer yet and that's why you're seeing garbled output; I do not know if it's a bug or working as intended, @igrr would probably be able to chime in on that.

In the meanwhile, why do you even need to do swap() so often? I can't think of a single scenario where a different approach wouldn't be saner.

me-no-dev, 2016-08-11:
just call Serial.flush(); before Serial.swap(); that should wait until all data is sent

PhilColbert, 2016-08-11:
What approach should be used for reading 2 lots of data, one from a GPS , one from a RF devices, pull from GPS, pull from RF?

Will try serial.flush, trying softwareserial at the moment.... well trying anyway :)

Thanks

Thanks

stickbreaker, 2016-08-14:
Serial.flush() calls this code in /esp8266/Arduino/blob/master/cores/esp8266/uart.c
```c++
void uart_wait_tx_empty(uart_t* uart)
{
    if(uart == NULL || !uart->tx_enabled) {
        return;
    }
    while(((USS(uart->uart_nr) >> USTXC) & 0xff) > 0) {
        delay(0);
    }
}
```
I don't know what USS() actually returns, but it looks like it it a count of the internal 128byte hardware TX buffer.  I don't see where it actually checks if the last bit of the currently transmitting byte has been sent out the pin?

Chuck.

vitalian12, 2016-08-23:
I have same issue, use Serial.flush() doesn't help...
and I think the mangled character come from RX not TX, so flushing TX buffer wouldn't help I think.
someone GPIO15 should be connected to GND with 10k resistor
https://github.com/esp8266/Arduino/issues/143
it also didn't help to remove the mangled character.
PhilColbert, if you solve this problem, could you leave a comment? or somebody knows about this problem, plz leave the comment

btsimonh, 2017-01-11:
If you swap the physical lines (which is what swap does; there is only one uart?) - then if this happens mid-character on rx, you will get garbage.  Basically, you can't swap() to read two different devices unless you know exactly then the data will come in.
e.g.
swap to serial 0.
send a command to get a response.
wait until the response comes or timeout.
swap to serial 1.
send a command to get a response.
wait until the response comes or timeout.

If the data comes in unsolicited from the device, you have a bit of a problem :).
If one of the feeds is slow enough, you MAY be able to find a software serial module and use a generic I/O pin for receive.

devyte, 2017-10-14:
@PhilColbert is this issue still valid with latest git?
#2427 has a workaround, have you tested that?

devyte, 2017-10-14:
@igrr what is your opinion on this? 
- should the workaround in #2427 be implemented inside flush()?
- is it the responsability of the user to make sure the rx buffer is empty before calling swap()?
- should flush() clear the rx buffer? (is this possible?)
- does SoftwareSerial suffer from something similar?

igrr, 2017-10-14:
I would go for fixing 'flush' as proposed (adding a small delay), and
documenting that 'flush' needs to be called before swap, otherwise
incomplete output will be generated.