esp8266/Arduino #2385: deleting MDNSResponder object will crash after some time

everslick, 2016-08-11:
dynamically creating a mDNS responder object on the heap is now possible, but when i delete the object the sketch will eventually crash. i have seen a timer function call after deleteing the object but i currently cannot, for the life of me, get the stack to dump a backtrace.

```
#include <Arduino.h>
#include <ESP8266mDNS.h>

const char *name = "esp";

static MDNSResponder *mdns = NULL;

void setup() {
  Serial.begin(115200);
  Serial.println();

  delay(1000);

  WiFi.mode(WIFI_STA);
  WiFi.begin("XXX", "xxxxxxxxxx");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  mdns = new MDNSResponder();

  if (mdns->begin(name)) {
    Serial.printf("MDNS: hostname set to '%s.local'\n", name);
  } else {
    Serial.print("MDNS: could not start mDNS responder\n");
  }

  Serial.println("connected!");

  Serial.println("deleting MDNS responder");
  delete (mdns);
  mdns = NULL;
}

void loop() {
  Serial.print(".");
  delay(333);
}
```

for me it looks like as there is some minor cleanup missing in the destructor, but i failed to figure it out myself. :-(

me-no-dev, 2016-08-11:
conn is not cleaned in the destructor. there might be more

everslick, 2016-08-12:
I put

```
  if (_conn) {
    _conn->unref();
  }
```

at the end of the destructor and this, indeed, seems to fix the crash. PR is on the way....

closed 2016-08-12 by everslick