esp8266/Arduino #2386: WiFi: Signal to Noise Ratio (SNR) and/or noise measurement

bitnapper, 2016-08-11:
I'd like to locate an ESP8266 with the [Mozilla Location Service](https://location.services.mozilla.com/api) (MLS) API. Therefore I need to measure the signal to noise ration (SNR). Is there a method like WiFi.noiseMeasurement() or WiFi.snr() to get this info, I could not find any?

stefanbode, 2016-08-18:
As far as I know there are two Options that would help:
WiFi.scanNetworks();
go through all available networks visible.
WiFi.RSSI(i) give the value you're looking for.

tablatronix, 2016-12-23:
RSSI is signal, noise is something else, you need noise for SNR.

gicho, 2016-12-23:
Sorry, my fingers was so fast.
Generally in packet networks SNR is not so easy to define. You need to know
if given channel is not busy, or you want to estimate the actual SNR during
your packets to the AP?
ESP has monitor mode so you can "listen" for some time, but this has many
caveats.

tablatronix, 2016-12-23:
I am talking about unmodulated RF noise floor

esp32 might have some packet noise, interesting

https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_wifi_types.h#L204

gicho, 2016-12-24:
> components/esp32/include/esp_wifi_types.h#L204

I haven't see exact match in 8266 API. This is somehow similar - this is
packet struct as reported to the promiscous mode callback:
https://github.com/david-griffith/esp8266-sniffer/blob/master/sniff.c
But still this is in case of valid received packet.

tablatronix, 2016-12-24:
Thanks, yeah I also did not see any mention of this in the forums, the RSSI is multiple packet averaged according to a response. Any rf stuff is not exposed.

yeah no actual docs anywhere on `wifi_promiscuous_cb_t`
You can find a version of `network_80211.h` with some comments for the buffers though, but that is all I found.

tablatronix, 2016-12-24:
It looks like espressif does not provide developers with any CSI or radio tap headers but they have documents on how developers should select channels. Seems odd to not provide this info most nics do.

devyte, 2017-10-14:
Nothing about noise or SNR in the api reference for SDK 2.1.0 either.
Closing.

closed 2017-10-14 by devyte