esp8266/Arduino #2387: Blink example won't finish compiling

bkhofmann, 2016-08-11:
### Basic Infos

#### Hardware
Hardware:			ESP-201
Core Version:      	?2.1.0-rc2? - not sure what this means

### Description

Brand new to using ESP8266 with Arduino IDE.   First trying to compile the "Blink" example.  The program will not finish compiling.  I copied the messages displayed in the compiler window of the IDE in the "Debug Messages" section of this form.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k/64k
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	80Mhz
Upload Using:		OTA
Reset Method:		?ck / nodemcu? - not sure what this means

### Sketch

```
/*
 ESP8266 Blink by Simon Peter
 Blink the blue LED on the ESP-01 module
 This example code is in the public domain
 
 The blue LED on the ESP-01 module is connected to GPIO1 
 (which is also the TXD pin; so we cannot use Serial.print() at the same time)
 
 Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```

### Debug Messages

```
Arduino: 1.6.8 (Windows XP), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

In file included from c:\documents and settings\benjamin c. hofmann\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62:0,

                 from C:\Documents and Settings\Benjamin C. Hofmann\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\Esp.cpp:24:

c:\documents and settings\benjamin c. hofmann\local settings\application data\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

krzychb, 2016-08-12:
Hi @bkhofmann,

This looks like the issue with path length of Windows XP.

Please check https://github.com/esp8266/Arduino/issues/2250 or https://github.com/esp8266/Arduino/issues/1631 for the full sory.

Krzysztof

bkhofmann, 2016-08-13:
Thank you @krzychb,

That was it!  I moved some of the files to a directory with a shorter path, and I also had to re-install the IDE on a different user account that had a shorter name.  After that, the compile worked as expected.  Thank you!

-Ben

closed 2016-08-13 by bkhofmann

udayaprakashjayaraman, 2019-08-23:
I too had same issues and now its rectified after following below mentioned steps:

Step 1: Download this file in link  -  https://drive.google.com/open?id=1ymTPH6M70T7Y5JrLw8QkMRD_Et3u7LSh

Step 2: go to this file location in your computer   

C:\Users\(User name of your Login)\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\bin

Step 3: Paste extracted file (xtensa-lx106-elf-g++ ,   xtensa-lx106-elf-gcc ) and replace it with the files downloaded from step 1 link.

( If unable to replace and get error ( File is running in other application) then open task manager and check any program (starts with *xtensa* this name) and end process of that and then try to copy that file. 

Now that error is rectified......!