esp8266/Arduino #2392: HTTPUpdate security

AdamMiltonBarker, 2016-08-13:
Using the MAC as an identifier for an update is not very secure as it can be easily spoofed, does anyone have any suggestions for a more secure approach?

carlosrodriguesIT, 2016-08-13:
Maybe you can use a MAC address and a static IP address to secure a connection...

AdamMiltonBarker, 2016-08-13:
Possibly although both could be found with an app like Fing. I was thinking of maybe adding api key and secret to spiffs and replicating my other API auth using an extra header in the update.

carlosrodriguesIT, 2016-08-13:
I think you have thought of a solution , now only have to apply, API key is a good solution, because is a long private string, not easy to discover, but HTTP is a weak protocol, any snnifer can be used to find your key.

AdamMiltonBarker, 2016-08-13:
Not if it is encrypted though. I have secure MQTT implemented as well any ideas for the future for support of MQTT

carlosrodriguesIT, 2016-08-13:
OK, if you use MQTT over SSL port 8883 by default is a good security measure, for MQTT I have many ideias, but still in resource.

AdamMiltonBarker, 2016-08-13:
Yes I have my own broker which is used for clients and devs and it only allows connection on port 8883.

carlosrodriguesIT, 2016-08-13:
Anyway, your broker is based in mosquitto, or do you have your own code?

AdamMiltonBarker, 2016-08-13:
Yes it is a mosquitto broker mixed with the auth plugin.

devyte, 2017-10-14:
Not the right place to have this discussion.
Closing as non-issue per #3655 .

closed 2017-10-14 by devyte