esp8266/Arduino #2394: Crashing after HTTPUpdate

AdamMiltonBarker, 2016-08-14:
I have the update working on an ESP01 but after the update and reset when it hits the update code again in the setup function it crashes:

```dhcp client start...
.........ip:192.168.1.12,mask:255.255.255.0,gw:192.168.1.2
.Successfully connected to WiFi
please start sntp first !
ssl->need_bytes=7568 > 6859
sleep disable
state: 5 -> 0 (0)
rm 0
del if0
bcn 0
del if1
usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
@cp:0
ld
System Initiating...
Mounted file system!
Waiting for Wifi to connect...
..............Successfully connected to WiFi

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
@cp:0
ld
ÿ```

If I remove the updateESPFirmWare it does not crash and continues to connect to MQTT any ideas guys ?

```
void updateESPFirmWare(){ 
    t_httpUpdate_return ret = ESPhttpUpdate.update("https://iot.techbubbletechnologies.com/updates/updater.php","","E1 D2 56 D8 9B 14 F2 8E 3D 36 D9 35 BC B3 4F 19 45 C1 97 98");
    switch(ret) {
         case HTTP_UPDATE_FAILED:
            Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
            Serial.println("");
            break;
        case HTTP_UPDATE_NO_UPDATES:
            Serial.println("HTTP_UPDATE_NO_UPDATES");
            break;
        case HTTP_UPDATE_OK:
            Serial.println("HTTP_UPDATE_OK");
            break;
    }   
}

void setup() {  
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  //SPIFFS.format();
  delay(10);
  Serial.println(F("System Initiating..."));
  client.setServer(mqtt_server,mqttPort);
  client.setCallback(callback);
  readSpiffs();  
  if ( esid != "" ) {    
      WiFi.begin(esid.c_str(), epass.c_str());      
      if (checkWifiConnection()) {        
        serverType = "WC";
        Serial.println(F("Successfully connected to WiFi")); 
        updateESPFirmWare();        
        return;        
      } 
  }  
  serverType = "AP";
  server=new(ESP8266WebServer);
  initiateAccessPoint();  
}```

AdamMiltonBarker, 2016-08-14:
I can see from my logging of updates in the database each time the HTTP_X_ESP8266_FREE_SPACE is getting smaller and smaller so maybe this is the issue, but how do you restore the space ?

AdamMiltonBarker, 2016-08-14:
Fixed

closed 2016-08-14 by AdamMiltonBarker

HassaanAkbar, 2017-02-24:
@adam.. I have a same problem... How did u solve it??