esp8266/Arduino #2395: ESP crashes, throwing exception 29

anujmattoo, 2016-08-15:
### Basic Infos

#### Hardware
Hardware: ESP-12F
Core Version: 2.3.0

### Description

I am getting exception 29 sometimes and it crashes the ESP12F. I am sending HTTP Request every 15-30ms to ESP. It sometimes crashes after thousand request and sometimes crashes after few hundred request. My project requires HTTP Request to be sent every 15-30ms. I have increased the CPU Frequency from 80Mhz to 120Mhz.

I have attached the ESP Exception stack decoder result.

Please help! @me-no-dev @igrr @majenkotech 

### Settings in IDE

Module: NodeMCU(12-E)
Flash Size: 4MB
CPU Frequency: 120Mhz
Flash Mode: qio
Flash Frequency: 40Mhz
Upload Using: SERIAL
Reset Method: ck / nodemcu

### Sketch

```
#include <ESP8266WiFi.h>
#include <EEPROM.h>
#include <ESP8266WebServer.h>

WiFiServer server(80);

void setup()
{
  Serial.begin(115200); //Start communication between the ESP8266-12E and the monitor window
}

void loop()
{
  WiFiClient client = server.available();
  if (!client)
  {
    return;
  }

  Serial.println("Somebody has connected :)");

  String incoming = client.readStringUntil('\r');
  Serial.println(incoming);
  client.flush();

  String s = "HTTP/1.1 200 OK\r\nConnection: close\r\nContent-Length: 0\r\n\r\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  Serial.println("Client disonnected");
}

```

### Debug Messages

[image]

igrr, 2016-08-16:
If you are sending "Connection: close" header, you probably want to wait until client actually closes the connection.
Alternatively, try git version of the core — it has a tweak by @me-no-dev to eliminate TCP connections in TIME_WAIT state, freeing up memory.

anujmattoo, 2016-09-10:
Can you please mention the link to the Git version of the core?

igrr, 2016-09-10:
See readme for instructions: https://github.com/esp8266/Arduino#using-git-version

anujmattoo, 2016-09-15:
I am unable to use git version for mac! Please help

igrr, 2016-09-15:
Sure, what's the problem?

anujmattoo, 2016-09-15:
The commands use in the instructions are for windows command prompt and how can use that with mac terminal. I have downloaded the python 2.7 for mac and how to clone the repository and to where in mac?

igrr, 2016-09-15:
These commands work in mac and linux as well.
Download arduino IDE to any location you like (though i suggest to use some directory under your home directory). Then open terminal and use cd command to go inside Arduino.app directory. 
For example, i have my Arduino.app in ~/Apps/arduino/nightly/. So in Terminal, i type:
```
cd ~/Apps/arduino/nightly/Arduino.app/Contents/Java/hardware
mkdir -p esp8266com
cd esp8266com
git clone https://github.com/esp8266/Arduino.git esp8266
cd esp8266/tools
python get.py
```

By the way, before you do that, make sure you uninstall the esp8266 package installed via boards manager. Failure to do that will cause errors in Arduino IDE.

igrr, 2016-09-15:
Also, you don't need to download python 2.7 if you're on a recent OS X version. OS X comes with python already installed.

anujmattoo, 2016-09-15:
Thanks! It worked. Can you please explain what is the difference between the git version and stable version.? How  TIME_WAIT solves the crash issue?

igrr, 2016-09-15:
You can navigate repository history using GitHub to see the changes since the last release.
TIME_WAIT doesn't solve the problem, it is a state defined by the TCP spec.
This commit adds a limit to the number of pcbs in time_wait state, violating the spec but adapting the stack to the limited RAM of the 8266: https://github.com/esp8266/Arduino/commit/07f4d4c241df2c552899857f39a4295164f686f2

devyte, 2017-10-14:
Per OP's last comment, closing as resolved.

closed 2017-10-14 by devyte