esp8266/Arduino #2399: Yield context issue

luc-github, 2016-08-15:
I am using Git version and ESP12F CPU 160Mz 4MB/1MB SPIFFS and SD card library https://github.com/greiman/SdFat/tree/master/SdFat

when using :
```
File datafile = SD.open((char *)path.c_str());
                  if (!datafile){
                       LOG("Cannot open file!\n");
                       web_interface->WebServer.sendContent_P(NOT_AUTH_NF);
                       page_not_found = true;
                    }
                if (web_interface->WebServer.streamFile(datafile, contentType) != datafile.size()) {
                    LOG("Sent less data than expected!\n");
                    }
                else LOG("Stream done\n");
```

I have a crash when downloading files bigger than few Kb but no issue with small files 
I have opened an issue on sdfat https://github.com/greiman/SdFat/issues/39 and conclusion it is yield issue on esp core
Discussion on gitter start here : https://gitter.im/esp8266/Arduino?at=57b1da0323f0cacc34e7e119
and end here: https://gitter.im/esp8266/Arduino?at=57b1e7dabe8025f16943d246

luc-github, 2016-08-16:
let me know if need more information or need any test
Thank you fir your great help

martinayotte, 2016-08-16:
I've try to dig the issue further. I've updated my SdFat with the latest version, placing back the yield() instead of delay(0) and I could not reproduce the issue while doing a read of 1MB file send over TCP telnet connection.
Luc ? maybe you should provide a link to some code where you can reproduce the issue. Maybe a strip-down version of your project where the issue still there. Because it could be a interaction with other libraries you are using.

igrr, 2016-08-16:
@martinayotte this issue will only happen with latest git version, not with 2.3.0.
I didn't try reproducing this but the cause of this issue is clear. It is due to a change in WiFiClient, which now reads Stream from TCP stack callback. This change was done to achieve higher throughput when sending data. However this change has an unintended consequence that now the Stream can not use `delay` or `yield` in its read method. I am thinking about reverting to the old slower way of sending data.

martinayotte, 2016-08-16:
Ah ! Ok ! I didn't know that Luc was using latest from git.

luc-github, 2016-08-16:
@martinayotte here my project tree/branch with SD: https://github.com/luc-github/ESP3D/tree/SDTest
 Still working on it but you can have a look - I do not use another external library but SdFat

martinayotte, 2016-08-16:
Can you confirm what IGRR wrote above, you are using github version of ESP framework not 2.3.0, right ?

luc-github, 2016-08-16:
yes git version, i mentioned in bug description, I will check 2.3.0 for this as was checking another issue

PR #3214 (2017-05-09): ClientContext: don’t read DataSource in TCP callback
Previously, `_write_some` function would be called each time TCP stack
notifies the application that some data was delivered (via the `sent`
callback). In turn, `_write_some` would obtain more data to be sent from
the DataSource. In case of a DataSource backed by a Stream, this would
read from a stream. Some libraries (such as SD) may call `yield` and
other blocking operations from Stream read function, which can not be
used in TCP stack callbacks.

This change moves the data sending loop back into the Arduino task, with
a negligible loss of [... cut]
commit 7de81270 (2017-05-09): ClientContext: don’t read DataSource in TCP callback

Previously, _write_some function would be called each time TCP stack
notifies the application that some data was delivered (via the `sent`
callback). In turn, _write_some would obtain more data to be sent from
the DataSource. In case of a DataSource backed by a Stream, this would
read from a stream. Some libraries (such as SD) may call `yield` and
other blocking operations from Stream read function, which can not be
used in TCP stack callbacks.

This change moves the data sending loop back into the Arduino task, with
a negligible loss of pe [... cut]
closed 2018-01-02 by igrr