esp8266/Arduino #2400: I2C with D6T-44L

vishalkothari, 2016-08-15:
Hardware - 
ESP 12E (NodeMCU Dev Kit V3 http://cdn.frightanic.com/blog/wp-content/uploads/2015/09/esp8266-nodemcu-dev-kit-v3-pins.jpg),
D6T 44L (http://www.mouser.com/pdfdocs/D6T01_ThermalIRSensorWhitepaper.pdf)

I had interfaced Arduino Uno with D6T 44L and below code. It worked fine.

I am trying to migrate the code to work with ESP8266 and D6T 44L, but I am getting all values as zero.
I have used SDA = GPIO4 and SCL = GPIO5 with 10k pull up resistors and 3.3v to 5v logic level translator.

Below is Arduino Sketch and Customized Library to read 35byte response.
[d6t arduino.ino.txt](https://github.com/esp8266/Arduino/files/418512/d6t.arduino.ino.txt)
[WireExt.h.txt](https://github.com/esp8266/Arduino/files/418511/WireExt.h.txt)
[WireExt.cpp.txt](https://github.com/esp8266/Arduino/files/418513/WireExt.cpp.txt)

Below is ESP sketch
[d6t esp.ino.txt](https://github.com/esp8266/Arduino/files/418514/d6t.esp.ino.txt)

Thank you for your help.

Regards,
Vishal Kothari

closed 2016-08-28 by vishalkothari

a321bhi, 2019-01-28:
Hi Vishal,
Why was this closed? Found a solution?
I'm facing a similar issue while interfacing esp8266 with d6t mems 44L

Bvgo, 2019-02-06:
What you solution. I am using a nodemcu and omron d6t 04. Kindly share

Thanks

vishalkothari, 2019-06-02:
@a321bhi and @Bvgo , As far as I remember, the issue was a hardware issue and software files attached did work ok.