esp8266/Arduino #2404: NodeMCU v3 ESP8266, ESP-12E error

Ivlad003, 2016-08-17:
hi i can't upload sketch(
please help 
Sketch uses 222,145 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,512 bytes (38%) of dynamic memory, leaving 50,408 bytes for local variables. Maximum is 81,920 bytes.
/Users/admin/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd nodemcu -cb 115200 -cp /dev/cu.wchusbserial1420 -ca 0x00000 -cf /var/folders/7j/lzcnfgm165d8f0f01bkw3klh0000gn/T/buildeb592bb0381bb07ab411ad4680536847.tmp/sketch_aug16b.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.wchusbserial1420
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.wchusbserial1420 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: C4 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: C4 instead of C0
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: C4 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: C4 instead of C0
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: C4 instead of C0
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: C4 instead of C0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

robertfoss, 2016-08-22:
I don't think NodeMCU v3 is made by the NodeMCU team. You may have counterfeit hardware.

Ivlad003, 2016-08-22:
Ok Thank’s)

devyte, 2017-10-14:
The relevant error is documented, as well as the possible causes. Closing.

closed 2017-10-14 by devyte