esp8266/Arduino #2406: WIFI + AVRISP + avrdude: can't upload hex file

daturach, 2016-08-17:
Hello

I am trying to use the sketch Arduino_Wifi_AVRISP, together with avrdude without success.

I have the following setup:
- Windows 7
- IDE 1.6.10
- Wemos D1 mini connected to ICSP on the Arduino Uno. I am using a level converter in between.

Avrdude command:

`avrdude -v -C avrdude.conf -c arduino -p atmega328 -P net:192.168.0.19:328 -U flash:w:Blink.cpp.hex:i`

Result:

```
avrdude: Version 6.3, compiled on Jun 22 2016 at 16:05:21
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "avrdude.conf"

         Using Port                    : net:192.168.0.19:328
         Using Programmer              : arduino
avrdude: ser_drain(): read error: ParamÞtre incorrect.

avrdude: ser_drain(): read error: ParamÞtre incorrect.

avrdude: ser_drain(): read error: ParamÞtre incorrect.

         AVR Part                      : ATmega328
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 4728160
         Firmware Version: 0.2
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.
```

Can you help?

Thanks for your support

Walter

VashTheProgrammer, 2017-04-14:
I have the same issue

Stonedfield, 2017-07-20:
Trying to find a solution for remote programming of an older Atmel AVR based project I had found this ESP8266 based Arduino_Wifi_AVRISP program. It seems not working at all when trying to connect with avrdude from Windows. Indeed there is a communication bug only in the Windows version of avrdude with the port redirect. Even the last error fix list of avrdude states port redirecting should work fine, this is not the case for the Windows version. There are only a few notices found in the Internet about this issue.

Under Linux avrdude works for the flash memory space well but the eeprom area has also big bug. After every 8 bytes the following 8 bytes are skipped in reading and writing. I think responsible for this bug is Arduino_Wifi_AVRISP and not avrdude. Further tests to be done.

devyte, 2017-10-14:
@daturach is this issue still valid with latest git?
@Stonedfield did you test further?

devyte, 2018-01-08:
Closing due to lack of feedback in over 2 months, and due to age.

closed 2018-01-08 by devyte

Stonedfield, 2018-01-08:
I had given it up to use this ESP8266 based Arduino_Wifi_AVRISP programmer as my ESP8266 was not able to run properly over several months. Instead now I am using a wired connection with RS422 line drivers