esp8266/Arduino #2407: Sd2Card.cpp problem in v2.3.0

cgaerter, 2016-08-17:
### Basic Infos

#### Hardware
Hardware:			ESP-12E

### Description

I want to access a SD card using esp8266 library v2.3.0 but card.init returns always false. With the esp8266 library v2.2.0 everything works fine (sketch unchanged, hardware unchanged). I tracked the issue down to Sd2Card.cpp which has changed from 2.2.0 to 2.3.0. Three times the following code has been added:
  _#ifdef ESP8266
  optimistic_yield(10000);
  #endif_
Unfortunately I do not have enough in-depth knowledge to investigate further. When I install the esp8266 library v2.3.0 and replace the Sd2Card.cpp with the one from v2.2.0 (which has not the optimistic_yield instructions) everything works fine.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```cpp

#include <SPI.h>
#include <SD.h>

Sd2Card card;
SdVolume volume;
SdFile root;
const int chipSelect = 16;

void setup() {
  if (!card.init (SPI_HALF_SPEED, chipSelect)) {
    Serial.print ("Initialization failed.\n");
  } else {
    Serial.print ("Initialization ok.\n");
  }
}

void loop() {
  //
}
```

reinaldoaraujo, 2016-08-28:
Hi cgaerter, I'm trying to run a sd card in my esp8266-07, but I'm not having success, I'm connecting:

GPIO-13 - SD / MOSI
GPIO-12 - SD / MISO
GPIO-14 - SD / SCLK
GPIO-16 - SD / CS

you are connecting this way?

cgaerter, 2016-08-29:
Hi reinaldoaraujo,

yes, I can confirm that I use the same connections as you do. But I have an ESP-_**12**_ and it does work with esp8266 library v2._**2**_.0 only and not with the newer esp8266 library v2._**3**_.0!

Greetings, Chris

probonopd, 2017-03-04:
Your sketch is missing `Serial.begin(9600);`.

Should look like:

```
#include <SPI.h>
#include <SD.h>

Sd2Card card;
SdVolume volume;
SdFile root;
const int chipSelect = 16;

void setup() {
  Serial.begin(9600);
  if (!card.init (SPI_HALF_SPEED, chipSelect)) {
    Serial.print ("Initialization failed.\n");
  } else {
    Serial.print ("Initialization ok.\n");
  }
}

void loop() {
  //
}
```

probonopd, 2017-03-04:
* On Arduino 1.6.11 with esp git 889775c I get "Initialization failed." with all but __one__ SD cards.

probonopd, 2017-03-04:
Using the SDWebServer example, http://esp8266sd.local/pins.png `Sent less data than expected!`.

Repeatedly reloading the page makes the ESP reboot.

```
Sent less data than expected!
Fatal exception 28(LoadProhibitedCause):
epc1=0x40203a9f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (28):
epc1=0x40203a9f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0b80 end: 3fff0f40 offset: 01a0
```

probonopd, 2017-03-06:
Turns out that my sd card issues may have been electrical... using a capacitor close to VCC and GND of the SD card does wonders for stability for me.

devyte, 2018-01-08:
Closing due to lack of feedback in over 2 months, and age.

closed 2018-01-08 by devyte