esp8266/Arduino #2409: SecureWifiClient::verify always returning false

stefanbode, 2016-08-18:
Reference: https://github.com/esp8266/Arduino/issues/2125

Connecting to an mosquitto MQTT on port 8333 does nut succeed in verifying the fingerprint. Mosquitto (1.4.9) is running with TSL1.2 verified by "MQTT SPY"

Sketch attached. Using NodeMCU 1.0 as hardware, but "generic ESP" board to simplify error search. Also removed all the PubSubClient stuff to simplify more and make sketch as simple as possible.

### Basic Infos

#### Hardware
Hardware:		ESP-12E (Nodemcu
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>

#define DEBUG_PRINT 1

#define MQTT_TEST_SERVER_HOST "xxxx.homeserver.com"
#define MQTT_TEST_USER "test"
#define MQTT_TEST_PASS "1234"

#define WIFI_SSID "wlan"
#define WIFI_PASSWORD "1234"

// *****************************
const char* ssid = WIFI_SSID;
const char* password = WIFI_PASSWORD;

const char* mqttuser = MQTT_TEST_USER;
const char* mqttpass = MQTT_TEST_PASS;
const char* mqtt_server_host = MQTT_TEST_SERVER_HOST;

//---------------------------
#define IPSET_STATIC { 192, 168, 178, 250 }
#define IPSET_GATEWAY { 192, 168, 178, 1 }
#define IPSET_SUBNET { 255, 255, 255, 0 }
#define IPSET_DNS { 192, 168, 178, 1 }

//
byte ip_static[] = IPSET_STATIC;
byte ip_gateway[] = IPSET_GATEWAY;
byte ip_subnet[] = IPSET_SUBNET;
byte ip_dns[] = IPSET_DNS;
// ****************

// openssl x509 -fingerprint -in  mqttserver.crt 
const char* fingerprint = "7F AE 74 09 5D 20 93 59 B2 E6 61 92 FF 22 AC 17 BB 4F 23 C9";
// also checked with the : between the numbers.

String clientName;

long lastReconnectAttempt = 0;
long lastMsg = 0;
int test_para = 2000;
unsigned long startMills;

//------------------------------
WiFiClientSecure wifiClient;

//----------------------

void verifytls() {
  // Use WiFiClientSecure class to create TLS connection
  Serial.print("connecting to ");
  Serial.println(mqtt_server_host);
  if (!wifiClient.connect(mqtt_server_host, 8883)) {
    Serial.println("connection failed");
    return;
  }

  if (wifiClient.verify(fingerprint, mqtt_server_host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
  }
}
  

void wifi_connect()
{
  if (WiFi.status() != WL_CONNECTED) {
    // WIFI
    if (DEBUG_PRINT) {
      Serial.println();
      Serial.print("===> WIFI ---> Connecting to ");
      Serial.println(ssid);
    }
    delay(10);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    WiFi.config(IPAddress(ip_static), IPAddress(ip_gateway), IPAddress(ip_subnet), IPAddress(ip_dns));

    int Attempt = 0;
    while (WiFi.status() != WL_CONNECTED) {
      if (DEBUG_PRINT) {
        Serial.print(". ");
        Serial.print(Attempt);
      }
      delay(100);
      Attempt++;
      if (Attempt == 150)
      {
        if (DEBUG_PRINT) {
          Serial.println();
          Serial.println("-----> Could not connect to WIFI");
        }
        ESP.restart();
        delay(200);
      }

    }

    if (DEBUG_PRINT) {
      Serial.println();
      Serial.print("===> WiFi connected");
      Serial.print(" ------> IP address: ");
      Serial.println(WiFi.localIP());
    }
  }
}

void setup()
{
  startMills = millis();

  if (DEBUG_PRINT) {
    Serial.begin(115200);
  }

  wifi_connect();

  clientName += "esp8266-";
  clientName += "-";
  clientName += String(micros() & 0xff, 16);
  verifytls();
  
}

void loop()
{

}
`

### Debug Messages
ALL

===> WIFI ---> Connecting to BodeNet_Keller
. 0. 1wifi evt: 0
wifi evt: 3

===> WiFi connected ------> IP address: 192.168.178.250
connecting to elocm.homeserver.com
[hostByName] request IP for: xxxx.homeserver.com
[hostByName] Host: xxx.homeserver.com IP: aaa.bbb.cccc.ddd
certificate doesn't match
pm open,type:2 0

CERT INFO:
pi@raspberrypi:/etc/mosquitto/certs $ openssl x509 -fingerprint -in  server.crt
SHA1 Fingerprint=7F:AE:74:09:5D:20:93:59:B2:E6:61:92:FF:22:AC:17:BB:4F:23:C9
-----BEGIN CERTIFICATE-----
MIIDaDCCAlACCQDcGEFaOdDLtzANBgkqhkiG9w0BAQsFADB2MQswCQYDVQQGEwJE
RTEMMAoGA1UECAwDTlJXMREwDwYDVQQHDAhCRVJHSEVJTTENMAsGA1UECgwEQk9E
RTEUMBIGA1UEAwwLU3RlZmFuIEJvZGUxITAfBgkqhkiG9w0BCQEWEnN0ZWZhbl9i

krzychb, 2016-08-18:
> Mosquitto (1.4.9) is running with TSL1.2

Hi @stefanbode,
WiFiClientSecure does support TLS 1.1. 
TLS 1.2 is not supported (https://github.com/esp8266/Arduino/issues/2298)
Krzysztof

stefanbode, 2016-08-18:
I was wrong. Using TLS1.1

mosquitto conf:
tls_version tlsv1

Additonally the PubSubClient seems to work. also encrypted as far as I can see. Just the "Verify" does not work. I my complete sketch the MQTT is working with the WifiSecureClient on Port 8883 and User/passord. Just also here Verify does not work.

chaeplin, 2016-08-18:
Use certificate CN of server.

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp#L468-L471

#define MQTT_SERVER_CN "mqttserver"
#define MQTT_SERVER { 192, 168, 10, 10 }

  if (!sslclient.connect(mqtt_server, 8883))
~~
  if (sslclient.verify(MQTT_FINGERPRINT, MQTT_SERVER_CN))

https://github.com/chaeplin/esp8266_and_arduino/blob/master/_72-homekit-accessories-Aircon/_72-homekit-accessories-Aircon.ino#L216-L233

stefanbode, 2016-08-19:
Ok, next step to get closer. Anyhow my CN has a space inside. For sure it does not work with this same "does not match". Will generate new server.crt without Space in the name, to be sure nothing goes wrong. What do I need to do to get the DEBUGV messages from the source code to my serial. I do not see any of these regardless what DEBUG option I choose. I think this would help get closer insight of the issue on my side.

stefanbode, 2016-08-19:
SOLVED. I put my name as CN with a space. This did not work. Now all lower case the fqdn of the server and magically it worked

closed 2016-08-19 by stefanbode

SERIDJ, 2020-01-31:
hi  @stefanbode  @stefanbode @chaeplin 

i wand to use WifiClientSecure to send data from ESP32 to Raspbeery server (script) .?