esp8266/Arduino #2410: Am I killing my flash? When will flash wear start happening?

craigfoo, 2016-08-18:
### Basic Infos

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.1.0-rc2

### Description/Problem

I'm using the Adafruit Huzzah which has the ESP-12E module on it. I have a power management circuit that powers the ESP8266 at a given interval. On a daily average, EEPROM is being read from 24 times and written to 12 times. I also read 2 files on every wakeup using SPIFFS 12 times a day for each file. Am I killing the flash? I have 10 Huzzahs doing this and one just crapped out on me. I'm hoping it's just this one but I'm a concerned. 

### Settings in IDE

Module:  			Adafruit Feather Huzzah 
Flash Size: 		4MB
CPU Frequency:	80Mhz

liquidfalcon, 2016-08-18:
Winbond, the manufacturer of the flash chip for the ESP-12 series quotes a write endurance of "More than 100,000 erase/program cycles", so, I doubt you were anywhere close to the wear limits of the flash. If you're really that concerned, try and push everything onto SPIFFS instead of the pseudo-EEPROM implementation, as SPIFFS is wear leveled.

craigfoo, 2016-08-18:
@liquidfalcon, that's the perfect answer. I was heading that way (exclusively using SPIFFS) already.

WereCatf, 2016-08-18:
In addition to what was said above, reading Flash doesn't wear it out, only writing to it does.

devyte, 2017-10-14:
Chinese modules with cheapo flash chips have lower write cycles, i.e.: 10000, but that still gives you over 2 years or so (assuming my sloppy math is correct).
Closing.

closed 2017-10-14 by devyte