esp8266/Arduino #2412: Home WiFi randomly switches to NodeMCU IP

balsimpson, 2016-08-19:
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

I am using a NodeMCU to do send and receive some MQTT messages from my accelerometer to my webserver. Trying to parse the MQTT data has been a nightmare but that's not the problem am facing. The problem is that my home WiFi automatically switches from my regular router port (192.168.10.1) to the NodeMCU port (192.168.4.1).

Am using Arduino IDE to code and flash the NodeMCU and in the code apart from starting the wifi I cannot see anything that would make it do this. Am I missing something?

### Settings in IDE

Module:  			NodeMCU v1
CPU Frequency:	80Mhz?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <PubSubClient.h>

// WiFi details
const char* ssid = "ssid";
const char* password = "password";

// cloudMQTT details
const char* mqtt_server = "m12.cloudmqtt.com";
int port = 18256;
const char* mqtt_user = "user";
const char* mqtt_pass = "password";

// Initialize
WiFiClient espClient;
PubSubClient client(espClient);

// MQTT message details
long lastMsg = 0;
char msg[128];
int value = 0;

void setup() {
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, port);
  client.setCallback(callback);
}

void setup_wifi() {
  delay(10);
  // Connect to WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

String buildJson() {
  String data = "{";
  data += "x:";
  data += (int)random(100);
  data += ",";
  data += "y:";
  data += (int)random(100);
  data += ",";
  data += "z:";
  data += (int)random(100);
  data += "}";
  return data;
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Connecting to cloudMQTT...");
    // Attempt to connect
    if (client.connect("something1", mqtt_user, mqtt_pass)) {
      Serial.println("connected");
      // Publish a state...
      client.publish("test", "fish connected");
      // subscribe to your topics
      client.subscribe("my/test1");
      client.subscribe("my/test");
      client.subscribe("my/outTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}
void loop() {

  if (!client.connected()) {
    reconnect();
  } else {
    long now = millis();
    if (now - lastMsg > 2000) {
      lastMsg = now;

      String json = buildJson();
      char jsonStr[25];
      json.toCharArray(jsonStr, 25);
      
      Serial.println(json);
      client.publish("my/test", jsonStr);
      client.publish("my/test", "17");
    }
  }
  client.loop();
}
```

### Debug Messages

```
messages here
```

WereCatf, 2016-08-19:
Are you deliberately using your Nodemcu as a WIFI AP+STA? By default it acts as both an AP and STA, which you may not have realized. If you set it to only work as a WIFI STA then your devices wouldn't try to connect to it. You can set it to STA by calling WiFi.mode(WIFI_STA); before WiFi.begin();

krzychb, 2016-08-19:
Another option, if you need NodeMCU running both soft-AP and STA mode, would be to use the option like "Forget this network" in any device / PC that is connecting to 192.168.4.1 instead of your regular router port (192.168.10.1). I.e. forget 192.168.4.1.

balsimpson, 2016-08-19:
Thanks for the replies. I'll try both and report back.

WereCatf, 2016-08-19:
@krzychb That's not how WiFi works; devices don't store WiFi-settings based on IP-addresses, they store them based on the SSID of the AP, as they obviously cannot even know what IP-address they're going to get until *after* they've already connected. @balsimpson has the issue where his Nodemcu is running as both an AP and a STA, but it's advertising the same SSID and his home-network does and uses the same password, and that's why his other devices randomly connect to it -- they cannot know which of the two SSID's is the "correct" one when they both have the same name and password.

He either has to change the SSID the Nodemcu advertises, or disable the AP on it and use STA-mode only.

balsimpson, 2016-08-20:
The easiest way I found was to add WiFi.mode(WIFI_STA) before starting your wifi setup. thanks for the answers.

devyte, 2017-10-14:
Per previous comment, closing as resolved.

closed 2017-10-14 by devyte