esp8266/Arduino #2414: ESP-12E troubleshooting (rst cause:2, boot mode:(3,6))

rkoptev, 2016-08-20:
### Basic Infos

#### Hardware
Hardware:		ESP-12E (ESP8266MOD)
Core Version:      	2.3.0

### Description

Hi. Previously, I was able to flash ESP through Arduino IDE. The code has worked, although very often controller restarted with no reason. "... wdt reset ..." has been written to the COM port. I tried to turn off the watch dog timer with the help of ESP.wdtDisable() or simply to increase the timeout with ESP.wdtEnable(65535). This partly solves the problem, but restarts still have a place to be.
But now, after a regular flashing I get this output in COM port:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

And that's all. I tried to change the module in the IDE to "ESP8266 Generic", and to change the size of the memory, but no result. What could be the problem? And how to fix this behavior?

### Settings in IDE

Module:  			NodeMCU 1.0
Flash Size: 		4MB/3MB
CPU Frequency:	80Mhz
Upload Using:		SERIAL

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <FS.h>

#define DEV true

#define STEPPER_EN_PIN   13
#define STEPPER_DIR_PIN  2
#define STEPPER_STEP_PIN 5

#define STEPPER_RPM 600
#define STEPPER_GEAR_RATIO 64
#define STEPPER_MICROSTEP 16
#define STEPPER_FULL_ROTATION 7000

#define SERVER_URL "rkoptev.shpp.me"
#define SERVER_PORT 8485
#define SERVER_CONNECTION_INTERVAL 10000

#define WIFI_DATA_FILENAME "/wifi.txt"
#define WIFI_CONNECTION_TIMEOUT 10000

ESP8266WebServer server(80);

char* ssid = new char[32];
char* password = new char[64];

bool jalousieState;

void moveStepper(unsigned long steps, bool direction) {
  unsigned long period = 60000000L / (STEPPER_RPM * STEPPER_GEAR_RATIO * STEPPER_MICROSTEP * 2UL);
  steps *= STEPPER_GEAR_RATIO;
  Serial.print("Period = ");
  Serial.println(period);
  digitalWrite(STEPPER_DIR_PIN, direction);
  digitalWrite(STEPPER_EN_PIN, LOW);
  for (int i = 0; i < steps; i++) {
    digitalWrite(STEPPER_STEP_PIN, HIGH);
    delayMicroseconds(period);
    digitalWrite(STEPPER_STEP_PIN, LOW);
    delayMicroseconds(period);
    ESP.wdtFeed();
  }
  digitalWrite(STEPPER_EN_PIN, HIGH);
}

void jalousieClose() {
  if (!jalousieState)
    return;
  Serial.println("Closing jalousie");
  moveStepper(STEPPER_FULL_ROTATION, 1);
  jalousieState = false;
}

void jalousieOpen() {
  if (jalousieState)
    return;
  Serial.println("Opening jalousie");
  moveStepper(STEPPER_FULL_ROTATION, 0);
  jalousieState = true;
}

void jalousieHome() {
  moveStepper(STEPPER_FULL_ROTATION, 1);
  jalousieState = false;
}

void handleWiFiCredentials() {
  if (!server.hasArg("SSID") || !server.hasArg("password")) { server.send(500, "text/plain", "Bag arguments"); return; }

  Serial.print("WiFi Credentials:\nSSID: ");
  Serial.println(server.arg("SSID"));
  Serial.print("Password: ");
  Serial.println(server.arg("password"));

  File wifiData = SPIFFS.open(WIFI_DATA_FILENAME, "w");
  wifiData.println(server.arg("SSID"));
  wifiData.println(server.arg("password"));
  wifiData.close();

  server.send(200, "text/plain", "Data saved");
  ESP.reset();
}

bool loadWiFiCredentials() {
  File wifiData = SPIFFS.open(WIFI_DATA_FILENAME, "r");
  if (!wifiData)
    return false;
  
  String _ssid = wifiData.readStringUntil('\n');
  _ssid.trim();
  _ssid.toCharArray(ssid, 32);
  
  String _password = wifiData.readStringUntil('\n');
  _password.trim();
  _password.toCharArray(password, 64);
  
  wifiData.close();
  return true;
}

void setup() {
  // Make the watch dog timeout longer
  ESP.wdtEnable(65535);
  
  Serial.begin(74880);
  Serial.setDebugOutput(DEV);

  pinMode(STEPPER_EN_PIN, OUTPUT);
  pinMode(STEPPER_DIR_PIN, OUTPUT);
  pinMode(STEPPER_STEP_PIN, OUTPUT);
  digitalWrite(STEPPER_EN_PIN, HIGH);

  // Initialize filesystem
  SPIFFS.begin();

  bool loaded = loadWiFiCredentials();
  if (loaded) {
    Serial.println("WiFi credentials loaded");
    unsigned long connectionTime = millis();
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
      delay(100);
      if (millis() - connectionTime > WIFI_CONNECTION_TIMEOUT) {
        Serial.println("Connection failed. Ereasing old Wifi credentials and starting AP to get new credentials.");
        SPIFFS.remove(WIFI_DATA_FILENAME);
        ESP.reset();
      }
    }
  } else {
    Serial.println("WiFi credentials not found, starting AP");

    WiFi.mode(WIFI_AP);
    WiFi.softAP("Pentagon228");
    
    server.serveStatic("/", SPIFFS, "/public_html/WifiSetup.html");
    server.serveStatic("/WifiSetup.css", SPIFFS, "/public_html/WifiSetup.css");
    server.serveStatic("/WifiSetup.js", SPIFFS, "/public_html/WifiSetup.jsl");
    server.on("/submit", HTTP_POST, handleWiFiCredentials);
    
    server.begin();

    while(1) {
      server.handleClient();
      ESP.wdtFeed();
    }
  }
  jalousieHome();
}

void loop() {
  delay(SERVER_CONNECTION_INTERVAL);
  Serial.println("I'm alive");
  
  WiFiClient client;
  if (!client.connect(SERVER_URL, SERVER_PORT))
    return;
  
  client.print(String("GET / HTTP/1.1\r\n") +
                      "Host:" + SERVER_URL + ":" + SERVER_PORT + "\r\n" + 
                      "Connection: close\r\n\r\n");
  unsigned long timeout = millis();
  while (client.available() == 0) {
    ESP.wdtFeed();
    delay(100);
    if (millis() - timeout > 5000) {
      client.stop();
      return;
    }
  }
  
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.println("Res: " + line);
    line.trim();
    if (line.equals(String("open"))) {
      jalousieOpen();
      return;
    } else if (line.equals(String("close"))) {
      jalousieClose();
      return;
    }
  }
}

```

WereCatf, 2016-08-20:
Have you connected some new devices to GPIO 0, 2 or 15? They control the boot-mode of the ESP and if some device you've connected pulls the wrong pin high or low it'll change boot-mode from normal to something else. See https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup

rkoptev, 2016-08-20:
I fundamentally don't use these pins in the project. Pin 15 is pulled down, pin 2 is pulled up, in normal mode pin 0 is pulled up, when I flash new firmware, I connect pin 2 to GND. There are no any devices connected to this pins.

WereCatf, 2016-08-20:
Well, the obvious next step would be to flash the simple "blink LED" - example sketch and see if it works -- you don't mention if you've tried that or not.

rkoptev, 2016-08-20:
I flashed Blink example and having the same output:

supersjimmie, 2016-08-22:
Did you run blink with the other hardware connected or everything disconnected?
And to what pin did you connect the LED (for blink)?

rkoptev, 2016-08-22:
I have some news regarding to this issue: the module has started to work, but very unstable. Ie yesterday it has started and worked full day. Then this error again. It seems that it occurs at all accidental. The most important thing is that I do not change a sketch and wiring, everything remains as before. It's like some kind of magic ((

@supersjimmie I tried to flash Blink example with disconnected peripheral. And in addition to described ablove message it gives me stack trace:
```
0x40216d48: tx_pwr_backoff at ?? line ?
0x40211771: set_txcap_reg at ?? line ?
0x402135d4: tx_pwctrl_init_cal at ?? line ?
0x40211771: set_txcap_reg at ?? line ?
0x40213b40: tx_pwctrl_init at ?? line ?
0x402153a4: chip_v6_initialize_bb at ?? line ?
0x402153b9: chip_v6_initialize_bb at ?? line ?
0x40100574: malloc at C:\Users\Ruslan Koptiev\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x40210f77: phy_init at ?? line ?
0x40101fea: register_phy_ops at ?? line ?
0x4021615a: register_chipv6_phy at ?? line ?
0x4020498e: wifi_station_set_default_hostname at ?? line ?
0x40201789: __wrap_register_chipv6_phy at C:\Users\Ruslan Koptiev\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_phy.c line 261
0x4022e340: rijndaelKeySetupEnc at ?? line ?
0x40101b1a: spi_flash_read at ?? line ?
0x40205d57: system_get_sdk_version at ?? line ?
0x402063e4: user_uart_wait_tx_fifo_empty at ?? line ?
0x402063bf: user_uart_wait_tx_fifo_empty at ?? line ?
0x4010171c: wdt_feed at ?? line ?
0x402241ed: ieee80211_node_pwrsave at ?? line ?
0x4010020c: umm_assimilate_up at C:\Users\Ruslan Koptiev\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1180
0x4010068c: _umm_realloc at C:\Users\Ruslan Koptiev\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1584
:  (inlined by) realloc at C:\Users\Ruslan Koptiev\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4020bd00: tcp_send_fin at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 131
0x4020a638: pbuf_alloc at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/pbuf.c line 329
0x4020c111: tcp_output at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 1025
0x4020ca70: udp_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/udp.c line 269

```
But that's just matter of chance: I can get this stack trace even with peripherals connected, and can not get even when it's disconnected.

Believe me, I tried different power supplies, testing them on an oscilloscope, added a capacitor 600 uF as near as possible to the board and I tried absolutely all combinations of pins 0, 2 and 15. I even replaced the board to the other. All to no avail. I'm in a state of frustration.
I will be very grateful for any hint or tip, as I spent on the solution of this problem for over a week.

RudyFiero, 2016-08-22:
What voltage regulator are you using? How far from the regulator to the ESP module? What are you using for interconnections?

I have found that capacitance alone does not deal with the dynamic current requirements. I suspect that you are having power issues.

rkoptev, 2016-08-22:
ok, in order:
I use LM2596 DC voltage regulator tuned to 3.3v. Datasheet says that it is able to provide a current of 3A. 
Also I use DC power adapter from old radio phone. It's provide 7.5V at 300mA.
For interconnections I use standart breadboard wires.

I tried to replace the power supply, but nothing has changed. Then, I tried to configure the voltage regulator to 3.7V, and ESP started! I do not know for how long, but if this problem is over, I'll write about it here.
Thank you for the advice, @RudyFiero .

supersjimmie, 2016-08-23:
I think that the 300mA from your DC power adapter is really "on the edge"...

RudyFiero, 2016-08-23:
I was a little concerned about that. But with a switching supply the current available at the output would be higher. The switcher acts more like a transformer and does a power conversion. A linear regulator just consumes the voltage difference and does not give any current gain.

WereCatf, 2016-08-23:
7.5V at 300mA would equal 2.25W of power and just assuming roughly a conversion-efficiency of 90% that'd result in 3.3V and ~600mA max current, if I understand things right. I think I read somewhere that the ESP8266 can peak at a little over 200mA, so that should still be plenty for it. That does make me wonder why his ESP would only turn on at 3.7V, not 3.3V -- there must be something else at play.

rkoptev, 2016-08-23:
@supersjimmie, it seems like you are right. I replaced this power supply with 24v@500mA and ESP works corectly and starts every time. It was extremely difficult to determine that the problem is in power. I am very astonished, because this chip is positioned as super energy efficient, capable of operating from a small CR2032 battary for a years.
In any case, I think that this problem can be considered solved. Thank you all for your help.

closed 2016-08-23 by rkoptev

architmuchhal12, 2017-06-06:
I am having the same problem when I connect the Adafruit GSM FONA808. But it works fine when I run other programs. 
Please suggest me a solution.
Thanks

kllsamui, 2017-07-27:
i see same when using NODE MCU 0.9 on USB of Raspberry Pi 3B and upload by (Raspbian OS) Arduino IDE 1.8.3.
( same code upload by PC is OK )

kamrulroky, 2017-10-05:
@architmuchhal12 
Have you figured out what is the solution? I am facing the same issue with SIM800l+Nodemcu to connecting adafruit MQTT

2000jago, 2017-12-09:
I was also having this issue powering my device (wemos D1 Mini) directly from an 18650.  It seems that the problem was that as the battery runs down it creates a power problem as RudyFiero above correctly suspected.  Replacing the single 18650 with a "powerbank" that puts out a constant 5V solved the problem for me immediately.

diraniyoussef, 2017-12-23:
I had the same error 
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
⸮
but it happened in a specific place of code, it's when I close the client socket, the NodeMCU being a server. It happens when the client closes from NodeMCU's side or from the client's side.

KlausNZ, 2018-01-09:
Hi folks,

I encountered the same issue with boot modes(1,6) and (3,6).
Development went perfectly fine and all was running smooth at 2am.
Revisiting the project the next morning all went to custard.

Finally I found the culprit and the cause for having the effect with the boot modes.

The exact reason for the continues boot mode errors shown I can;t give, but in my project it was some code failing in a class initialization of a global instance which runs before setup() is called.
Took me a while to figure out, but I got there.

So my recommendation if you encounter the same issue:
Look to your global class instances and see if code is causing exceptions before setup() is called!

Cheers,
Klaus

devyte, 2018-01-10:
@KlausNZ you can't have global classes that on construction depend on on other global classes being already constructed. The order of construction of global classes is non-deterministic in C++ (it depends on the order that Translation Units are linked, which can't be controlled).

fahrvergnuugen, 2018-01-20:
I'm having this same issue. ESP-12E on a Mini dev board. I'm trying to upload Blink Sketch, getting `rst cause:2, boot mode:(3,6)`

My board is powered by USBC (thunderbolt macbook pro) so I don't think it's a power supply issue... 
I have other identical boards that flash fine, and I have flashed this one a dozen times, now all of a sudden it won't flash. I don't have anything physically wired to the board.

Any other suggestions?

bkrajendra, 2018-01-20:
Something weird is happening recently.
All our ESP boards stopped working
Getting Blank after flashing program. Nothing happens. After checking by changing baud rate to 74880
this is what we getting
```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xef
csum 0xef
csum err
ets_main.c 
```

gbetous, 2018-01-27:
Hi,

I bought a few D1 Mini Lite, and every one is only responding with uart @74880 and this message : 

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c 

I tried to flash NodeMCU, I have no errors on esptool, but nothing change : I still have this issue, and NodeMCU is not loaded

damo1023, 2018-01-30:
> Something weird is happening recently.
 All our ESP boards stopped working

Also my boards.... tried different ESP's, power supplies, etc...  Looks like there might be a virus in IDE?

Any success solving this?

matiascolon, 2018-01-31:
i have same problem

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xef
csum 0xef
csum err
ets_main.c

WereCatf, 2018-01-31:
@damo1023 I have one board that used to work in QOUT-mode, but all of sudden it now only works in DIO-mode, so I would recommend you try changing the board to a generic ESP8266, then test if changing the flash-mode to something else works. (DOUT-mode is the slowest and most compatible one, so I'd start with that.) If it does, then I suppose the issue needs further investigation.

tstepnia, 2018-02-09:
In my case cause of resets and exceptions was FT232 TX line to ESP. 
Now after program upload I disconnect it, reset ESP leaving RX connection only to see debug output.

Chriss217, 2018-03-07:
I have a UNO+WiFi R3 ATmega328P+ESP8266 32Mb board (with 8 dip switches). I tried many times to flash the firmware without success using the esptool.py. By setting sw7 to off, running miniterm.py and pressing reset button I received the same message as matiascolon e.g "ets Jan 8 2013, rst, cause:2, boot modes:(3,6)" followed by the other lines and stopping at "ets_main.c".
Following the info on "https://www.bountysource.com/issues/47667636-esp-link-on-a-robotdyn-atmega2560-esp8266" I d/l esp-link v3.0.14 and reflashed the firmware. This failed as before until I added the option "-fm dio". Now when I connected via miniterm.py and hit RST I see the result:
--------------------
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

rf cal sector: 128
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.0.0(5a875ba) S <binary chars>
------------------
The sequence to flash it was:
SW7 on
Reset
run "esptool.py -p /dev/ttyUSB0 erase_flash"
Reset
run "esptool.py -p /dev/ttyUSB0 write_flash -fm dio -fs 32m -ff 80m 0x00000 boot_v1.6.bin 0x01000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin"
Set SW7 off
Reset.

Using my phone I can now (at last see) a wifi station called "ESP_809A70".
I haven't tried any further tests so far, but thought I would post these results in case it helps others progress. Best. Chris

Chriss217, 2018-03-07:
Apologies for the text in bold. This was not intended and I can't unbold it.

GrantGMiller, 2018-04-12:
I was having the same issue with a LOL1n NodeMCU board purchased from a Aliexpress.
I was able to fix the issue by adding the "flash_mode" parameter when loading micropython firmware using esptool.

Here is the command I used on my windows machine to load firmware:

esptool --port COM7 --baud 460800 write_flash --flash_mode dio --flash_size=detect 0 C:/Users/public/documents/esp8266-20170612-v1.9.1.bin

Hope the helps someone in the future :-)

bkrajendra, 2018-04-13:
Be aware there are many ESP 12 variants in market, that looks like to be fake or having something changed in terms of hardware.
Recently i got an ESP12E model with completely plain top metal cover without any logo or any text, not even CE approved symbol.

tried to flash it but its not responding.

[image]

matiascolon, 2018-04-13:
I had the same problem with the DOIT brand, I spent a lot of time searching the internet and trying more esp from the same brand and none worked, then I asked for the refund and the company where I bought them gave me the refund.
I recommend AI THINKER or nodeMCU brand AMICA.

fred2088, 2018-06-22:
have the same problem with 3 pcs of esp8266, is there any solution ?

WereCatf, 2018-06-25:
@jozles Try with setting flash-mode to DOUT.

fred2088, 2018-06-25:
Find the problem, since I am new to esp8266, don't know bad program will cause rest loop of hardware ...., Download another program solve the problem ...

jozles, 2018-06-25:
Incredible...

slept 6 hours and... it works ; didnt touch anything !

@WereCatf thx, flash mode is still QIO ... I'll try if pb again

MikaelBrenner, 2018-08-16:
I had the same problem. It was caused by using `delayMicroseconds()` many times consecutively.
 Apparently, `delayMicroseconds` interferes with the watchdog, preventing the board of working correctly.

hassin23ayz, 2018-11-01:
Hi, 
I have solved the problem by 
Tools > Erase Flash > All contents 
compiled and uploaded again 
the boot problem did not occur  anymore ... until now !

gnzg, 2018-11-18:
Guys, my ESP01 crashed constantly (and randomly). Solved by prepping it to an LM1117 voltage regulator. Schematics online.

w8an, 2018-12-16:
I've found that when I have trouble flashing at high speed, it will work when I set the baud rate to 74880 and use DOUT mode.

BobGag, 2018-12-18:
Read the comments/suggestions above. Tried one of the suggestions to Erase Flash: All Flash Contents
Working fine now.. so far.

madzur, 2018-12-26:
> @damo1023 I have one board that used to work in QOUT-mode, but all of sudden it now only works in DIO-mode, so I would recommend you try changing the board to a generic ESP8266, then test if [...]

Awesome tip man. I thought a bunch of ESP01s I had bought were toast. This helped me save all of them. Now they can be programmed and are all working. I’m setting off the fireworks after trying 2 months to get this working!!! :):)

JamesNewton, 2019-02-28:
Another vote for power supply causing this: In our case, soldering a 2200uF cap across the 3.3 volt line managed it. Actually running at 3.2 volts, but very stable now.

ferm10n, 2019-03-06:
I just want to throw a note in here, I was having the same issue and I believe it was caused by a watchdog timer or something. Initially I had a loop function with no delays. the device would reboot about every 5 or 6 seconds.
Adding in a `delay(0)` in the loop corrected the issue. Removing it brings it back.

GHorneczki, 2019-03-16:
> Another vote for power supply causing this: In our case, soldering a 2200uF cap across the 3.3 volt line managed it. Actually running at 3.2 volts, but very stable now.

Thanks! This tip solved my issue too. I put a 1000uF to 3.3V and no reboots or freeze anymore.

Nkolev06, 2019-03-27:
I have same error

Septikos94, 2019-04-13:
> @damo1023 I have one board that used to work in QOUT-mode, but all of sudden it now only works in DIO-mode, so I would recommend you try changing the board to a generic ESP8266, then test if [...]

This worked for me! By default, flash mode DIO is selected in the Arduino IDE. I have an ESP-01S, so change flash mode to DOUT and just worked great. Thanks.

LanFly, 2019-05-23:
> _edit: reformat_
> 
> I have a UNO+WiFi R3 ATmega328P+ESP8266 32Mb board (with 8 dip switches). I tried many times to flash the firmware without success using the esptool.py. By setting sw7 to off, running miniterm.py and pressing reset button I received the same message as matiascolon e.g "ets Jan 8 2013, rst, cause:2, boot modes:(3,6)" followed by the other lines and stopping at "ets_main.c".
> 
> Following the info on "https://www.bountysource.com/issues/47667636-esp-link-on-a-robotdyn-atmega2560-esp8266" I d/l esp-link v3.0.14 and reflashed the firmware. This failed as before until I added the option "-fm dio". Now when I connected via miniterm.py and hit RST I see the result:
> 
> ```
>  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> load 0x40100000, len 2408, room 16 
> tail 8
> chksum 0xe5
> load 0x3ffe8000, len 776, room 0 
> tail 8
> chksum 0x84
> load 0x3ffe8310, len 632, room 0 
> tail 8
> chksum 0xd8
> csum 0xd8
> 
> 2nd boot version : 1.6
>   SPI Speed      : 80MHz
>   SPI Mode       : DIO
>   SPI Flash Size & Map: 32Mbit(512KB+512KB)
> jump to run user1 @ 1000
> 
> rf cal sector: 128
> rf[112] : 00
> rf[113] : 00
> rf[114] : 01
> 
> SDK ver: 2.0.0(5a875ba) S <binary chars>
> ```
> 
> The sequence to flash it was:
> 
> ```
> SW7 on
> Reset
> run "esptool.py -p /dev/ttyUSB0 erase_flash"
> Reset
> run "esptool.py -p /dev/ttyUSB0 write_flash -fm dio -fs 32m -ff 80m 0x00000 boot_v1.6.bin 0x01000 user1.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin"
> Set SW7 off
> Reset.
> ```
> 
> Using my phone I can now (at last see) a wifi station called "ESP_809A70".
> I haven't tried any further tests so far, but thought I would post these results in case it helps others progress. Best. Chris

Hi, I recently encountered this problem when designing open source hardware.

`ets Jan  8 2013,rst cause:2, boot mode:(3,7)`

The reason for this problem is that the current of the power supply is insufficient.

`Solution：At least 3.3V 300mA of power supply.`

I hope this will help you.

molsondry, 2019-07-15:
Hi guys,
I had the same problem: after 3 or 4 connection fails the system restarted with the codes mentioned above. I just moved the Wifi connection routine to the beginning of the setup routine. Before I started two ISRs before starting the Wifi. Now it works.

void setup() {
  // put your setup code here, to run once:
  // Console parameters
  Serial.begin(115200);

 // Wifi startup, should be here in the beginning of setup, further down it caused periodically restarts
  connectWiFi();

  // Capacitive Sensor for detecting door open-close/ 
  // Oscillator configurations
  pinMode(oscillatorPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterru........................

void volatile connectWiFi (){
  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  /* Explicitly set the ESP8266 to be a WiFi-client, otherwise, it by default,
     would try to act as both a client and an access-point and could cause
     network-issues with your other WiFi-devices on your WiFi-network. */
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Try and create TCP connections
  /*
  if (!sta.connect("heise.de", 80)) {
    Serial.println("connection failed");
    delay(5000);
    return;
  }
  */

naorellana, 2019-07-29:
i started the connection to wifi again before to use the http.begin(). it worked for me

omerfraz, 2019-08-01:
In my case when i removed yield() from the loop ... the problem was solved

theicfire, 2019-08-16:
Some details I found:
1) `rst cause:2, boot mode:(3,6)` happens on startup for all my ESP8266 devices. (WeMos D1, NodeMCU v1.0). I didn't realize this was the case because at startup by default the baud rate is 74880. However, after this message prints everything works fine
2) I was finding that sometimes my devices would continually boot loop with this message. I haven't found the root cause, but I had 3 devices on at the same time and the third one was always boot looping. Very odd, but probably just my own problem

quanngovan98, 2019-09-22:
> Hi,
> I have solved the problem by
> Tools > Erase Flash > All contents
> compiled and uploaded again
> the boot problem did not occur anymore ... until now !

thank you, it works greatly!!

dr3g0ng44m, 2019-09-29:
I dont think anyone get in rediculous situation as me, but if you have #define<iostream> in cpp or header file, lets remove it first

muito93, 2019-10-09:
> Hi,
> I have solved the problem by
> Tools > Erase Flash > All contents
> compiled and uploaded again
> the boot problem did not occur anymore ... until now !

@hassin23ayz  It worked, thank you

shubham-kulkarni, 2019-10-18:
I am facing the same issue. I accidentally pressed "reset" button on the board while uploading was in progress (nearly 60%). Now when I am connecting the board to my PC, its detected. I am also able to upload the sketch to the board. But its misbehaving. Inbuilt LED is constantly flashing and I see the mentioned error on serial screen (Baud Rate 115200 / NL&CR).

vivekanandRdhakane, 2019-12-03:
> 
> 
> Hi,
> I have solved the problem by
> Tools > Erase Flash > All contents
> compiled and uploaded again
> the boot problem did not occur anymore ... until now !

thanks. It worked

faelpinho, 2020-01-03:
Have no solution?
I got only this message when GPIO_0 is GND:
" ets Jan  8 2013,rst cause:2, boot mode:(3,6)

ets_main.c "

Nothing more. 
And I got this message when GPIO_0 is VCC:
" ets Jan  8 2013,rst cause:2, boot mode:(1,7)"
Nothing more.

I have clicked "Erase" button on ESP8266 Download Tool V.3.6.8, and the prompt shows this message:
CONNECT BAUD: 115200
==========
.Uploading stub...
Running stub...
Stub running...
------------------
crc_efuse_4bit: 0
crc_calc_4bit: 8
48bit mac
"

[image]

bbombb-bbombb, 2020-04-01:
Hi,
I solve this problem when download older version of the esp8266 to version 2.5.0 and everything works again.

skybadger, 2020-04-27:
I am using ESP8266-12 and 2.6.3 core. 
I had this problem and walked back to code, sorted out some bad variables ( clash with internal libraries vs local include settings, turned off pre-cache to ensure libraries got rebuilt and fixed my pins allocation - I was forgetting pins are not the same as GPIOs and it all works. 

Oh, and having downloaded the GH latest head ( v4.1) , I still needed to add the ICACHE RA ATTR in front of all the ISR functions. 
But it is working and with interrupts too. 
Very nice. Took about 3 hours of faff when it should have taken 5 mins but that's hardware.

k-byte-rgb, 2020-07-26:
> Find the problem, since I am new to esp8266, don't know bad program will cause rest loop of hardware ...., Download another program solve the problem ...

 which programmer?

[12 more replies]