esp8266/Arduino #2415: ESP8266 After BasicOTA OTA network port not showing up

SrikanthBala, 2016-08-21:
----------------------------- Remove above -----------------------------

### Basic Infos

Hardware:		ESP-12 on Node MCU Tried V0.9 and V1.0 (12E)
Core Version:      	2.3.0
Arduino IDE version 1.6.11

As per instructions (latest) for OTA I have been attempting to set it up for a week without success. Tried BasicOTA. Flashes well via serial port. Shows IP address. I can ping the port and get a reply. But no OTA port seen. I switched off my Windows firewall (Using Windows 10), without effect. I checked my Router settings (Netgear N600). Nothing seems to be set to block anything, but being a novice, I may be missing something here.
Still no success. The port just does not show up. Tried with another PC and another router (TP Link) with no success. One more thing, when looking at the verbose output whilst programming I don't find python 2.7 being used anywhere. It says something about espttool. Why do we need to download python 2.7 then ? Am I missing something ? Do I need a different firmware on my ESP to support OTA ? Also I don't think either of the NodeMCU 0.9 or 1.0 are origionals. They seem to be clones. However all other sketches work and I tried a plain ESP8266 12 module too using FTDI for programming without OTA success. Please help.

Module:  			ESP8266 12 and 12 E on Node MCU 0.9 and 1.0
Flash Size: 		4MB/1MB
CPU Frequency:	80Mhz
Flash Frequency:	40Mhz?
Upload Using:		OTA Not possible / SERIAL 115200
Reset Method:		nodemcu Tried V0.9 and 1.0

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

krzychb, 2016-08-21:
Hi @SrikanthBala,

Regarding your particular concerns:

> Why do we need to download python 2.7 then ? 

Python is required to load the module over the OTA port. Since you do not see the port, you could not use / see it in action yet.

> I don't think either of the NodeMCU 0.9 or 1.0 are origionals

If you are able to load these modules using serial port and they operate correctly, including [HelloServer.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino) sketch, then OTA will work as well. Please verify if this sketch is working for you.

> Do I need a different firmware on my ESP to support OTA ?

No, you do not. Proceed according to latest [OTA documentation](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md), use latest Arduino IDE and [stable esp8266 / Arduino](https://github.com/esp8266/Arduino#stable-version-) core. I have just loaded Arduino 1.6.11 portable and [BasicOTA.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino) is working for me. Network port is showing up and I am able to load the module. I am using Windows 7. Later I will try my other PC with Windows 10.

Basing on you description you have already made the standard checks (tried another PC, tried another router, disabled firewall and successfully ping the module). I propose three actions:

1. Check basic operation of your module in your network using [HelloServer.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino) sketch.

2. There seems to be something with your network that is preventing the OTA port to show up. Maybe this is with Arduino IDE installation / Java.  If not done already, try portable installation [Windows ZIP file for non admin install](https://www.arduino.cc/download_handler.php?f=/arduino-1.6.11-windows.zip).

3. If problem is related to your network, then check it from another angle by trying  [OTA with a web browser](https://github.com/esp8266/Arduino/tree/master/doc/ota_updates#web-browser). Use the latest Aduino IDE 1.6.11 and esp8266 / Arduino core 2.3.0.

Krzysztof

SrikanthBala, 2016-08-21:
At the beginning I thank you very much for a prompt and detailed reply. My problem is still not solved but I see the following -
1. The HelloServer.ino sketch seems to work over serial and when I type the URL on my browser I get 'hello from esp8266!' as expected. So far so good
2. I did not yet try the portable install as I had Arduino IDE 1.6.9 before and as OTA did not work I uninstalled it and reinstalled the IDE 1.6.11 and reinstalled the ESp8266 through Boards manager which I think would have downloaded the latest version. So would this qualify as a clean install of IDE ?
3.I tried the OTA with a web browser. I did not install the http://www.apple.com/support/bonjour/  as I have a Windows machine and though the web page indicated says it is needed for Windows I though this may be a type ?
ON uploading the Here I hit upon two problems. One updating the example ESP8266HTTPUpdateServer.ino sketch, I could see the serial output asking me to log on as http://esp8266-webupdate.local/update but I could not as my browser said it couldn't reach the page. However I could log on using 192.168.1.2/update and the web page showed a button for 'Browse' and 'Update'. I could load a file for update and the update process (Here your instructions website mentions that this happens if  'host software installed in step 2 does not work'. I'm not sure I understand that as the Serial output on installing the sketch seems as expected).

Ok. That said, when I tried to search for a file "C:\-----------\WebUpdater.cpp.bin using the path provided by the verbose output, I could not find the file but instead found WebUpdater.ino.bin no 'cpp' file. Does that indicate some setting problem on my Arduino IDE ? 
Anyway I uploaded the WebUpdater.ino.bin using the Upload button and it flashed (blinking on my ESP12) but the Serial output remained blank and after a few secs said wdt reset. Since the watchdog has reset, probably I uploaded some junk instead of the code.
This is till where I have reached. As I said earlier I have not tried the Arduino Non Admin portable install. I will probably try that whilst waiting for a reply.

Once again my sincere thanks for the support I am receiving

SrikanthBala, 2016-08-21:
One more thing. My Install directory in Preferences shows as C:\Users\********\Documents\Arduino but I don't see my ESP2866 directory there under libraries. I don't know if this means anything. Its probably someplace else as my HelloServer and several other ESP sketches work. In fact OTA was the first sketch which failed. FTP server, SD card server all work well. Thanks.

krzychb, 2016-08-21:
1. HelloServer.ino works indeed :+1:

  2. Try it. It is quick and easy. The package is so you do not need to install Arduino IDE. After downloading you need to unzip `arduino-1.6.11` folder and just run `arduino.exe`. You may need to reinstall esp8266/Arduino json. 

  3. Bonjour for Windows is needed so you can use `http://esp8266-webupdate.local/update` instead of `http://192.168.1.2/update`. On [Bonjour](https://support.apple.com/bonjour) page you should follow  [Download Bonjour Print Services for Windows v2.0.2](https://support.apple.com/kb/DL999?locale=en_US). If it does not work, then this may indicate the same root cause as you have with not seeing the OTA port in Arduino IDE. 

  Regarding missing 'cpp' file, I have checked and indeed you need to load `WebUpdater.ino.bin`. I guess this has changed in Arduino IDE since release I prepared this manual. Please try it again. It should work. Reset the module once after initial loading using serial.

> My Install directory in Preferences shows as C:\Users********\Documents\Arduino but I don't see my ESP2866 directory there under libraries.

This is normal / not the problem. ESP8266 libraries are in directory you can check under _File > Preferences_ as marked below
[image]

SrikanthBala, 2016-08-21:
OK. Thanks again for the promptness

I tried a few things.
1. Did a clean portable install. Was very hopeful as everything seemed to reinstall from scratch on my USB stick. Flashed BAsic OTA. Waited few secs. No port. Closed IDE restarted. No Port. Do I need to reset the Node MCU before I see anything. My MCU has 2 buttons one says USER the other Flash. I pressed each one for a2 secs. Seems to do nothing. Node MCU does not restart the prog on pressing these buttons and Only when I power cycle can I see the serial port rejoin to the port.
Anyway no port

2. I tried a new approach. Thinking it may be my router problem I used my phone as an AP and connected my PC and ESP to this. New 'pingable' IP showing on the Serial output but again no port. Now switched back to my original router as AP

3. I tried installing Bonjour and found it was already there ( I had done this sometime earlier during my last week of struggle I guess :-) ) Anyway I did another install over it. 
Do I need to do anything to configure Bonjour or something ? Because  http://esp8266-webupdate.local/update  still does not work while instead of  http://192.168.1.2/update does
Does that identify my problem ?

Also when I was looking at my router's 'attached devices' to correctly identify the ESP's IP, the ESP is showing as ESP_F531CC and not ESP8266 as expected. Is it something to do with the underscore character not being acceptable as I read on the forum? How do I change the name of the device ?

Thanks in advance

SrikanthBala, 2016-08-21:
One more comment. I also own a Linkit Smart 7688 Duo which is like an Arduino Yun somewhat. LAst week while playing with it I found that OTA was easy and I could see my port. That was I think IDE 1.6.7 or was it (1.6.9 ?) Anyway on connecting that I cannot see that OTA port either. It was definitely working back then. It also had underscores on its name. Is that the problem ?

SrikanthBala, 2016-08-21:
Ok. I added WiFi.hostname("ESP826612E") thinking this would make a difference. It didn't.
You're right. Using the WebUpdate example, I can upload a sketch successfully. So I finally did my first OTA ! But I still need to use 192.168.1.12 and not http://esp8266-webupdate.local/update. So something wrong with my Bonjour, UDP or some other such setup. Rechecked with my router. It is not set to block any services and I think Bonjour installation should anyway configur my Windows firewall

krzychb, 2016-08-21:
This is great that WebUpdate.ino example finally worked for you! Bonjour on Windows is required to be able to use address like http://esp8266-webupdate.local/update in web browser instead of IP address

I have checked BasicOTA.ino on my Windows 10 PC and sure enough it works.

Arduino OTA is advertised using mDNS. Arduino IDE is able to recognise it starting from release 1.6.7 when [PR-4107](https://github.com/arduino/Arduino/pull/4107) was merged.

Now I would check if your PC is getting mDNS packets sent by ESP. You can do it with http://hobbyistsoftware.com/bonjourBrowser. 

On my PC I see it as below:

[image]

SrikanthBala, 2016-08-22:
Installed Bonjour Browser. On flashing BasicOTA It does not show any running services though I can still web update. How Do I get my PC to see the mDNS packages ?
Also tried a portable install of IDE 1.6.7 with ESP ver 2.3.0 without success

krzychb, 2016-08-22:
> My MCU has 2 buttons one says USER the other Flash. I pressed each one for 2 secs each. Seems to do nothing

Please post a picture of your board. Maybe somebody else can comment.

> Does this isolate the problem to my PC or my Router ?

Given you have tried couple of routers I guess the problems is rather with your PC blocking mDNS packets. 

As you can ping your module I would try to run Arduino OTA update manually. Assuming that IP of your module is 192.168.1.12,, you can run the the following:

```
(...)l\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools\espota.py -i 192.168.1.12 -p 8266 --auth= -f (...)BasicOTA.ino.bin
```

Replace `(...)` with paths to particular files.

When I am back home later today, I will try to check what else you can do to diagnose why Bonjour Browser and Arduino IDE are unable to discover module advertising OTA.

SrikanthBala, 2016-08-22:
**OK ! OTA is working !!!! The port shows up !!! A zillion thanks !!**

Now the truth is I'm not sure how but it is something to do with my router. Yes I had tried another router at my workplace too and it was a different company (TP-link) but I did not work so long and so intensely on it (maybe it has a similar problem which I need to look into) . 
I started googling about this NETGEAR N600 router and posts talked about disabling IPv6. I found mine diabled and I set it to pass thru (why I don't know). 
Then there was a UPnP setting which looked promising. The router online help on it said Turn UPnP On
"UPnP can be enabled for automatic device configuration, or it can be disabled. The default setting for UPnP is enabled. If this feature is disabled, the router will not allow any device to automatically control the resources, such as port forwarding (mapping), of the router. " 
It was already on. So I turned it off. I dont know what did the trick but my OTA port suddenly showed up on Arduino IDE and Bonjour
Now I tried setting IPv6 back to the original setting of 'Disabled' but my port still shows. I set UPnP back to On and my port still shows.

Great to have it working. Would have been greater for me and others to know why ? 

Dear krzychb to quote you from last year "This is great! If I were a woman I would give you a kiss
But I am not – so what about a hug?" . Exactly my feelings 

Thanks a zillion again. From a week of struggle you solved my problem in less than a day !

krzychb, 2016-08-22:
Thanks a lot for feedback and nice words!
:smile: :+1:

SrikanthBala, 2016-08-26:
Hi !. Just reopened this issue to tell you that the problem reappeared and I figured out what was wrong but not why.
http://esp8266sd.local

Then I flashed a mDNS webserver which again worked only when I typed in the IP address. Then it suddenly struck me to check the running windows services and sure enough Bonjour had been listed as stopped, though I don't know why.
It has been set be default to auto restart on reboot. So I rebooted my system with no changes and everything works.

So for some reason Windows 10 had swtched off bonjour services and, since, during my entire week of trouble OTAing, I shut down my system just once, bonjour services was probably off. The one time it worked was when I think I rebooted , but it slipped by notice then.

Just thought I will share this, as it may help someone else.

Sorry to have taken so much of your time last week.

Thanks and Regards.

krzychb, 2016-08-26:
Hi @SrikanthBala,

I am happy to hear beck from you with the follow up and update.

Such very issues pop up every couple of weeks on esp8266 forum or here. Thanks to your persistence this is first case I see it pined down to weird router settings or bonjour service not being restarted by Windows.

Thanks a lot! 

Krzysztof

SrikanthBala, 2016-08-26:
Ok. Regards. Am closing the issue.

closed 2016-08-26 by SrikanthBala

ltvanderkrogt, 2017-04-18:
Hi, 
since a few days I was not able to see my OTA enabled sensors anymore on my Macbook Air. I tried everything I found at all the fora to solve the problem but nothing worked. It took me a lot of time but I solved this issue finally. The Over The Air(OTA) programming is using the Bonjour protocol which is using MulticastDomainNameService(mDNS). So I installed two tools to make the devices visible; 
1) nDNS browser Chrome(https://github.com/GoogleChrome/chrome-app-samples/tree/master/samples/mdns-browser) 
2) Bonjour Browser(http://www.tildesoft.com/) 

My issue occurred after an upgrade of McAfee "Total protection" which apparently switched on its firewall(I did n't know it had..) I already switched off the OSX firewall but this didn't solve the problem. 
Switich off both firewalls did the trick. So I made a firewall rule for McAfee firewall and now it works fine.  

Firewall rule for incoming traffic mdns/Bpnjour
Port:**5353**
TCP/UDP: **UDP** 	
Service/Protocol Name: Multicast DNS (MDNS)
RFC: 3927
Service Name: mdns
Used by: Bonjour, AirPlay, Home Sharing, Printer Discovery, Back to My Mac

AusUser5, 2017-08-29:
Hi

I'm having issue with not seeing OTA devices. This is the first time I've tried OTA so i'm never seen this work on my setup. I'm running Ardunio IDE on Windows 10 Virtual machine hosted on Windows 10 host. I've have NodeMCU board with a sketch that has ArdunioOTA library and config, the NodeMCU has connected to the network and I can ping its IP from all devices.

I'm glad to find this thread and so done the following:

Ardunio IDE 1.8.2 installed on Windows 10
Bonjour Service installed (via intunes) and is running.
Windows 10 Host and Guest Firewall disabled.
Bonjour Browser installed  - 1 device found Iphone (while i have 5 apple device in house, so this is strange!)
mDNS Browser installed - nothing found.

Is there a way I can confirm he NodeMCU is listening on the OTA ports, so that isolate this from the issue?