esp8266/Arduino #2422: Alternate bin file sources for OTA?

Matthew-Bradley, 2016-08-22:
Hi everyone,

I'm looking at a neat application where people can submit code to run a robot, and a server will download the compiled bin file on a "schedule" to the ESP8266, and presumably store it on the flash. I'm wondering if there's a way to point OTA at a stored bin file (or perhaps a string containing hex?), or if this is what the stream interface is supposed to accomplish. (as yet undocumented on the OTA page)

If it works, I think this could really open up OTA and enable people to come up with their own update mechanisms to suit their purpose.

Thanks for all the amazing work you do,
    Matt

brendanmatkin, 2016-08-23:
Does the update server not accomplish this? http://esp8266.github.io/Arduino/versions/2.3.0/doc/ota_updates/readme.html#http-server

Matthew-Bradley, 2016-08-24:
The update server is nice, but it does not.

The update server fetches a file from a web address, but doesn't allow you to load in a file from the ESP8266's flash or other memory. What I'm talking about is a far more general method of loading a file that's been saved in flash or other source, such that it doesn't matter where it came from. If you wanted to, you could rig up some insane arrangement with a laser and photocell link to the ESP8266, write some code to parse and save a bin file to the flash, and then use that to program the ESP8266. You could even write code that parses the code from button presses. Anything, really.

That way, as people write libraries and find new novel ways to connect to an ESP8266, it could support OTA via that method (they only have to write code on each side for a method of encoding/decoding). I can only speculate that the stream interface might allow for this, as it isn't documented.

Matthew-Bradley, 2016-10-01:
To clarify, I'm only looking to have the ability to take a .bin of the firmware that came from a source unspecified by implementation (it doesn't matter *how* it got to be on the ESP) and be able to use it like anything else to reflash the board. This could, for example, be implemented to pull from a byte array in memory, or a .bin file in the flash file-system. Right now, the only way to achieve this functionality would be to have the ESP upload a file to itself, acting as a client to it's own server in the existing server upload example.

This would allow for the vast diversification of OTA methods.

igrr, 2016-10-01:
If your update source provides a Stream interface (or if you can write a wrapper), then you can simply pass the Stream instance to `Updater` global object. It will read binary from stream and perform update. You still need to pass binary image size and md5 signature to Updater, but this doesn't limit your choice of update source.

devyte, 2017-10-14:
Yes, it's possible, per previous comment.
Not the right place to ask such questions. Closing per #3655 .

closed 2017-10-14 by devyte