esp8266/Arduino #2425: Cannot upload sketch by Arduino IDE to NodeMCU on mac

fantasymaker-cn, 2016-08-23:
### Basic Infos

#### Hardware
Hardware:			NodeMCU
Core Version:      	

### Description

Sketch compiled successfully and it has been uploaded and run before. Now I'm using Mac OS and compiling and uploading same sketch. I hava CP210x driver installed. It compiles success but when upload there's no response in the console. Until I unplug the usb cable, the console shows error message:

```
Uploading 249584 bytes from /var/folders/5f/89fpkjd11fgdddkktd98gt2w0000gn/T/buildce5c794997922f8532c793f696916553.tmp/esp-reporter.ino.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 4 bytes
error: failed sending 0xC0
warning: espcomm_send_command: didn't receive command response
error: espcomm_upload_mem failed
```

### Settings in IDE

Module:  			NodeMCU1.0
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

brendanmatkin, 2016-08-23:
I was having a few issues with NodeMCU modules. Maybe not a proper fix but just try programming it as a generic ESP8266 module. That worked well for me. 

If I remember correctly, the downside is that you loose the labelled pin mapping...

fantasymaker-cn, 2016-08-24:
👍@brendanmatkin Thanks very much! This solves the problem! Also the pin mapping of Arduino is lost, it can be used by GPIO number instead.

brendanmatkin, 2016-08-24:
I wonder if we should leave this issue open - even though the work around is easy, it is still broken. Either repairing the issue or just cloning the generic and adding the pin mapping might be the correct solution?

fantasymaker-cn, 2016-08-24:
@brendanmatkin You're very thoughtful. This should be an issue.

devyte, 2017-10-14:
@lixplor is this issue still valid?

devyte, 2018-01-08:
Closing due to lack of feedback in over 2 months, and due to age.

closed 2018-01-08 by devyte