esp8266/Arduino #2428: Unable to program flash: "espcomm_send_command: didn't receive command response"

OrganicIrradiation, 2016-08-24:
### Basic Infos

#### Hardware
Hardware:			Wemos D1 mini (ESP-12F, CH340G USB<->Serial, 4M bytes flash)
Core Version:      	?

### Description

I've been having lots of issues trying to reflash a Wemos D1 mini (ESP-12F, CH340G USB<->Serial, 4M bytes flash). I was previously able to flash it with the Arduino IDE, but I'm worried I've soft-bricked it.

I'm able to read the chip_id and MAC, but apparently not the flash_id:

    $ esptool.py --port /dev/tty.wchusbserial1410 chip_id
    esptool.py v1.1
    Connecting...
    Chip ID: 0x00------
    $ esptool.py --port /dev/tty.wchusbserial1410 read_mac
    esptool.py v1.1
    Connecting...
    MAC: 5c:cf:7f:--:--:--
    $ esptool.py --port /dev/tty.wchusbserial1410 flash_id
    esptool.py v1.1
    Connecting...
    
    A fatal error occurred: Timed out waiting for packet header

If I connect with a serial terminal at 74880 baud I can see the module communicates from the bootloader when I press the restart button:

    ets Jan 8 2013,rst cause:2, boot mode:(3,6)

I've tried on Mac OS X 10.11.6 (with the [signed driver](http://www.wch.cn/download/CH341SER_MAC_ZIP.html)) and on a Windows 10 machine (with automatically installed driver), both with the same results. I've also tried with virtual machines and a multitude of cables. Nothing seems to reflash it (I've tried esptool.py, esptool-ck, NodeMCU Flasher, and the espressif flash tool).

Any suggestions??

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB (3M SPIFFS)
CPU Frequency:		80Mhz
Upload Speed:		115200
Port:		/dev/tty.wchusbserial1410

### Sketch

```cpp

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```

### Debug Messages

```
Sketch uses 222,149 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,524 bytes (38%) of dynamic memory, leaving 50,396 bytes for local variables. Maximum is 81,920 bytes.
~/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd nodemcu -cb 115200 -cp /dev/cu.wchusbserial1410 -ca 0x00000 -cf /var/folders/g1/j8sdbq2x04x0b1dwrwlz_0vw0000gp/T/buildcc7633746e9ac2df9f3c43e9ebc9d838.tmp/Blink.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.wchusbserial1410
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.wchusbserial1410 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 226304 bytes from /var/folders/g1/j8sdbq2x04x0b1dwrwlz_0vw0000gp/T/buildcc7633746e9ac2df9f3c43e9ebc9d838.tmp/Blink.ino.bin to flash at 0x00000000
	erasing flash
	size: 037400 address: 000000
	first_sector_index: 0
	total_sector_count: 56
	head_sector_count: 16
	adjusted_sector_count: 40
	erase_size: 028000
An error occurred while uploading the sketch
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting timeout 100
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
closing bootloader
error: espcomm_upload_mem failed
```

WereCatf, 2016-08-24:
AFAIK there is no way, whatsoever, of soft-bricking an ESP8266, maybe you have broken a resistor or something reset-circuit and that causes your problem -- it looks to me like the IDE or the esptool isn't able to reset the board into programming-mode. You could manually place the ESP into UART programming-mode using pull-up and pull-down resistors according to https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup before trying to flash something, then when flashed remove the resistors and reset the board and see if it works.

OrganicIrradiation, 2016-08-24:
Thanks for the suggestion. According to the link you sent, I should have the following:

> GPIO 0: LOW
> GPIO 2: HIGH
> GPIO 15: LOW

The [D1 mini schematic](http://www.wemos.cc/Products/images/d1_mini.pdf) shows that GPIO 0 and 2 have 10k pull-up resistors and GPIO 15 has a pull-down. So, I'll try with a pull-down on Pin D3 (GPIO 0) and GND tomorrow morning and report back. I had tried with a 4.7k (only loose resistor available on hand right now), which did not change the reported Arduino output. However, the 4.7k pull-down & 10k pull-up might not have created a sufficiently low voltage divider (~1.5V if my math is correct).

OrganicIrradiation, 2016-08-24:
Thanks for the suggestion, @WereCatf! Fixed by putting a 10 Ohm resistor between pins D3 (DIO 0) and GND! Seems as though the reset & programming circuit pull-down timing, that I had relied upon for so long, is off (see right side of USB to UART section of the [Wemos D1 mini schematic](http://www.wemos.cc/Products/images/d1_mini.pdf))

    Sketch uses 222,149 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
    Global variables use 31,524 bytes (38%) of dynamic memory, leaving 50,396 bytes for local variables. Maximum is 81,920 bytes.
    ~/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9/esptool -vv -cd nodemcu -cb 115200 -cp /dev/cu.wchusbserial1410 -ca 0x00000 -cf /var/folders/g1/j8sdbq2x04x0b1dwrwlz_0vw0000gp/T/buildcc7633746e9ac2df9f3c43e9ebc9d838.tmp/Blink.ino.bin 
    esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 115200
        setting port from /dev/tty.usbserial to /dev/cu.wchusbserial1410
        setting address from 0x00000000 to 0x00000000
        espcomm_upload_file
        espcomm_upload_mem
    opening port /dev/cu.wchusbserial1410 at 115200
        tcgetattr
        tcsetattr
        serial open
    opening bootloader
    resetting board
    trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
    trying to connect
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
    Uploading 226304 bytes from /var/folders/g1/j8sdbq2x04x0b1dwrwlz_0vw0000gp/T/buildcc7633746e9ac2df9f3c43e9ebc9d838.tmp/Blink.ino.bin to flash at 0x00000000
        erasing flash
        size: 037400 address: 000000
        first_sector_index: 0
        total_sector_count: 56
        head_sector_count: 16
        adjusted_sector_count: 40
        erase_size: 028000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting timeout 15000
        setting timeout 100
        espcomm_send_command: receiving 2 bytes of data
        writing flash
    ................................................................................ [ 36% ]
    ................................................................................ [ 72% ]
    .............................................................                    [ 100% ]
    starting app without reboot
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
    closing bootloader

closed 2016-08-24 by OrganicIrradiation

sabag, 2018-03-07:
Hi, 
Did someone got to solve this ?
I am still (2018-03) getting this error while using Arduino IDE 1.8.5 , trying to flash a "previously flashed many times" ESP12-F using a powered module with 3v3 and FTDI. but I am getting the exact output attached above.
i tried 1M and the default but nothing works.

nophead, 2018-04-01:
I had this problem and solved it by adding a 100nF ceramic decoupling capacitor across the supply rails, close to the module.

nophead, 2018-08-17:
Here is my explanation to why programming fails sometimes: http://hydraraptor.blogspot.com/2018/04/esp8266-contention-during-programming.html

Buffer8888, 2019-01-29:
It worked, however I had to also unplug USB first, connect D3 to GND, plug in USB again, run upload sketch, disconnect D3 afterwards. Good Luck.

jandrassy, 2019-03-01:
10 Ohm resistor? use plain wire.