esp8266/Arduino #2429: Cheap Ebay esp01 programmer

penfold42, 2016-08-24:
http://vi.raptor.ebaydesc.com/ws/eBayISAPI.dll?ViewItemDescV4&item=222121082897&category=4660&pm=1&ds=0&t=1472017038142

Has anyone worked out how to get this working ?

Neither nodemcu nor ck reset methods work

WereCatf, 2016-08-24:
The description in your link rather clearly says that it doesn't have a reset-method, you need to implement it yourself by e.g. copying the circuit from the Nodemcu-schematics and implementing it yourself.

penfold42, 2016-08-24:
Thanks for your reply - True - I misread their note before posting this.

I would recommend against buying these - they are not a programmer as shipped and getting access to the DTR  and RTS pins on the ch340 isn't easy with the cap in the way.

Shame really - 2 transistors and resistors would have been easy to include at design time.

I just finished putting a couple of buttons on reset and gpio0 which is good enough for occasional use

closed 2017-10-14 by devyte