esp8266/Arduino #2430: Start Serial with only TX enabled

supersjimmie, 2016-08-24:
#### Hardware
Hardware:			NodeMCU v2
Core Version:      	Latest GIT

### Description

Problem description

### Settings in IDE

Module:  			NodeMCU
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch
#include "DHT.h"
#define DHTTYPE DHT22
#define DHTPIN D9  //  D9 is Serial RX
DHT dht(DHTPIN, DHTTYPE);

```cpp
void setup() {
  Serial.begin(115200);
  dht.begin();
}

void loop() {
  Serial.println(dht.readTemperature());
}
```

This is a stripped version of my code, which just describes the question.
I would like to use pin D9 for DHT (input), because I only use Serial for debugging output.
Is it possible to begin Serial with some parameter so it will only use it's TX pin and leave the RX pin free/available for other usage?

WereCatf, 2016-08-24:
Looking at the HardwareSerial.h it looks like `Serial.begin(115200, SERIAL_8N1, SERIAL_RX_ONLY);` would do the trick.

supersjimmie, 2016-08-24:
I tried this and it compiles and outputs my debugging:
```
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY, 1);   //     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin);
```
But it seems that the D9 is still not usable, because the DHT still does not communicate.

WereCatf, 2016-08-24:
Does it work if you try it on some other pin?

supersjimmie, 2016-08-24:
You're just a second faster than me.
Reverted back to D4 where it worked and that stopped working, so I guess I have another hardware issue now.  Will come back later...

supersjimmie, 2016-08-24:
Thanks @WereCatf 
Yep, works now.
I can upload a new sketch with the DHT connected to D9 (RX) and use the DHT when running.
(only my DHT22 modules are getting crappy, sometimes they don't work at any pin anymore)

closed 2016-08-24 by supersjimmie

WereCatf, 2016-08-24:
Good to hear :)