esp8266/Arduino #2433: err already associed!

EagleSmith, 2016-08-25:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12
Core Version:      	2.2.0

### Description

Problem description

### Settings in IDE

Module:  			ESPDUINO
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		ck

### Sketch

:ww
Sent WiFi page
:ur 1
:close
WS:dis
:del
wifi evt: 7
wifi evt: 7
wifi evt: 7
err already associed!
station: 84:3a:4b:3e:b2:88 leave, AID = 2
rm 2
wifi evt: 6

EagleSmith, 2016-08-25:
Sorry,the hardware and Core vesion is wrong.

Hardware: ESP-13
Core Version: 2.3.0

tablatronix, 2017-09-15:
I have a similar problem, i get random disconnects like this

```
err already associed!
station: 00:24:d7:9c:85:74 leave, AID = 1
rm 1
add 1
aid 1
station: 00:24:d7:9c:85:74 join, AID = 1
LmacRxBlk:1
```

devyte, 2017-10-14:
@EagleSmith can't help without a MCVE sketch.
I've seen this when you try to call WiFi.begin() twice.
Closing due to lack of info and/or user error, see #3655 .

closed 2017-10-14 by devyte

tablatronix, 2017-10-14:
There is no repro sketch, I get this just randomly, Will reopen if I find a way to reproduce.
Worth noting I have 30 dup SSID on this network

kapyaar, 2018-03-05:
@tablatronix Just to let you know, I came across this issue and after many things, the solution that worked was to restart the router.

tablatronix, 2018-03-05:
Yeah I still see this every now and then, but never in any reproducible method, so that makes sense that it could be router issue, some sort of client caching or session.

wkraft-fablabka, 2019-09-26:
Hello,

when flashing a couple "Wemos D1 mini" for an workshop I ran into this problem.  I was not able to connect to the ESP8266 either
 * because I could not see the SSID on my client at all
 * I saw the SSID, but received logs like:

>SDK:2.2.2-dev(38a443e)/Core:2.5.2-151-g3733ece7=20502151/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-8-g7958710/BearSSL:89454af
>add 1
>aid 1
>station: 00:21:6a:71:be:2e join, AID = 1
>err already associed!
>station: 00:21:6a:71:be:2e leave, AID = 1
>rm 1
>max connection!
>max connection!
>max connection!
>max connection!
>add 1
>aid 1
>station: 00:21:6a:71:be:2e join, AID = 1
>err already associed!
>station: 00:21:6a:71:be:2e leave, AID = 1
>rm 1
>max connection!
>max connection!
>max connection!
>add 1
>aid 1
>station: 00:21:6a:71:be:2e join, AID = 1
>err already associed!
>station: 00:21:6a:71:be:2e leave, AID = 1
>rm 1
>max connection!
>add 1
>aid 1
>station: 00:21:6a:71:be:2e join, AID = 1
>err already associed!
>station: 00:21:6a:71:be:2e leave, AID = 1
>rm 1
>max connection!

until the Client told me, that a connection could not be established. 
If the WiFi connection could be established, the TCP connections always went down.

Trying with different Clients and WLAN Adapters I figured out that this happened under Windows 10, Windows 7 and Ubuntu 18-04 with an "Intel Wifi Link 5300" or an "Realtek RTL8188ETV" WLAN-Chipset.
Using an Atheros9271 with all this three OS everything was OK.

I have currently 10 broken devices and 8 working onces from one order Batch.  Besides one, which has Chinese laser-labeling on the ESP, all the others are looking the same.